RainerKuemmerle/g2o #209: error: no matching constructor for initialization of 'g2o::OptimizationAlgorithmLevenberg'

Kay1794, 2017-09-26:
Hi,
I'm new to g2o.When I am trying to compile lsd-slam on mac using the newest g2o version I met the problem below:
`error: no matching constructor for
      initialization of 'BlockSolver' (aka 'g2o::BlockSolver<g2o::BlockSolverTraits<7, 3> >')
        BlockSolver* blockSolver = new BlockSolver(solver);
`

`error: no matching constructor for
      initialization of 'g2o::OptimizationAlgorithmLevenberg'
        g2o::OptimizationAlgorithmLevenberg* algorithm = new g2o::OptimizationAlgorithmLevenberg(blockSolver);`

Does anyone have idea what happened here?

jumonatr, 2017-09-26:
Types now require that you pass in unique_ptrs when an object assumes ownership of the types passed in. You can call g2o::make_unique if you don't have C++17 enabled or std::make_unique if you do.

g2o\examples\ba\ba_demo.cpp:137 is an example of how you'd create the objects now.

Kay1794, 2017-09-26:
Exactly, I just fixed it up follow the example. thank you so much! @jumonatr

closed 2025-02-10 by github-actions[bot]