RainerKuemmerle/g2o #210: Undefined symbols for architecture x86_64 using Macos 10.12.6

Kay1794, 2017-09-27:
[image]
Hi,

I met this problem when I built a project using g2o.
My cmake flags are -DCMAKE_BUILD_TYPE=Release -DBUILD_CSPARSE=ON
Not sure if it is a link problem using g2o on mac

Does anyone have the same problem?

Dnaed, 2017-10-31:
Stuck with the same for about a week. Any solutions ??

Dnaed, 2017-11-01:
Stuck with the same for about a week. Any solutions?

sjulier, 2017-11-01:
One idea is that you seem to be linking with a static library (liblsd_slam.a). This won't, automatically, pull in all the dependent libraries. Can you run:

make VERBOSE=1

and see if it's trying to link in the csparse libraries at all? If not, you'll have to manually include csparse in the target_link_libraries.

(There is a branch on github which uses new-style cmake options which should hopefully address this problem.)

niebayes, 2020-05-14:
1. pull the latest repo of g2o from Github.
2. compile it.
3. find_package() 
4. include_directories()
5. target_link_libraries() --- NOTICE: G2O has two libraries ${G2O_CORE_LIBRARY} and ${G2O_STUFF_LIBRARY}, instead of the ordinary one ${G2O_LIBRARIES}. You need to specify both of them in this command.

wbrennan899, 2020-06-09:
> * pull the latest repo of g2o from Github.
> * compile it.
> * find_package()
> * include_directories()
> * target_link_libraries() --- NOTICE: G2O has two libraries ${G2O_CORE_LIBRARY} and ${G2O_STUFF_LIBRARY}, instead of the ordinary one ${G2O_LIBRARIES}. You need to specify both of them in this command.

@niebayes Could you please provide a little more detail on how to implement this? Do you think it will solve the issue I just posted? https://github.com/RainerKuemmerle/g2o/issues/423

closed 2025-02-09 by github-actions[bot]