RainerKuemmerle/g2o #212: Process finished with exit code 139 (interrupted by signal 11: SIGSEGV) when g2o_types_sba linked

soaph, 2017-09-30:
My Development Environment is:
OS: Arch Linux x86_64
cmake: 3.9.3
gcc: 7.2.0
eigen: 3.3.4
suitesparse 4.5.4
opencv: 3.3.0

After I compiled g2o, I tried it with a simple curve fitting program:

```cpp
#include <iostream>
#include <g2o/core/base_vertex.h>
#include <g2o/core/base_unary_edge.h>
#include <g2o/core/block_solver.h>
#include <g2o/core/optimization_algorithm_levenberg.h>
#include <g2o/core/optimization_algorithm_gauss_newton.h>
#include <g2o/core/optimization_algorithm_dogleg.h>
#include <g2o/solvers/dense/linear_solver_dense.h>
#include <Eigen/Core>
#include <opencv2/core/core.hpp>
#include <cmath>
#include <chrono>
using namespace std;

class CurveFittingVertex : public g2o::BaseVertex<3, Eigen::Vector3d>{
public:
    EIGEN_MAKE_ALIGNED_OPERATOR_NEW
    virtual void setToOriginImpl(){
        _estimate << 0,0,0;
    }
    virtual void oplusImpl(double const * update){
        _estimate += Eigen::Vector3d(update);
    }
    virtual bool read(istream& in){}
    virtual bool write(ostream& out) const {}
};

class CurveFittingEdge : public g2o::BaseUnaryEdge<1,double,CurveFittingVertex>{
public:
    EIGEN_MAKE_ALIGNED_OPERATOR_NEW
    CurveFittingEdge(double x) : BaseUnaryEdge(), _x(x) {}
    void computeError(){
        const CurveFittingVertex* v = static_cast<const CurveFittingVertex*>(_vertices[0]);
        const Eigen::Vector3d abc = v->estimate();
        _error[0] = _measurement - std::exp(abc(0,0)*_x*_x+abc(1,0)*_x+abc(2,0));
    }
    virtual bool read(istream& in) {}
    virtual bool write(ostream& out) const {}
private:
    double _x;
};

int main(){
    double a = 1.0, b = 2.0, c = 1.0;
    int N = 100;
    double w_sigma = 1.0;
    cv::RNG rng;
    double abc[3] = {0,0,0};
    vector<double> x_data, y_data;

    cout << "generating date ..." << endl;
    for (int i = 0; i < N; ++i){
        double x = i/(double)N;
        x_data.push_back(x);
        y_data.push_back(std::exp(a*x*x+b*x+c) + rng.gaussian(w_sigma));
        cout << x_data[i] << " " << y_data[i] << endl;
    }

    typedef g2o::BlockSolver<g2o::BlockSolverTraits<3,1>> Block;
    typedef g2o::LinearSolverDense<Block::PoseMatrixType> LinearSolver;
    g2o::OptimizationAlgorithmLevenberg* solver = new g2o::OptimizationAlgorithmLevenberg(
            g2o::make_unique<Block>(g2o::make_unique<LinearSolver>())
    );

    g2o::SparseOptimizer optimizer;
    optimizer.setAlgorithm(solver);
    optimizer.setVerbose(true);

    CurveFittingVertex* v = new CurveFittingVertex();
    v->setEstimate(Eigen::Vector3d(0,0,0));
    v->setId(0);
    optimizer.addVertex(v);

    for (int i = 0; i < N; ++i){
        CurveFittingEdge* edge = new CurveFittingEdge(x_data[i]);
        edge->setId(i);
        edge->setVertex(0, v);
        edge->setMeasurement(y_data[i]);
        edge->setInformation(Eigen::Matrix<double,1,1>::Identity()/(w_sigma*w_sigma));
        optimizer.addEdge(edge);
    }

    cout << "start optimization" << endl;
    chrono::steady_clock::time_point t1 = chrono::steady_clock::now();
    optimizer.initializeOptimization();
    optimizer.optimize(100);
    chrono::steady_clock::time_point t2 = chrono::steady_clock::now();
    chrono::duration<double> time_used = chrono::duration_cast<chrono::duration<double>>(t2-t1);
    cout << "solve time cost = " << time_used.count() << " seconds. " << endl;

    Eigen::Vector3d abc_estimate = v->estimate();
    cout << "estimate model: " << abc_estimate.transpose() << endl;

    return 0;
}
```

and the following is the CMakeLists.txt:

```cmake
cmake_minimum_required(VERSION 3.8)
project(g2o_curve_fitting)

set(CMAKE_CXX_STANDARD 11)

list(APPEND CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR}/cmake-build-debug)

find_package(OpenCV REQUIRED)
find_package(Eigen3 REQUIRED)
find_package(G2O REQUIRED)

include_directories(${G2O_INCLUDE_DIRS} ${EIGEN3_INCLUDE_DIRS})

set(SOURCE_FILES main.cpp)
add_executable(g2o_curve_fitting ${SOURCE_FILES})

target_link_libraries(g2o_curve_fitting ${OpenCV_LIBS} g2o_core g2o_stuff g2o_types_sba)
```

there is a FindG2O.cmake in the `cmake-build-debug` subdirectory, with the same content as the one in this repository.

Compiling this is OK, but when I run the binary, even before the first line of main function, it failed:

```Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)```

But if I remove the last linked library `g2o_types_sba`, the program can run normally.

**Note**: the last linked library `g2o_types_sba` is not used anywhere in this .cpp file, but I deliberately added it, because I had tried to use sparse bundle adjustment in another program but had failed, so I tried to test it in this program. It seems that this .so file is broken? But it's pretty strange that the examples compiled during compiling g2o can run without any incident.

Please help me ~ I don't know how to deal with it now .....

jvhoven, 2017-11-17:
Hi,

We're currently trying to implement Bundle Adjustment for our school project and we ran into  similar problems. In order to bypass these problems we started using an older version of g2o to link against ([this one to be specific](https://github.com/RainerKuemmerle/g2o/tree/2e3566968c1f0e5608155a2a26154b98d6d752dc)). 

I'm not sure what the actual problem is but there have been updates concerning memory usage and I'm guessing that has to do with the problem.

Somliga, 2018-04-09:
I also roll backed to the version mention above and it worked, my issue was also with the types_sba library

DellBrother, 2018-05-01:
Same case for me. Sometime just linking (no invoking) the g2o_types_sba.so will cause the problem. sometimes instance creation such as g2o::VertexSE3Expmap * pose = new g2o::VertexSE3Expmap()  will cause the problem. 

sb. says that Eigen 3.2.10 can fix this issue. I tried but not working. 
sb. says it is caused by environment setup. I am not sure.

be waiting for answer. thanks.

YangSiri, 2019-11-09:
Hi, @DellBrother 
You statement seems to be my problem too, I'm wondering if you have any solution? That would be great!

Looking forward to you reply. THANKS

closed 2025-02-09 by github-actions[bot]