RainerKuemmerle/g2o #220: Covariance recovery success depends on the solver

mmattamala, 2017-10-23:
Hi @RainerKuemmerle,

First, thank you for the library, I has been very useful for my research :)

I'm facing some issues to recover the covariance while using the `computeMarginals`
 method, similar to #207, I think. Particularly, I'm running the `static_target.cpp` example and I obtain different results depending on the solver I choose (I'm editing [this](https://github.com/RainerKuemmerle/g2o/blob/4b9c2f5b68d14ad479457b18c5a2a0bce1541a90/g2o/examples/target/static_target.cpp#L59) line).

For instance, the default solver is CHOLMOD. In that case, I get the following output for the covariance:
```
covariance
RBI: 1 3
CBI: 1 3
BLOCK: 0 0
0.100002       -0       -0
      -0 0.100002       -0
      -0       -0 0.100002
```

However, if I change the solver to Eigen, Dense or PCG, I get the following:
```
covariance
RBI: 0
CBI: 0
```

In addition, If I run the example with gdb, I got the following error:
```
#0  0x00000000005bc654 in std::_Rb_tree<int, std::pair<int const, Eigen::Matrix<double, -1, -1, 0, -1, -1>*>, std::_Select1st<std::pair<int const, Eigen::Matrix<double, -1, -1, 0, -1, -1>*> >, std::less<int>, std::allocator<std::pair<int const, Eigen::Matrix<double, -1, -1, 0, -1, -1>*> > >::_M_begin (this=0x0) at /usr/include/c++/5/bits/stl_tree.h:652
#1  0x00000000005ba763 in std::_Rb_tree<int, std::pair<int const, Eigen::Matrix<double, -1, -1, 0, -1, -1>*>, std::_Select1st<std::pair<int const, Eigen::Matrix<double, -1, -1, 0, -1, -1>*> >, std::less<int>, std::allocator<std::pair<int const, Eigen::Matrix<double, -1, -1, 0, -1, -1>*> > >::find (this=0x0, __k=@0x7fffffffd134: 0) at /usr/include/c++/5/bits/stl_tree.h:2295
#2  0x00000000005b821b in std::map<int, Eigen::Matrix<double, -1, -1, 0, -1, -1>*, std::less<int>, std::allocator<std::pair<int const, Eigen::Matrix<double, -1, -1, 0, -1, -1>*> > >::find (this=0x0, __x=@0x7fffffffd134: 0) at /usr/include/c++/5/bits/stl_map.h:846
#3  0x00000000005b50d7 in g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::block (this=0x7fffffffd220, r=0, c=0, 
    alloc=false) at /home/matias/libs/g2o/g2o/core/sparse_block_matrix.hpp:92
#4  0x00000000005aeb7e in main () at /home/matias/libs/g2o/g2o/examples/target/static_target.cpp:126
```

I'm particularly concerned about Eigen solver because this problem began while recovering the covariance in ORB_SLAM, which uses it. Maybe my problem would be solved If I change the solver, but I want to understand why it fails in this case. Any advice about this issue would be very useful.

Thanks!

RainerKuemmerle, 2017-11-01:
Not all solvers implement the solveBlocks method which is required for recovering the covariance.
The CSparse solver does, Eigen so far not.

mmattamala, 2017-11-01:
Great! Thank you Rainer for the clarification.

closed 2017-11-01 by mmattamala