RainerKuemmerle/g2o #225: Can 3D line and 3D point be involved simultaneously in SBA?

rikpires, 2017-11-24:
I try to implement a point and line slam. It works fine in pose only estimation step. In the SBA step, I find I don't know how to involve 3D line and 3D point simultaneously. Because point landmark has a dimension of 3 while a line has a dimension of 6, I cannot use BlockSolver_6_3 directly.
Is it possible to do this in g2o? Thanks.

ananhan, 2017-12-07:
Have you already solved the problem,I also want to combine point and line together in VSLAM

decster, 2018-03-11:
Also wonder how this could be done

Mechazo11, 2024-04-06:
Leaving this here for future reference. Found this when I was looking into object detection in VSLAM pipelines

[point-plane-object-slam](https://github.com/benchun123/point-plane-object-SLAM)

closed 2025-02-09 by github-actions[bot]