RainerKuemmerle/g2o #231: g2o_viewer

Young532042725, 2017-12-14:
Ubuntu 16.04, when i run the g2o_viewer,some problem occur. i cannot solve this problem myself. could someone help me? thank you very very much.
[image]

CCB307, 2018-01-08:
Did you compile g2o_viewer with Qt5? I have a error when I compile g2o_viewer.
[image]

osiz-blockchainapp, 2019-03-18:
I am facing the same error, did you got any solution? 
/usr/lib/x86_64-linux-gnu/qt5/bin/uic: symbol lookup error: /usr/lib/x86_64-linux-gnu/qt5/bin/uic: undefined symbol: qt_qhash_seed

closed 2021-02-06 by RainerKuemmerle