RainerKuemmerle/g2o #242: Build breaks with the latest changes

yurivict, 2017-12-24:
```
/usr/ports/science/g2o/work-nosimd/g2o-20170730_git-51-gc6e5df8/g2o/stuff/sparse_helper.h:39:69: error: unknown type name 'number_t'
  G2O_STUFF_API bool writeVector(const std::string& filename, const number_t*v, int n);
                                                                    ^
/usr/ports/science/g2o/work-nosimd/g2o-20170730_git-51-gc6e5df8/g2o/stuff/sparse_helper.h:44:120: error: unknown type name 'number_t'
  G2O_STUFF_API bool writeCCSMatrix(const std::string& filename, int rows, int cols, const int* p, const int* i, const number_t* v, bool upperTriangleSymmetric = true);
                                                                                                                       ^
```

yurivict, 2017-12-24:
The revision 56b0344 still breaks:

```
In file included from /usr/ports/science/g2o/work-sse2/g2o-20170730_git-58-g56b0344/g2o/stuff/sparse_helper.cpp:27:
/usr/ports/science/g2o/work-sse2/g2o-20170730_git-58-g56b0344/g2o/stuff/sparse_helper.h:40:69: error: unknown type name 'number_t'
  G2O_STUFF_API bool writeVector(const std::string& filename, const number_t*v, int n);
                                                                    ^
/usr/ports/science/g2o/work-sse2/g2o-20170730_git-58-g56b0344/g2o/stuff/sparse_helper.h:45:120: error: unknown type name 'number_t'
  G2O_STUFF_API bool writeCCSMatrix(const std::string& filename, int rows, int cols, const int* p, const int* i, const number_t* v, bool upperTriangleSymmetric = true);
                                                                                                                       ^
/usr/ports/science/g2o/work-sse2/g2o-20170730_git-58-g56b0344/g2o/stuff/sparse_helper.cpp:43:7: error: unknown type name 'number_t'
      number_t x;
      ^
/usr/ports/science/g2o/work-sse2/g2o-20170730_git-58-g56b0344/g2o/stuff/sparse_helper.cpp:44:36: error: unknown type name 'number_t'
      TripletEntry(int r_, int c_, number_t x_) : r(r_), c(c_), x(x_) {}
                                   ^
```

RainerKuemmerle, 2017-12-24:
Can you please post your g2o/config.h
This should include a definition for number_t

yurivict, 2017-12-24:
```
#ifndef G2O_CONFIG_H
#define G2O_CONFIG_H

#define G2O_HAVE_OPENGL 1
#define G2O_OPENGL_FOUND 1
/* #undef G2O_OPENMP */
#define G2O_SHARED_LIBS 1
#define G2O_LGPL_SHARED_LIBS 1

// available sparse matrix libraries
/* #undef G2O_HAVE_CHOLMOD */
#define G2O_HAVE_CSPARSE 1

/* #undef G2O_NO_IMPLICIT_OWNERSHIP_OF_OBJECTS */

#ifdef G2O_NO_IMPLICIT_OWNERSHIP_OF_OBJECTS
#define G2O_DELETE_IMPLICITLY_OWNED_OBJECTS 0
#else
#define G2O_DELETE_IMPLICITLY_OWNED_OBJECTS 1
#endif

/* #undef G2O_SINGLE_PRECISION_MATH */
#ifdef G2O_SINGLE_PRECISION_MATH
    #define G2O_NUMBER_FORMAT_STR "%g"

    #ifdef __cplusplus
        using number_t = float;
    #else
        typedef float number_t;
    #endif
#else
    #define G2O_NUMBER_FORMAT_STR "%lg"

    #ifdef __cplusplus
        using number_t = double;
    #else
        typedef double number_t;
    #endif
#endif

#define G2O_CXX_COMPILER "Clang /usr/bin/c++"

#ifdef __cplusplus
#include <g2o/core/eigen_types.h>
#endif

#endif
```

RainerKuemmerle, 2017-12-24:
What build environment are you using?
All CI and my local builds are running smoothly.
Maybe there is some older config.h included instead of the one from the build server.

Can you run
make VERBOSE=1 stuff
to see how the compiler call looks like.
Then copy&paste this call, remove the object generation and specify to see the pre-processor output (-E):
For me this gives:
`/usr/bin/c++   -DUNIX -Dstuff_EXPORTS -I/home/goki/workspace/g2o -I/usr/include/eigen3 -I/home/goki/workspace/g2o/build  -Wall -W -O3 -DNDEBUG -O3 -march=native -fPIC   -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -std=gnu++11 -E -c /home/goki/workspace/g2o/g2o/stuff/sparse_helper.cpp | grep number_t`

Output is:
```c++
using number_t = double;
  using VectorN = Eigen::Matrix<number_t, N, 1, Eigen::ColMajor>;
  typedef Eigen::Matrix<number_t,2,2,Eigen::ColMajor> Matrix2;
  typedef Eigen::Matrix<number_t,3,3,Eigen::ColMajor> Matrix3;
  typedef Eigen::Matrix<number_t,4,4,Eigen::ColMajor> Matrix4;
  typedef Eigen::Matrix<number_t,Eigen::Dynamic,Eigen::Dynamic,Eigen::ColMajor> MatrixX;
  typedef Eigen::Transform<number_t,2,Eigen::Isometry,Eigen::ColMajor> Isometry2;
  typedef Eigen::Transform<number_t,3,Eigen::Isometry,Eigen::ColMajor> Isometry3;
  typedef Eigen::Transform<number_t,2,Eigen::Affine,Eigen::ColMajor> Affine2;
  typedef Eigen::Transform<number_t,3,Eigen::Affine,Eigen::ColMajor> Affine3;
  typedef Eigen::Rotation2D<number_t> Rotation2D;
  typedef Eigen::Quaternion<number_t> Quaternion;
  typedef Eigen::AngleAxis<number_t> AngleAxis;
  bool writeVector(const std::string& filename, const number_t*v, int n);
  bool writeCCSMatrix(const std::string& filename, int rows, int cols, const int* p, const int* i, const number_t* v, bool upperTriangleSymmetric = true);
      number_t x;
      TripletEntry(int r_, int c_, number_t x_) : r(r_), c(c_), x(x_) {}
  bool writeVector(const string& filename, const number_t*v, int n)
  bool writeCCSMatrix(const string& filename, int rows, int cols, const int* Ap, const int* Ai, const number_t* Ax, bool upperTriangleSymmetric)
```
which indicates that number_t is defined.

yurivict, 2017-12-24:
> What build environment are you using?
FreeBSD 11.1

The previous revisions worked fine.

This command:
```
/usr/bin/c++  -DUNIX -Dstuff_EXPORTS -I/usr/local/include -I/usr/ports/science/g2o/work-nosimd/g2o-20170730_git-58-g56b0344 -I/usr/local/include/eigen3 -I/usr/ports/science/g2o/work-nosimd/.build -O2 -pipe -fno-omit-frame-pointer -fstack-protector -fno-strict-aliasing -fno-omit-frame-pointer  -Wall -O2 -pipe -fno-omit-frame-pointer -fstack-protector -fno-strict-aliasing -fno-omit-frame-pointer -fPIC   -std=gnu++11 -Wall -W -O3 -DNDEBUG -O3 -march=native -fPIC -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -std=gnu++11 -E /usr/ports/science/g2o/work-nosimd/g2o-20170730_git-58-g56b0344/g2o/stuff/sparse_helper.cpp | grep number_t
```

Prints this:
```
                bool writeVector(const std::string& filename, const number_t*v, int n);
                bool writeCCSMatrix(const std::string& filename, int rows, int cols, const int* p, const int* i, const number_t* v, bool upperTriangleSymmetric = true);
      number_t x;
      TripletEntry(int r_, int c_, number_t x_) : r(r_), c(c_), x(x_) {}
  bool writeVector(const string& filename, const number_t*v, int n)
  bool writeCCSMatrix(const string& filename, int rows, int cols, const int* Ap, const int* Ai, const number_t* Ax, bool upperTriangleSymmetric)
```
It doesn't look like your output.

RainerKuemmerle, 2017-12-24:
Maybe there is some older g2o/config.h lying around in the paths specified
by -I and is used instead of the current generated one.
I assume that an installed older version could lead to this effect
(g2o/config.h under /usr/local/include).
Commit 46f615ca32bd98a0f8a7ab1cb429566c43000e98 includes this change.

yurivict, 2017-12-24:
You're right, removing the previous installed version fixed the problem.

commit 85385331 (2017-12-24): add missing include

fix #242
closed 2017-12-24 by yurivict