RainerKuemmerle/g2o #246: Please explain in README how do you depend on SSE level?

yurivict, 2017-12-27:
CMakeLists.txt supplies --msse{N} options based on autodetection or user choice.
Could you please explain which operations and how actually depend on SSE?

RainerKuemmerle, 2017-12-29:
http://eigen.tuxfamily.org/index.php?title=FAQ#Vectorization

yurivict, 2017-12-29:
Hi Rainer,

FYI, I created the FreeBSD port for g2o. https://reviews.freebsd.org/D13610 However, you might be interested to know that SIMD use runs into some packaging issues and spurs some controversy.

When SIMD is selected during build, as opposed to during runtime, this causes the need to have flavors of this application for different SIMD levels. If properly implemented, SIMD selection should happen at the package manager level. It should choose the right SIMD flavor for the system.

However, this isn't properly implemented in FreeBSD. I doubt this is properly implemented in RedHat, Debian or Gentoo either. In FreeBSD, SIMD can be mapped to port options, which can then be manually chosen by users. Package needs to be manually rebuilt.

It is better if the project can do the runtime SIMD selection. I would like to ask you a question: do you think g2o can be modified to do the runtime SIMD selection? My guess is that this isn't likely because it's Eigen that does SIMD for g2o.

closed 2018-03-28 by yurivict