RainerKuemmerle/g2o #248: Cross compile for rpi3 64 bits

Minipada, 2017-12-30:
I am currently trying to cross compile the software for a rpi3 in 64 bits (aarch64) with GCC/GXX 5.4

I am have managed to set the configuration flags but I'm coming across this issue:

    error: cannot convert 'ptrdiff_t* {aka long int*}' to 'int*' in assignment

This seems to be due to architecture. Do you know if there is a flag to enable to handle it?

Thanks

jumonatr, 2018-01-02:
It's probably just a case where the code is assigning to the wrong type, but it happens to be the right one when building for x86. What line is this occurring on?

vyi, 2020-12-07:
Hi Guys,

I came across this same issue while building g2o for arm64-v8a (on Android) using CMake based ndk-toolchain. For me this issue was coming while compiling the csparse solver. 

I tried to manually cast `ptrdiff_t` to `int *` but there are too many places where this is needed (in files **linear_csparse_solver**, **csparse_extension**, **core/marginal_covariance_cholesky** ..) and I'm afraid this might break other things !!! 

I found that **rk** had used the following define in the included csparse lib (file **EXTERNAL/csparse/cs.h**):

```
// rk: We define csi to be int to be backward compatible with older CSparse releases.
//     At some point we might adapt the internal structures of g2o to use the same type
//     as the one given is this header file.
#define csi int

```

This was helping with the build. However it was throwing compile time error when using the library for a **arm64-v8a** platform.

Now what is the correct way to fix this issue? 

Using the following branch : 20200410_git

jiapei100, 2021-12-31:
Same here... On **X86_64**.
Any solutions?

closed 2025-02-08 by github-actions[bot]