RainerKuemmerle/g2o #250: I got an error about Eigen3("data is not aligned") when using my own edge and vertex to do optimization

zouxianghong, 2018-01-15:
1.Problem Description: 
when using my own edge() and vertex() to do optimization, i got an error: "((size_t(m_data) % (((int)1 >= (int)internal::traits<Derived>::Alignment) ? (int)1 : (int)internal::traits<Derived>::Alignment)) == 0) && \"data is not aligned\""
2.Code:
2.1.Definition of Vertex and Edge
`enum BlendType
  {
    POSE_ESTIMATION_2D2D = 0,
    POSE_ESTIMATION_2D3D_1 = 1,
    POSE_ESTIMATION_2D3D_2 = 2,
    POSE_ESTIMATION_3D2D_1 = 3,
    POSE_ESTIMATION_3D2D_2 = 4,
    POSE_ESTIMATION_3D3D_1 = 5,
    POSE_ESTIMATION_3D3D_2 = 6,
    POSE_ESTIMATION_3D3D_3 = 7
  };

/** \brief Vertex for blend: 2d-2d/2d-3d/3d-2d/3d-3d bundleadjustment(pose only)
 * Pose: {theta_x, theta_y, theta_z, T_1, T_2, T_3}
 */ 
class VertexBlendPoseOnly: public g2o::BaseVertex<6, Vector6d>
{
public:

  virtual void setToOriginImpl()
  {
    _estimate.fill(0.);
  }
  
  virtual void oplusImpl(const double* update)
  {
    Eigen::Map<const Vector6d> v(update);
    _estimate += v;
  }
  
  virtual bool read(std::istream& in)
  {
    Vector6d lv;
    for (int i=0; i<6; i++)
      in >> _estimate[i];
    return true;
  }
  virtual bool write(std::ostream& out) const
  {
    Vector6d lv=estimate();
    for (int i=0; i<6; i++){
      out << lv[i] << " ";
    }
    return out.good();
  }
  
public:
  EIGEN_MAKE_ALIGNED_OPERATOR_NEW
  
};

// Edge for blend: 2d-2d/2d-3d/3d-2d/3d-3d bundleadjustment(pose only)
class EdgeBlendPoseOnly: public g2o::BaseUnaryEdge<1, Eigen::Matrix<double,1,1>, VertexBlendPoseOnly>
{
public:
  /**/
  EdgeBlendPoseOnly(
    const BlendType& blend_type,
    const Eigen::Vector3d& Pc_1,
    const Eigen::Vector3d& Pc_2)
  : blend_type_(blend_type),
    Pc_1_(Pc_1),
    Pc_2_(Pc_2)
  {
    information().setIdentity();
  }
[... 315 lines cut ...]
 Derived = Eigen::Map<Eigen::Matrix<double, 1, 6, 1, 1, 6>, 32, Eigen::Stride<0, 0> >]") at assert.c:101
#4  0x00007ffff7b343ec in Eigen::MapBase<Eigen::Map<Eigen::Matrix<double, 1, 6, 1, 1, 6>, 32, Eigen::Stride<0, 0> >, 0>::checkSanity (this=<optimized out>)
    at /usr/include/eigen3/Eigen/src/Core/MapBase.h:168
#5  Eigen::MapBase<Eigen::Map<Eigen::Matrix<double, 1, 6, 1, 1, 6>, 32, Eigen::Stride<0, 0> >, 0>::MapBase (cols=6, rows=1, dataPtr=<optimized out>, this=<optimized out>)
    at /usr/include/eigen3/Eigen/src/Core/MapBase.h:155
#6  Eigen::MapBase<Eigen::Map<Eigen::Matrix<double, 1, 6, 1, 1, 6>, 32, Eigen::Stride<0, 0> >, 1>::MapBase (cols=6, rows=1, dataPtr=<optimized out>, this=<optimized out>)
    at /usr/include/eigen3/Eigen/src/Core/MapBase.h:243
#7  Eigen::Map<Eigen::Matrix<double, 1, 6, 1, 1, 6>, 32, Eigen::Stride<0, 0> >::Map (stride=..., cols=6, rows=1, dataPtr=<optimized out>, this=<optimized out>)
    at /usr/include/eigen3/Eigen/src/Core/Map.h:151
#8  g2o::BaseUnaryEdge<1, double, RGBD_IndoorMapping::VertexBlendPoseOnly>::linearizeOplus (this=<optimized out>, jacobianWorkspace=...)
    at /usr/local/include/g2o/core/base_unary_edge.hpp:86
#9  0x00007ffff7b187ed in g2o::BlockSolver<g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> > >::buildSystem (this=0x13321ed0) at /usr/local/include/g2o/core/block_solver.hpp:512
#10 0x00007ffff1f96be9 in g2o::OptimizationAlgorithmLevenberg::solve(int, bool) () from /usr/local/lib/libg2o_core.so
#11 0x00007ffff1f8e5d7 in g2o::SparseOptimizer::optimize(int, bool) () from /usr/local/lib/libg2o_core.so
#12 0x00007ffff7b31740 in RGBD_IndoorMapping::poseEstimationBlendG2O (state=@0x7fffffffbeac: RGBD_IndoorMapping::OK, pts_2d2d_1=std::vector of length 0, capacity 0, 
    pts_2d2d_2=std::vector of length 0, capacity 0, pts_2d3d_1=std::vector of length 4, capacity 4 = {...}, pts_2d3d_2=std::vector of length 4, capacity 4 = {...}, 
    pts_3d2d_1=std::vector of length 1, capacity 1 = {...}, pts_3d2d_2=std::vector of length 1, capacity 1 = {...}, pts_3d3d_1=std::vector of length 267, capacity 512 = {...}, 
    pts_3d3d_2=std::vector of length 267, capacity 512 = {...}, inliers_2d2d_index=std::vector of length 0, capacity 0, 
    inliers_2d3d_index=std::vector of length 4, capacity 4 = {...}, inliers_3d2d_index=std::vector of length 1, capacity 1 = {...}, 
    inliers_3d3d_index=std::vector of length 267, capacity 512 = {...}, camera=0x9f2590, distCoeffs=..., initi_T21=...)
    at RGBD_IndoorMapping/src/PoseEstimation.cpp:395
#13 0x00007ffff7af5d44 in RGBD_IndoorMapping::Mapping::trackWithFrame (this=this@entry=0x632c00, F=0x15986f40)
    at RGBD_IndoorMapping/src/Mapping.cpp:262`
4.Asking for help
I don't know why this bug happens, though i do everything i can. Anyone can help me? Thanks!

YangSiri, 2019-11-09:
Hi, Have you tried Release mode, it works for me.

balarayen-blippar, 2020-03-06:
@zouxianghong Did you find the answer for this?

wangtsiao, 2021-08-21:
check your compiler option, i think maybe we should remove -march=native

VRichardJP, 2021-12-08:
Could it be related to https://github.com/RainerKuemmerle/g2o/issues/557 ? I get the "data is not aligned" assertion failure when `-mf16c` compilation flag is used (which is included by `-march=native` on my machine). Running in Release mode is just a way to ignore the assert but does not solve the underlying problem.

zouxianghong, 2021-12-08:
谢谢，收到

Crydsch, 2023-04-04:
I encountered the same problem with g2o and eigen (and pcl).

Turns out it really is because of SSE and AVX flags. If any of those are in use, Eigen changes its behavior to align data.
This includes special allocators and overwriting new/delete/malloc/free calls.
Which in turn cause this assert error or even segfaults (double free) if data is not correctly handled.
It happens when different parts of your application (such as the app itself and/or two libraries exchange data)
and one was compiled with SSE/AVX (and thus euler) and the other was not.

This happens when the libraries and your application use different build flags.
In particular:
- In release builds g2o (and pcl) use these flags
in my case it was: `-march=native -mavx2 -msse4.2 -fpmath=sse`
=> So if you link with these libraries your application **must** be built with the same flags.
- In debug builds g2o does not use these flags
(For pcl I had to manually deactivate them even in a debug build)
=> And thus neither should your application.

tl;dr; Make sure all your libraries and your application are built with the same SSE/AVX flags.

zouxianghong, 2023-04-04:
谢谢，收到

Doflamingo-swj, 2024-12-03:
I use pcl-1.8 to solve this problem.

zouxianghong, 2024-12-03:
谢谢，收到

zouxianghong, 2025-02-08:
谢谢，收到

closed 2025-04-26 by github-actions[bot]

zouxianghong, 2025-04-26:
谢谢，收到