RainerKuemmerle/g2o #259: how to remove a single edge

anirudt, 2018-02-20:
Hi,

I have a simple question. After running ```optimize()``` once, I am finding a high number of outliers. Is there any way to delete a single edge out of the graph and run ```optimize``` again, or do I have to construct it again?

sjulier, 2018-02-20:
The method removeEdge() removes the edge from a graph and unlinks it from all of the vertices it was attached to. After you call it on your edges, I think you need to call  initializeOptimization() to reset all of g2o's internal data structures to the new graph configuration. You should then be able to call optimize().

anirudt, 2018-02-20:
Thank you!

What does setLevel() do? I've seen it being used in a similar context; I wanted to know what does setLevel(1) do, and setLevel(0) do. (this is coming from ORB SLAM code: https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/Optimizer.cc#L398

RainerKuemmerle, 2018-02-21:
You may also think about initializing the optimization based on a subset of edges instead of completely removing it.

izhengfan, 2018-02-22:
`setLevel(jnt )` is useful when you call `optimizer.initializeOptimization(int )`. If you assign  `initializeOptimization(0)`, the optimizer will include all edges up to level 0 in the optimization, and edges set to level >=1  will not be included.

anirudt, 2018-02-22:
@izhengfan Thank you so much for that explanation; it was really short and precise.

@RainerKuemmerle Thanks for the hint! :)

closed 2018-02-22 by anirudt