RainerKuemmerle/g2o #260: example simulator_3d_line get "Segmentation fault (core dumped)"

maShimizu, 2018-02-26:
Hi here

I get "Segmentation fault (core dumped)" using simulator_3d_line example code.
log:
# Using CSparse poseDim -1 landMarkDim -1 blockordering 0
Creating simulator
Creating robot
Creating line sensor
Creating landmark line 1
Creating landmark line 2
Creating landmark line 3
Segmentation fault (core dumped)

And, I try 3d slam and run original code. but sometimes get "Segmentation fault" same as above, too.
I have suspicious about "g2o::ParameterSE3Offset", because I just try simple code as below and get "segmentation fault".
Please anyone let me know where I should modify this code or check.

Code( a port of ):
#include "g2o/core/sparse_optimizer.h"
#include "g2o/core/block_solver.h"
#include "g2o/core/linear_solver.h"
#include "g2o/core/factory.h"
#include "g2o/core/optimization_algorithm_factory.h"
#include "g2o/core/optimization_algorithm_gauss_newton.h"
#include "g2o/core/optimization_algorithm_levenberg.h"
#include "g2o/solvers/csparse/linear_solver_csparse.h"
#include "g2o/solvers/eigen/linear_solver_eigen.h"
#include "g2o/types/slam3d/types_slam3d.h"

void nviewTracker::BundleAdjustment( std::vector<Frame*>* pframes )
{
	g2o::SparseOptimizer optimizer;
	optimizer.setVerbose(false);

	std::unique_ptr<g2o::BlockSolver_6_3::LinearSolverType> linearSolver;
	linearSolver = g2o::make_unique<g2o::LinearSolverCSparse<g2o::BlockSolver_6_3::PoseMatrixType>>();

	g2o::OptimizationAlgorithmLevenberg* solver = new g2o::OptimizationAlgorithmLevenberg( g2o::make_unique<g2o::BlockSolver_6_3>(std::move(linearSolver)) );
	optimizer.setAlgorithm(solver);

	g2o::ParameterSE3Offset* odomOffset = new g2o::ParameterSE3Offset;
	//odomOffset->setId(0);
	//optimizer.addParameter(odomOffset);

}

Thank you.

maShimizu, 2018-03-06:
Hi all

I've already posted above, and I try to resolve now.
I run g2o code on AWS ubuntu. AWS instance don't have GPU,
I install and use OpenGL/mesa.
I guess OpenGL/mesa have any trouble.

Does anyone try g2o on AWS?

01001HR, 2018-03-21:
I guess there were many bugs after merging "Make G2O Compatible With Hardware That Only Supports Single Precision Floating Point Types",  try use commit before that.

jumonatr, 2018-03-21:
I don't think that check-in has anything to do with it, as it just adds a using definition for changing the default number type.

Have you tried attaching a debugger to it to figure out what the callstack is?

maShimizu, 2018-03-22:
I don't change types, I use just original slam_3d example types.
Anyway, I try a debugger. But, I can't find attaching a debugger in your comment.
or is a debugger in g2o?

RainerKuemmerle, 2018-03-23:
http://senarvi.github.io/2016/07/05/stack-trace-with-gdb

RainerKuemmerle, 2018-03-23:
regarding OpenGL. There is a build config variable to exclude OpenGL.
Set G2O_USE_OPENGL to false in your cmake configuration.

closed 2025-02-08 by github-actions[bot]