RainerKuemmerle/g2o #272: Nothing displayed in g2o_viewer

Bardo91, 2018-05-04:
I am trying to visualize my graph using the g2o_viewer but what I see is just the coordinate system.
[image]
This is my file:
[g2o_graph.g2o.tar.gz](https://github.com/RainerKuemmerle/g2o/files/1976166/g2o_graph.g2o.tar.gz)

I tried the viewer with the venice dataset (http://ais.informatik.uni-freiburg.de/projects/datasets/g2o-datasets/), but It still the same.
[image]

What I do is run the viewer by callin `g2o_viewer` in the command line, and then using the drop down menu I load the files. Is there something that I am missing? or should it work just with loading the files?

Zander-yang, 2019-01-17:
Did you define the vertex and edge by yourself and load type， solver via a share library(.so)？ I loaded the .g2o file generated by example/tutorial_slam2d. The result of g2o_viewer seems like yours.

YangSiri, 2019-11-14:
Hi,
I have the same question. It says there are some vertexs and some measurements, but there is nothing displayed. I used self-defined edge and also SE3Edge. But i tried other people's *.g2o. It works fine. Do you have any new comments?

Looking forward to your help. Thanks.

RainerKuemmerle, 2021-01-05:
On custom types you need to provide the classes to draw them. See the existing types for examples on how to draw.

closed 2021-01-05 by RainerKuemmerle