RainerKuemmerle/g2o #274: undefined reference to `g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(g2o::Solver*)'

voyage19941119, 2018-05-10:
My error info:

[ 94%] Linking CXX executable ../bin/crowdsource
../lib/libcrowdsource_lane.so: **## undefined reference to `g2o::OptimizableGraph::addVertex(g2o::HyperGraph::Vertex*, g2o::OptimizableGraph::Data*)'**
../lib/libcrowdsource_lane.so: ## **undefined reference to `g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(g2o::Solver*)'**
collect2: error: ld returned 1 exit status
  
I have linked libg2o_core.so, and other references are fine after link g2o_core
Thanks for reply  @RainerKuemmerle

voyage19941119, 2018-05-10:
I have fix this using another version of g2o

closed 2018-05-10 by voyage19941119

simonprovost, 2018-11-29:
@voyage19941119  which one ?

sjulier, 2018-11-29:
Context?