RainerKuemmerle/g2o #276: Switching to multiedges is 4-5x slower

divyasharma2701, 2018-05-18:
Hi @RainerKuemmerle
 
I was using the graph to solve for pose, 3d point earlier and now i added an extra vertex for extrinsic (3-dof for rotation only) which started using multiedge instead of binary edge. This has made the system 4-5x slower than before. I also tried to fix that vertex and compare the timing it still takes 3-4x longer than before. I used batchstatics to understand how much structure was changed, the no of vertices, edges and hessian dimension is same in both before and after (fixed extrinsic vertex ) formulation. So not sure what could be the issue. can it be memory allocations related ?
I was wondering if there are some obvious optimization i can employ from g2o.
Appreciate any pointers.

Thanks
Divya

divyasharma2701, 2018-05-18:
After further digging using batch stats .. it seems like its taking 10x time on buildSystem() https://github.com/RainerKuemmerle/g2o/blob/master/g2o/core/block_solver.hpp#L463

PeterQFR, 2018-11-18:
This may not be your issue, but solvers will run orders of magnitude slower if not compiled as a release. Check to see that your build is a release build.

MinaHenein, 2019-10-15:
I'm running into the same problem even in release build mode, any recommendations?

YangSiri, 2019-10-30:
Hi, @divyasharma2701 
Could you help me with the multiedge definition? I'm a little confused while using it myself for the first time. Thanks a lot.
(any code would be great.)

MinaHenein, 2019-10-30:
Have a look here, I found it very useful when implementing my multi edge factors
https://github.com/RainerKuemmerle/g2o/issues/165

YangSiri, 2019-10-30:
yes, i checked, @MinaHenein 
can you explain to me further about your mulitedge ?