RainerKuemmerle/g2o #279: linearSolver errors, code from examples

antithing, 2018-05-27:
Hi, i am on windows, and am trying to compile some third party code that uses g2o.

The code:

```
g2o::BlockSolver_6_3::LinearSolverType* linearSolver;
linearSolver = new g2o::LinearSolverCholmod<g2o::BlockSolver_6_3::PoseMatrixType>();
```

gave me 'object of absract class type is not allowed`

so i have adjusted to:

```
 g2o::BlockSolver_6_3::LinearSolverType* linearSolver;
linearSolver = g2o::make_unique<g2o::LinearSolverCholmod<g2o::BlockSolver_6_3::PoseMatrixType>>();
```

Which seems fine.

The next lines:

```
 g2o::BlockSolver_6_3* solver_ptr = new g2o::BlockSolver_6_3(linearSolver);
    g2o::OptimizationAlgorithmLevenberg* solver = new g2o::OptimizationAlgorithmLevenberg(solver_ptr);
```

are giving me:

`no instance of the constructor matches the argument list`.

What changes can i make to get this compiling?

thanks!

antithing, 2018-05-27:
Ah, of course I found it soon after posting. I needed to add:

`#include "g2o/solvers/cholmod/linear_solver_cholmod.h"`

The errors have gone. One more thing, i am attempting to run some third party code, and have replaced:

```
 g2o::SparseOptimizer optimizer;
    optimizer.setVerbose(false);
    g2o::BlockSolver_6_3::LinearSolverType* linearSolver;
    linearSolver = new g2o::LinearSolverCholmod<g2o::BlockSolver_6_3::PoseMatrixType>();
    g2o::BlockSolver_6_3* solver_ptr = new g2o::BlockSolver_6_3(linearSolver);
    g2o::OptimizationAlgorithmLevenberg* solver = new g2o::OptimizationAlgorithmLevenberg(solver_ptr);
```

which would not compile, with:

```
	g2o::SparseOptimizer optimizer;
	optimizer.setVerbose(false);
	std::unique_ptr<g2o::BlockSolver_6_3::LinearSolverType> linearSolver;

	linearSolver = g2o::make_unique<g2o::LinearSolverCholmod<g2o::BlockSolver_6_3::PoseMatrixType>>();

	g2o::OptimizationAlgorithmLevenberg* solver = new g2o::OptimizationAlgorithmLevenberg(
		g2o::make_unique<g2o::BlockSolver_6_3>(std::move(linearSolver)));
```
As above. This compiles, but is this correct?

Thanks again!

antithing, 2018-05-29:
Thank you! i have changed to:

```
	g2o::OptimizationAlgorithmLevenberg solver(g2o::OptimizationAlgorithmLevenberg(
		g2o::make_unique<g2o::BlockSolver_6_3>(std::move(linearSolver))));
	

    solver.setUserLambdaInit(1e-10);
    optimizer.setAlgorithm(&solver);
```

and it won't compile, giving me:

`'g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(const g2o::OptimizationAlgorithmLevenberg &)': attempting to reference a deleted function`

Have i done this wrong?

jumonatr, 2018-05-29:
Yes, here you're trying to build an OptimizationAlgorithmLevenberg object using another OptimizationAlgorithmLevenberg object.

Just do:

g2o::OptimizationAlgorithmLevenberg solver(g2o::make_unique<g2o::BlockSolver_6_3>(std::move(linearSolver)));

antithing, 2018-05-29:
`g2o::OptimizationAlgorithmLevenberg solver(g2o::make_unique<g2o::BlockSolver_6_3>(std::move(linearSolver)));
`

compiles and runs great. Thank you again for your help.

closed 2018-08-07 by RainerKuemmerle