RainerKuemmerle/g2o #280: making g20, encounter make  erro

airsmailed, 2018-06-04:
My Development Environment is:
Ubuntu 14.04  64-bits
cmake g2o  is no any question,  but  make  g2o,  it appears  this erro,  

[  1%] Built target freeglut_minimal

[  2%] Built target opengl_helper

[  6%] Built target stuff

[ 17%] Built target core

[ 18%] Built target g2o_cli_library

[ 19%] Built target g2o_cli_application

[ 20%] Building CXX object g2o/types/slam3d/CMakeFiles/types_slam3d.dir/edge_se3_pointxyz.cpp.o

[ 20%] Building CXX object g2o/types/slam3d/CMakeFiles/types_slam3d.dir/edge_se3_pointxyz_disparity.cpp.o

[ 21%] Building CXX object g2o/types/slam3d/CMakeFiles/types_slam3d.dir/edge_se3_pointxyz_depth.cpp.o

[ 21%] Building CXX object g2o/types/slam3d/CMakeFiles/types_slam3d.dir/edge_se3_prior.cpp.o

[ 21%] Building CXX object g2o/types/slam3d/CMakeFiles/types_slam3d.dir/edge_se3_lotsofxyz.cpp.o

[ 22%] Building CXX object g2o/types/slam3d/CMakeFiles/types_slam3d.dir/edge_pointxyz.cpp.o

[ 22%] Building CXX object g2o/types/slam3d/CMakeFiles/types_slam3d.dir/types_slam3d.cpp.o

Linking CXX shared library ../../../../lib/libg2o_types_slam3d.so

[ 26%] Built target types_slam3d

Scanning dependencies of target g2o_hierarchical_library

[ 26%] Building CXX object g2o/apps/g2o_hierarchical/CMakeFiles/g2o_hierarchical_library.dir/edge_labeler.cpp.o

[ 27%] Building CXX object g2o/apps/g2o_hierarchical/CMakeFiles/g2o_hierarchical_library.dir/edge_creator.cpp.o

[ 27%] Building CXX object g2o/apps/g2o_hierarchical/CMakeFiles/g2o_hierarchical_library.dir/star.cpp.o

[ 28%] Building CXX object g2o/apps/g2o_hierarchical/CMakeFiles/g2o_hierarchical_library.dir/edge_types_cost_function.cpp.o

[ 28%] Building CXX object g2o/apps/g2o_hierarchical/CMakeFiles/g2o_hierarchical_library.dir/backbone_tree_action.cpp.o

[ 29%] Building CXX object g2o/apps/g2o_hierarchical/CMakeFiles/g2o_hierarchical_library.dir/simple_star_ops.cpp.o

In file included from /usr/include/eigen3/Eigen/Core:278:0,

                 from /home/xiaoqiang/slam_ku/g2o/g2o/core/eigen_types.h:30,

                 from /home/xiaoqiang/slam_ku/g2o/build/g2o/config.h:18,

                 from /home/xiaoqiang/slam_ku/g2o/g2o/core/openmp_mutex.h:30,

                 from /home/xiaoqiang/slam_ku/g2o/g2o/core/optimizable_graph.h:37,

                 from /home/xiaoqiang/slam_ku/g2o/g2o/core/sparse_optimizer.h:32,

                 from /home/xiaoqiang/slam_ku/g2o/g2o/apps/g2o_hierarchical/simple_star_ops.h:6,

                 from /home/xiaoqiang/slam_ku/g2o/g2o/apps/g2o_hierarchical/simple_star_ops.cpp:1:

/usr/include/eigen3/Eigen/src/Core/DenseBase.h: In instantiation of ‘class
 Eigen::DenseBase<Eigen::GeneralProduct<Eigen::Matrix<double, -1, -1>, Eigen::Block<const Eigen::Map<const Eigen::Matrix<double, -1,

 1>, 0, Eigen::Stride<0, 0> >, -1, 1, false>, 4> >’:

/usr/include/eigen3/Eigen/src/Core/MatrixBase.h:48:34:   required from ‘class

 Eigen::MatrixBase<Eigen::GeneralProduct<Eigen::Matrix<double, -1, -1>, Eigen::Block<const Eigen::Map<const Eigen::Matrix<double,

 -1, 1>, 0, Eigen::Stride<0, 0> >, -1, 1, false>, 4> >’

/usr/include/eigen3/Eigen/src/Core/ProductBase.h:63:7:   required from ‘class

 Eigen::ProductBase<Eigen::GeneralProduct<Eigen::Matrix<double, -1, -1>, Eigen::Block<const Eigen::Map<const Eigen::Matrix<double,

 -1, 1>, 0, Eigen::Stride<0, 0> >, -1, 1, false>, 4>, Eigen::Matrix<double, -1, -1>, Eigen::Block<const Eigen::Map<const

 Eigen::Matrix<double, -1, 1>, 0, Eigen::Stride<0, 0> >, -1, 1, false> >’

/usr/include/eigen3/Eigen/src/Core/GeneralProduct.h:327:7:   required from ‘class Eigen::GeneralProduct<Eigen::Matrix<double, -1,

 -1>, Eigen::Block<const Eigen::Map<const Eigen::Matrix<double, -1, 1>, 0, Eigen::Stride<0, 0> >, -1, 1, false>, 4>’

/home/xiaoqiang/slam_ku/g2o/g2o/core/matrix_operations.h:50:64:   required from here

/usr/include/eigen3/Eigen/src/Core/DenseBase.h:424:21: internal compiler error: Segmentation fault

     CoeffReturnType value() const

                     ^
Please submit a full bug report,

with preprocessed source if appropriate.

See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.

The bug is not reproducible, so it is likely a hardware or OS problem.

make[2]: *** [g2o/apps/g2o_hierarchical/CMakeFiles/g2o_hierarchical_library.dir/simple_star_ops.cpp.o] Error 1

make[1]: *** [g2o/apps/g2o_hierarchical/CMakeFiles/g2o_hierarchical_library.dir/all] Error 2

make: *** [all] Error 2

RainerKuemmerle, 2018-06-25:
Please try with Ubuntu 16.04 or newer.

airsmailed, 2018-06-26:
Thank you very much for your answer, but I can't upgrade the Ubuntu version now. Is there any other solution?@RainerKuemmerle

sjulier, 2018-06-26:
Since it's an internal compiler error, I've found that sometimes doing "transparent" things (like reordering includes, or using typedefs) causes the problem to go away. Another thing is to disable different optimisation settings to see if that makes any difference. That would impact performance, of course.

Another thing you could do is see if you can build and manually install a more recent version of your compiler.

closed 2018-08-07 by RainerKuemmerle