RainerKuemmerle/g2o #282: Segmentation Fault g2o::VertexSE3::VertexSE3()    from /usr/local/lib/libg2o_types_slam3d.so

manhha1402, 2018-06-16:
Hi,
I install g2o branch 20170730_git and test an example with g2o but the program couldn't run. I use gdb to debug and error from file "g2o::VertexSE3::VertexSE3()    from /usr/local/lib/libg2o_types_slam3d.so"
I am using ubuntu 16.04 eigen 3.4
How can I solve this problem ?
Thanks !

sjulier, 2018-06-20:
Which example did you try? Which operating system?

manhha1402, 2018-06-22:
Hi @sjulier,
I use Ubuntu 16.04 eigen 3.4. Here is example that I tried 
https://github.com/yezhenzhao/RGBD_SLAM/blob/master/g2o_test.cpp
I debug with gdb and got :
Program received signal SIGSEGV, Segmentation fault.
 0x00007ffff7715184 in g2o::VertexSE3::VertexSE3() () from /usr/local/lib/libg2o_types_slam3d.so

sjulier, 2018-06-22:
Thanks. I currently don't have my computer available, but could you build in debug and get a full stack trace?

apozharski, 2018-07-09:
I currently have the same problem (a segfault in the VertexSE3 constructor) though not in an example but in a slam algorithm using g2o. The full trace from valgrind is  is: 
> ==12909== Process terminating with default action of signal 11 (SIGSEGV)
==12909==  General Protection Fault
==12909==    at 0x7469E0A: g2o::VertexSE3::VertexSE3() (in /usr/local/lib/libg2o_types_slam3d.so)
==12909==    by 0x4F1F3BE: dvo_slam::internal::LocalMapImpl::addFrameVertex(ros::Time const&) (in /home/anton/dvo-ws/devel/lib/libdvo_slam.so)
==12909==    by 0x4F1FFB0: dvo_slam::internal::LocalMapImpl::LocalMapImpl(boost::shared_ptr<dvo::core::RgbdImagePyramid> const&, Eigen::Transform<double, 3, 2, 0> const&) (in /home/anton/dvo-ws/devel/lib/libdvo_slam.so)
==12909==    by 0x4F1EF8B: dvo_slam::LocalMap::LocalMap(boost::shared_ptr<dvo::core::RgbdImagePyramid> const&, Eigen::Transform<double, 3, 2, 0> const&) (in /home/anton/dvo-ws/devel/lib/libdvo_slam.so)
==12909==    by 0x4F1EFDD: dvo_slam::LocalMap::create(boost::shared_ptr<dvo::core::RgbdImagePyramid> const&, Eigen::Transform<double, 3, 2, 0> const&) (in /home/anton/dvo-ws/devel/lib/libdvo_slam.so)
==12909==    by 0x4F210AF: dvo_slam::LocalTracker::initNewLocalMap(boost::shared_ptr<dvo::core::RgbdImagePyramid> const&, boost::shared_ptr<dvo::core::RgbdImagePyramid> const&, dvo::DenseTracker::Result&, Eigen::Transform<double, 3, 2, 0> const&) (in /home/anton/dvo-ws/devel/lib/libdvo_slam.so)
==12909==    by 0x4F2250C: dvo_slam::LocalTracker::update(boost::shared_ptr<dvo::core::RgbdImagePyramid> const&, Eigen::Transform<double, 3, 2, 0>&) (in /home/anton/dvo-ws/devel/lib/libdvo_slam.so)
==12909==    by 0x4EBE8A7: dvo_slam::KeyframeTracker::Impl::update(boost::shared_ptr<dvo::core::RgbdImagePyramid> const&, ros::Time const&, Eigen::Transform<double, 3, 2, 0>&) (in /home/anton/dvo-ws/devel/lib/libdvo_slam.so)
==12909==    by 0x4EEF6E7: dvo_slam::CameraKeyframeTracker::handleImages(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&, boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&, boost::shared_ptr<sensor_msgs::CameraInfo_<std::allocator<void> > const> const&, boost::shared_ptr<sensor_msgs::CameraInfo_<std::allocator<void> > const> const&) (in /home/anton/dvo-ws/devel/lib/libdvo_slam.so)
==12909==    by 0x7C46015: boost::detail::function::void_function_obj_invoker9<boost::function<void (boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&, boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&, boost::shared_ptr<sensor_msgs::CameraInfo_<std::allocator<void> > const> const&, boost::shared_ptr<sensor_msgs::CameraInfo_<std::allocator<void> > const> const&, boost::shared_ptr<message_filters::NullType const> const&, boost::shared_ptr<message_filters::NullType const> const&, boost::shared_ptr<message_filters::NullType const> const&, boost::shared_ptr<message_filters::NullType const> const&, boost::shared_ptr<message_filters::NullType const> const&)>, void, boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const>, boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const>, boost::shared_ptr<sensor_msgs::CameraInfo_<std::allocator<void> > const>, boost::shared_ptr<sensor_msgs::CameraInfo_<std::allocator<void> > const>, boost::shared_ptr<message_filters::NullType const>, boost::shared_ptr<message_filters::NullType const>, boost::shared_ptr<message_filters::NullType const>, boost::shared_ptr<message_filters::NullType const>, boost::shared_ptr<message_filters::NullType const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const>, boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const>, boost::shared_ptr<sensor_msgs::CameraInfo_<std::allocator<void> > const>, boost::shared_ptr<sensor_msgs::CameraInfo_<std::allocator<void> > const>, boost::shared_ptr<message_filters::NullType const>, boost::shared_ptr<message_filters::NullType const>, boost::shared_ptr<message_filters::NullType const>, boost::shared_ptr<message_filters::NullType const>, boost::shared_ptr<message_filters::NullType const>) (in /home/anton/dvo-ws/devel/lib/libdvo_ros.so)
==12909==    by 0x7C5B24B: boost::function9<void, boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const>, boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const>, boost::shared_ptr<sensor_msgs::CameraInfo_<std::allocator<void> > const>, boost::shared_ptr<sensor_msgs::CameraInfo_<std::allocator<void> > const>, boost::shared_ptr<message_filters::NullType const>, boost::shared_ptr<message_filters::NullType const>, boost::shared_ptr<message_filters::NullType const>, boost::shared_ptr<message_filters::NullType const>, boost::shared_ptr<message_filters::NullType const> >::operator()(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const>, boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const>, boost::shared_ptr<sensor_msgs::CameraInfo_<std::allocator<void> > const>, boost::shared_ptr<sensor_msgs::CameraInfo_<std::allocator<void> > const>, boost::shared_ptr<message_filters::NullType const>, boost::shared_ptr<message_filters::NullType const>, boost::shared_ptr<message_filters::NullType const>, boost::shared_ptr<message_filters::NullType const>, boost::shared_ptr<message_filters::NullType const>) const (in /home/anton/dvo-ws/devel/lib/libdvo_ros.so)
[... cut ...] locator<void> > const> const&, ros::MessageEvent<sensor_msgs::CameraInfo_<std::allocator<void> > const> const&, ros::MessageEvent<message_filters::NullType const> const&, ros::MessageEvent<message_filters::NullType const> const&, ros::MessageEvent<message_filters::NullType const> const&, ros::MessageEvent<message_filters::NullType const> const&, ros::MessageEvent<message_filters::NullType const> const&) (in /home/anton/dvo-ws/devel/lib/libdvo_ros.so)

This is with g2o compiled with Eigen 3.3.90.

apozharski, 2018-07-10:
As an update I cannot reproduce this issue when I used g2o built in debug mode, which means I can't provide a trace beyond the constructor. I honestly find that a bit odd.

YoshuaNava, 2018-07-15:
Hi,
I experienced the same issue, and ended up rolling back to commit `a48ff8c42136f18fbe215b02bfeca48fa0c67507`from jlblancoc, which works well.

lambdald, 2018-08-07:
same problem. A older version works well.

MMYY, 2018-08-13:
same problem.
vesion:[8564e1e365d4e719dceca6269a0ba203f7f43dec](https://github.com/RainerKuemmerle/g2o/commit/8564e1e365d4e719dceca6269a0ba203f7f43dec)
`Segmentation fault (core dumped)`

call stack:
#0 g2o::VertexSE3::VertexSE3()() at /usr/local/lib/libg2o_types_slam3d.so
#1 main() at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:239

closed 2025-02-07 by github-actions[bot]