RainerKuemmerle/g2o #288: Does anyone know how to add GPS coordinates in optimization of monocular ORBSLAM?

nikhilarane, 2018-08-01:


nikhilarane, 2018-08-01:
I am thinking an error function of (X - Xgps)^2 + reprojection error. What will be the structure for this?

RainerKuemmerle, 2018-08-07:
How about modelling GPS by an EdgeXYZPrior

shanmukhananda, 2018-08-09:
You can try below approach.
- create a new custom unary edge  which implements `BaseUnaryEdge<3, Eigen::Vector3d, VertexSE3Expmap>`. This is a prior edge
- you can also constrain the camera nodes based on GPS distance for this you have to create custom binary edge which implements `BaseBinaryEdge<1, double, VertexSE3Expmap, VertexSE3Expmap>`.  Error will be `distance(camera_pos) - distance(gps_pos)`.
- add the prior edge (and/or binary distance edge) in global bundle adjustment and local bundle adjustment.
- adding in global bundle adjustment fixes initial scale issue(in monocular case). adding these edge in local bundle adjustment improves the trajectory and fixes the scale drift. 
- setting correct information matrix is crucial. 
- to use prior edge you will have to convert gps coordinate to local.

nikhilarane, 2018-08-29:
Thanks, Rainer and Shanmukhananda.

"create a new custom unary edge which implements BaseUnaryEdge<3, Eigen::Vector3d, VertexSE3Expmap>. This is a prior edge"

What will be the Jacobian for this edge

sume-cn, 2019-04-02:
> Thanks, Rainer and Shanmukhananda.
> 
> "create a new custom unary edge which implements BaseUnaryEdge<3, Eigen::Vector3d, VertexSE3Expmap>. This is a prior edge"
> 
> What will be the Jacobian for this edge

You can refer to gtsam's sample: LocalizationExample.cpp

sume-cn, 2019-08-09:
> You can try below approach.
>
> * create a new custom unary edge  which implements `BaseUnaryEdge<3, Eigen::Vector3d, VertexSE3Expmap>`. This is a prior edge
> * you can also constrain the camera nodes [...]

I have the same question, if the error is like:
`class  EdgeXYZPriorSE3Expmap : public  BaseUnaryEdge<3, Vector3d, VertexSE3Expmap>
{
    void computeError()
    {
        const VertexSE3Expmap* v = static_cast<const VertexSE3Expmap*>(_vertices[0]);
        Vector3d obs(_measurement);
        _error = obs - SE3Quat(v->estimate()).translation();
    }

    virtual void linearizeOplus();
};`
how to compute the Jacobian Matrix?
thanks a lot!

sume-cn, 2019-08-26:
Solved, just leave linearizeOplus() not implemented.
BaseUnaryEdge<3, Eigen::Vector3d, VertexSE3Expmap> works, but some tricks must be used to get a fine trajectory.

radeonwu, 2019-10-15:
> Solved, just leave linearizeOplus() not implemented.
> BaseUnaryEdge<3, Eigen::Vector3d, VertexSE3Expmap> works, but some tricks must be used to get a fine trajectory.

Hi, 
I followed shanmukhananda 's suggestion to add GPS absolute observation of XYZ, into ORB-SLAM2's mono operation pipeline. Below is what I did 
- defined a custom edge type EdgeSE3ExpmapXYZPrior() : BaseUnaryEdge<3, Vector3, g2o::VertexSE3Expmap>(), leaving linearizeOplus unimplemented
- added the above edges in both LocalBundleAdjustment and GlobalBundleAdjustment, and information is set to identity()

As a result, the modified ORB-SLAM2 program can run to finish map building, but the result is more or less the same as before adding the GPS constraints, especially for the map scale which is far less than the ground truth.

Any comments or suggestions to improve will be welcome.

Wu

whu-lyh, 2021-07-29:
@radeonwu have u figure it out or not?

closed 2025-02-06 by github-actions[bot]