RainerKuemmerle/g2o #290: G2O BlockSolver Initialization Crash on Unix System

simonprovost, 2018-08-03:
I have a project which possesses and uses G2O library, it runs on both platforms(win/Unix).

***(WINDOW PASSED / UNIX CRASHED)***

We can see in both platforms, these lines:

    g2o::SparseOptimizer optimizer;
	g2o::BlockSolver_6_3::LinearSolverType * linearSolver;

	linearSolver = new g2o::LinearSolverDense<g2o::BlockSolver_6_3::PoseMatrixType>();

Next steps, in window os we have this:

    g2o::BlockSolver_6_3 * solver_ptr = new g2o::BlockSolver_6_3(linearSolver);
	g2o::OptimizationAlgorithmLevenberg* solver = new g2o::OptimizationAlgorithmLevenberg(solver_ptr);

But Unix system can't compile those lines because it says

> ***my_file_G2o.cpp:*** *In member function 'int *Refiner_G2O::refinePose(cv::Mat&, const std::vector<cv::Point3_<float> >&, const std::vector<cv::KeyPoint>&, const cv::Mat&, float, std::vector<int>&)':* **my_file_G2o.cpp** **--> no matching function for call to** *'g2o::BlockSolver<g2o::BlockSolverTraits<6, 3>::BlockSolver(g2o::BlockSolver<g2o::BlockSolverTraits<6, 3>::LinearSolverType*&)'*
^ 
In file included from *G2o/include/g2o/core/block_solver.h:199:0*,
G2o/include/g2o/core/***block_solver.hpp:40:1***: **note: candidate:** g2o::BlockSolver<Traits>::BlockSolver(std::unique_ptr<typename Traits::LinearSolverType>) [with Traits = g2o::BlockSolverTraits<6, 3>; typename Traits::LinearSolverType = g2o::LinearSolver<Eigen::Matrix<double, 6, 6, 0> >]
 BlockSolver<Traits>::BlockSolver(std::unique_ptr<LinearSolverType> linearSolver)

When I see these errors, I complete my Unix code with this new block

    auto solver_ptr = g2o::make_unique<g2o::BlockSolver_6_3>(linearSolver); // [SEGFAULT LINE]
	auto solver = g2o::make_unique<g2o::OptimizationAlgorithmLevenberg>(solver_ptr);
    optimizer.setAlgorithm(solver.get());

So now, I can build/run but I meet a segfault on the line with **[SEGFAULT LINE]** tag.

**EDIT: After step to step debug with debug build I have a few new pieces of information:**

    g2o::SparseOptimizer optimizer; // debug says: "optimize: g2o::SparseOptimize
	g2o::BlockSolver_6_3::LinearSolverType * linearSolver = NULL; // debug says : linearSolver: NULL
	linearSolver = new g2o::LinearSolverDense<g2o::BlockSolver_6_3::PoseMatrixType>(); // debug says:  linear solver: 0x7fe39cba9140

	if (linearSolver == NULL) // debug says : not null.
	    exit (84);
	auto solver_ptr = g2o::make_unique<g2o::BlockSolver< g2o::BlockSolver_6_3 >>(linearSolver);
// DEBUG --> And here, with "step into" he go to this "/usr/include/c++/6/bits/move.h"
and in the ligne 48 we can see this function:

    `template<typename T, typename ...ArgTs>
     std::unique_ptr<T> make_unique(ArgTs&& ...args) //DEBUG says: args#0: 
     -var-create: unable to create variable object.
      {
        return std::unique_ptr<T>(g2o::make_unique<T>(std::forward<ArgTs> 
        (args)...));
       };`

And he segfaults because she has a bad args.

I don't understand, why? if someone has an idea I want to understand how in Window it works and not in Unix system, will feel wonderful if you help.

LINUX Version: Ubuntu 16.04<br/>
CMAKE Version: 3.11.4

Best regards,

simonprovost, 2018-08-03:
SOLVED:

G2O optimise:

    g2o::SparseOptimizer optimizer;
	
    auto linearSolver = std::make_unique<g2o::LinearSolverDense<g2o::BlockSolver_6_3::PoseMatrixType>>();

    auto solver = new g2o::OptimizationAlgorithmLevenberg(std::make_unique<g2o::BlockSolver_6_3>(std::move(linearSolver)));

    optimizer.setAlgorithm(solver);

MISC.H:

    template<typename T, typename ...ArgTs>
    std::unique_ptr<T> make_unique(ArgTs&& ...args)
    {
      return std::unique_ptr<T>(new T(std::forward<ArgTs>(args)...));
    };

jumonatr, 2018-08-03:
So the reason you were seeing issues was because you were mixing and matching the old raw new/delete way with the newer safer smart pointer way. All the samples have been updated with the new patterns if you want to crib off of those.

The reason the argument wasn't correct when calling g2o::make_unique was that you were trying to build a BlockSolver with a raw pointer, but the constructor takes a unique_ptr because it assumes ownership of the solver.

In case you were wondering, the g2o::make_unique function in misc.h is defined because g2o supports C++11, but the standard std::make_unique function is only defined in compilers supporting C++14 and up.

In your solved case you may want to adjust it in the following way:

```
auto linearSolver = std::make_unique<g2o::LinearSolverDense<g2o::BlockSolver_6_3::PoseMatrixType>>();

g2o::OptimizationAlgorithmLevenberg solver{std::make_unique<g2o::BlockSolver_6_3>(std::move(linearSolver))};

g2o::SparseOptimizer optimizer;
optimizer.setAlgorithm(&solver);
```

The main change is to no longer allocate your solver using the operator new, because setAlgorithm doesn't assume ownership of it, which means you'd be leaking it every time you call your function.

Another more minor change is to flip the order of the algorithm and the optimizer construction, this ensures that your optimizer will get destroyed before your algorithm. In this case it doesn't really matter because the SparseOptimizer destructor doesn't use the algorithm pointer, but since the SparseOptimizer has a non-owning pointer to the algorithm, we should try to never have a dangling pointer to a deleted object.

closed 2018-08-07 by RainerKuemmerle