RainerKuemmerle/g2o #300: Seg fault and infinite loop in computeEntry() while computing marginal covariance

Willyzw, 2018-08-29:
Hallo, I am using the g2o for 3d pose graph optimization under the ros kinetic environment. For some usecases, we need the information about the uncertainty of 3d pose. So I tried to compute the marginal covariance. However sometimes the segmentation fault and infinite loop happens. 

Following is the gdb backtrace for the case of infinite loop:

Program received signal SIGINT, Interrupt.
0x00007ffff7136df6 in ?? () from /opt/ros/kinetic/lib/libg2o_core.so
(gdb) bt
#0  0x00007ffff7136df6 in ?? () from /opt/ros/kinetic/lib/libg2o_core.so
#1  0x00007ffff7137229 in g2o::MarginalCovarianceCholesky::computeEntry(int, int) () from /opt/ros/kinetic/lib/libg2o_core.so
#2  0x00007ffff713718a in g2o::MarginalCovarianceCholesky::computeEntry(int, int) () from /opt/ros/kinetic/lib/libg2o_core.so
#3  0x00007ffff713718a in g2o::MarginalCovarianceCholesky::computeEntry(int, int) () from /opt/ros/kinetic/lib/libg2o_core.so
#4  0x00007ffff713718a in g2o::MarginalCovarianceCholesky::computeEntry(int, int) () from /opt/ros/kinetic/lib/libg2o_core.so
#5  0x00007ffff713718a in g2o::MarginalCovarianceCholesky::computeEntry(int, int) () from /opt/ros/kinetic/lib/libg2o_core.so
#6  0x00007ffff713718a in g2o::MarginalCovarianceCholesky::computeEntry(int, int) () from /opt/ros/kinetic/lib/libg2o_core.so
#7  0x00007ffff713718a in g2o::MarginalCovarianceCholesky::computeEntry(int, int) () from /opt/ros/kinetic/lib/libg2o_core.so
#8  0x00007ffff713718a in g2o::MarginalCovarianceCholesky::computeEntry(int, int) () from /opt/ros/kinetic/lib/libg2o_core.so
#9  0x00007ffff713718a in g2o::MarginalCovarianceCholesky::computeEntry(int, int) () from /opt/ros/kinetic/lib/libg2o_core.so
#10 0x00007ffff713718a in g2o::MarginalCovarianceCholesky::computeEntry(int, int) () from /opt/ros/kinetic/lib/libg2o_core.so
#11 0x00007ffff713718a in g2o::MarginalCovarianceCholesky::computeEntry(int, int) () from /opt/ros/kinetic/lib/libg2o_core.so
#12 0x00007ffff713718a in g2o::MarginalCovarianceCholesky::computeEntry(int, int) () from /opt/ros/kinetic/lib/libg2o_core.so
#13 0x00007ffff713718a in g2o::MarginalCovarianceCholesky::computeEntry(int, int) () from /opt/ros/kinetic/lib/libg2o_core.so
#14 0x00007ffff713718a in g2o::MarginalCovarianceCholesky::computeEntry(int, int) () from /opt/ros/kinetic/lib/libg2o_core.so
#15 0x00007ffff713718a in g2o::MarginalCovarianceCholesky::computeEntry(int, int) () from /opt/ros/kinetic/lib/libg2o_core.so
#16 0x00007ffff713718a in g2o::MarginalCovarianceCholesky::computeEntry(int, int) () from /opt/ros/kinetic/lib/libg2o_core.so
#17 0x00007ffff713718a in g2o::MarginalCovarianceCholesky::computeEntry(int, int) () from /opt/ros/kinetic/lib/libg2o_core.so

RainerKuemmerle, 2018-08-29:
Can you provide a g2o file (obtained by calling save on your graph)?

Willyzw, 2018-08-29:
Sorry I can not provide the data due to the project restriction. But I will try if I can re-produce the issue from the simulated data.

Maybe I should provide more information regarding the issue. So initially we did not encounter the issue by querying the marginal covariance of the vertex pose. However when we did one step more to query the cross correlation among the poses.More specifically, the off-diagonal block of inverted Hessian matrix, then the segmentation fault and infinite loop happen occasionally.

Willyzw, 2018-08-30:
After trying some simulated data as well as the real datasets found online, I haven't got same issue. So firstly I need to clarify that the issue about the infinite loop seems a false alarm, as for big graph one may need more patience with regard to the complexity O(n^3). One more remark is that the computing of cross correlation (off-diagonal blocks) seems be sped up a lot by adding edges with zero information matrix to isolated sub-graphs. I guess it might relate to the underlying sparse structure in linear solver (I am really newbie in the field linear system, so very likely I am totally wrong).

Regarding the segmentation fault in function computeEntry, it happened to the graph with about 200 000 poses and 1000 000 edges. I am not sure why it happened. Considering it occurred only for this graph, it might be the fault of graph itself. Could you share with me if such error has ever happened to you? Or under which circumstance it could happen?

P. S. while playing the graph, one strange error happened to me and it took me quite much time to  debug. Finally it turns out that the binary edge which connects two same vertices was added accidentally. The memory leak can occur when such edge is removed due to the assert at line 219 in hyper_graph.cpp. I can't tell if it is a bug, but can we maybe change the assert to e.g. an exception or maybe prevent the user from creating such edge

RakshaMu, 2018-09-03:
What would the computational complexity for a graph with 10000 nodes be?

RakshaMu, 2018-09-04:
Thank you!

Currently, I'm working with M10000 dataset which has 10.000 nodes in it. The dataset is in ".graph" or Toro format. How do I generate an optimized graph using g2o? Or is there any dataset (of 10.000 nodes) available in .g2o format?

Thanks!

Willyzw, 2018-09-05:
@RainerKuemmerle Thank you sincerely for the reply. The details described in paper help a lot in understanding the code! 
@RakshaMu During the survey I have found this website helpful, which provides some datasets in g2o format. https://lucacarlone.mit.edu/datasets/

RakshaMu, 2018-09-06:
@Willyzw  Thanks! I have gone through those data sets. There is no data set which has >10000 nodes.

rjasd1128hf, 2019-09-06:
that is because stack overflow, you may adjust your reserved stack size larger!

Willyzw, 2020-07-06:
> 
> 
> that is because stack overflow, you may adjust your reserved stack size larger!

Willyzw, 2020-07-06:
@rjasd1128hf I can confirm the stack overflow issue. One can try to increase the stack size for large dataset or rewrite the `computeEntry` in a buttom-up manner to avoid recursive calls.

closed 2020-07-06 by Willyzw

uestcyile, 2024-11-06:
@Willyzw  Do you fix this issue by increase the stack size ? for example, by setting "ulimit -s 16384" ?