RainerKuemmerle/g2o #305: BA_demo not working if just fixing one frame

Bardo91, 2018-09-23:
I am trying to run BA in my project but I experienced some issues that I am not sure to undersand. So at first I copied the ba_demo and added a piece of code for plotting the result.

The following picture shows the result of the example (as it is in the repo) with first two vertices fixed.
[image]

And this picture shows the result of the example if only fixing the first vertex (https://github.com/RainerKuemmerle/g2o/blob/c2f4eb8271d19476fa7934c485d2508f3277fbd3/g2o/examples/ba/ba_demo.cpp#L174) 
[image]

Does anyone experienced something similar? My concern is that I cannot set two first frames fix because if the second one might have noise. If I fix the second frame the noise propagates.

RainerKuemmerle, 2018-10-03:
You need to provide the scale to the optimizer which is done by fixing two frames.

Bardo91, 2018-10-17:
That makes a lot of sense. But, then, what confuses me is how is it used in the ORB_SLAM2 library (https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/Optimizer.cc#L79) it seems to me that it is only fixing the first frame, so, how is possible that it works? Is it because the algorithm is close to the minimum?

closed 2020-10-24 by RainerKuemmerle