RainerKuemmerle/g2o #307: g2o_viewer(core dumped)

Junkai-Liu, 2018-09-25:
when i run g2o_viewer sphere.g2o the output is :
* Error in `g2o_viewer': realloc(): invalid pointer: 0x00007ff6307a0820 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7ff6310067e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x85d80)[0x7ff631014d80]
/lib/x86_64-linux-gnu/libc.so.6(realloc+0x22f)[0x7ff6310138ef]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN9QListData7reallocEi+0x1f)[0x7ff62c6c89cf]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN9QListData6appendEi+0x81)[0x7ff62c6c8aa1]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x1d6d78)[0x7ff62c794d78]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_Z21qRegisterResourceDataiPKhS0_S0_+0x2e6)[0x7ff62c790b16]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x7bcc3)[0x7ff62c639cc3]
/lib64/ld-linux-x86-64.so.2(+0x106ba)[0x7ff632e9f6ba]
/lib64/ld-linux-x86-64.so.2(+0x107cb)[0x7ff632e9f7cb]
/lib64/ld-linux-x86-64.so.2(+0xc6a)[0x7ff632e8fc6a]
======= Memory map: ========
00400000-00402000 r-xp 00000000 08:14 3563307                            /usr/local/bin/g2o_viewer
00601000-00602000 r--p 00001000 08:14 3563307                            /usr/local/bin/g2o_viewer
00602000-00603000 rw-p 00002000 08:14 3563307                            /usr/local/bin/g2o_viewer
02590000-025c2000 rw-p 00000000 00:00 0                                  [heap]
7ff620000000-7ff620021000 rw-p 00000000 00:00 0 
7ff620021000-7ff624000000 ---p 00000000 00:00 0 
7ff62726a000-7ff628b20000 r-xp 00000000 08:14 3558617                    /usr/lib/x86_64-linux-gnu/libicudata.so.55.1
7ff628b20000-7ff628d1f000 ---p 018b6000 08:14 3558617                    /usr/lib/x86_64-linux-gnu/libicudata.so.55.1
7ff628d1f000-7ff628d20000 r--p 018b5000 08:14 3558617                    /usr/lib/x86_64-linux-gnu/libicudata.so.55.1
7ff628d20000-7ff628d21000 rw-p 018b6000 08:14 3558617                    /usr/lib/x86_64-linux-gnu/libicudata.so.55.1
7ff628d21000-7ff628d45000 r-xp 00000000 08:14 3544653                    /usr/lib/x86_64-linux-gnu/libgraphite2.so.3.0.1
7ff628d45000-7ff628f44000 ---p 00024000 08:14 3544653                    /usr/lib/x86_64-linux-gnu/libgraphite2.so.3.0.1
7ff628f44000-7ff628f46000 r--p 00023000 08:14 3544653                    /usr/lib/x86_64-linux-gnu/libgraphite2.so.3.0.1
7ff628f46000-7ff628f47000 rw-p 00025000 08:14 3544653                    /usr/lib/x86_64-linux-gnu/libgraphite2.so.3.0.1
7ff628f47000-7ff628f4c000 r-xp 00000000 08:14 3548172                    /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7ff628f4c000-7ff62914b000 ---p 00005000 08:14 3548172                    /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7ff62914b000-7ff62914c000 r--p 00004000 08:14 3548172                    /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7ff62914c000-7ff62914d000 rw-p 00005000 08:14 3548172                    /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7ff62914d000-7ff6291b1000 r-xp 00000000 08:14 3549138                    /usr/lib/x86_64-linux-gnu/libpcre16.so.3.13.2
7ff6291b1000-7ff6293b1000 ---p 00064000 08:14 3549138                    /usr/lib/x86_64-linux-gnu/libpcre16.so.3.13.2
7ff6293b1000-7ff6293b2000 r--p 00064000 08:14 3549138                    /usr/lib/x86_64-linux-gnu/libpcre16.so.3.13.2
7ff6293b2000-7ff6293b3000 rw-p 00065000 08:14 3549138                    /usr/lib/x86_64-linux-gnu/libpcre16.so.3.13.2
7ff6293b3000-7ff629532000 r-xp 00000000 08:14 3558621                    /usr/lib/x86_64-linux-gnu/libicuuc.so.55.1
7ff629532000-7ff629732000 ---p 0017f000 08:14 3558621                    /usr/lib/x86_64-linux-gnu/libicuuc.so.55.1
7ff629732000-7ff629742000 r--p 0017f000 08:14 3558621                    /usr/lib/x86_64-linux-gnu/libicuuc.so.55.1
7ff629742000-7ff629743000 rw-p 0018f000 08:14 3558621                    /usr/lib/x86_64-linux-gnu/libicuuc.so.55.1
7ff629743000-7ff629747000 rw-p 00000000 00:00 0 
7ff629747000-7ff629999000 r-xp 00000000 08:14 3558618                    /usr/lib/x86_64-linux-gnu/libicui18n.so.55.1
7ff629999000-7ff629b99000 ---p 00252000 08:14 3558618                    /usr/lib/x86_64-linux-gnu/libicui18n.so.55.1
7ff629b99000-7ff629ba8000 r--p 00252000 08:14 3558618                    /usr/lib/x86_64-linux-gnu/libicui18n.so.55.1
7ff629ba8000-7ff629ba9000 rw-p 00261000 08:14 3558618                    /usr/lib/x86_64-linux-gnu/libicui18n.so.55.1
7ff629ba9000-7ff629c05000 r-xp 00000000 08:14 3548814                    /usr/lib/x86_64-linux-gnu/libharfbuzz.so.0.10000.1
7ff629c05000-7ff629e05000 ---p 0005c000 08:14 3548814                    /usr/lib/x86_64-linux-gnu/libharfbuzz.so.0.10000.1
7ff629e05000-7ff629e06000 r--p 0005c000 08:14 3548814                    /usr/lib/x86_64-linux-gnu/libharfbuzz.so.0.10000.1
7ff629e06000-7ff629e07000 rw-p 0005d000 08:14 3548814                    /usr/lib/x86_64-linux-gnu/libharfbuzz.so.0.10000.1
7ff629e07000-7ff629e0b000 r-xp 00000000 08:14 524317                     /lib/x86_64-linux-gnu/libuuid.so.1.3.0
7ff629e0b000-7ff62a00a000 ---p 00004000 08:14 524317                     /lib/x86_64-linux-gnu/libuuid.so.1.3.0
7ff62a00a000-7ff62a00b000 r--p 00003000 08:14 524317                     /lib/x86_64-linux-gnu/libuuid.so.1.3.0
7ff62a00b000-7ff62a00c000 rw-p 00004000 08:14 524317                     /lib/x86_64-linux-gnu/libuuid.so.1.3.0
7ff62a00c000-7ff62a013000 r-xp 00000000 08:14 3548551                    /usr/lib/x86_64-linux-gnu/libffi.so.6.0.4
7ff62a013000-7ff62a212000 ---p 00007000 08:14 3548551                    /usr/lib/x86_64-linux-gnu/libffi.so.6.0.4
7ff62a212000-7ff62a213000 r--p 00006000 08:14 3548551                    /usr/lib/x86_64-linux-gnu/libffi.so.6.0.4
7ff62a213000-7ff62a214000 rw-p 00007000 08:14 3548551                    /usr/lib/x86_64-linux-gnu/libffi.so.6.0.4
7ff62a214000-7ff62a282000 r-xp 00000000 08:14 529237                     /lib/x86_64-linux-gnu/libpcre.so.3.13.2
7ff62a282000-7ff62a482000 ---p 0006e000 08:14 529237                     /lib/x86_64-linux-gnu/libpcre.so.3.13.2
7ff62a482000-7ff62a483000 r--p 0006e000 08:14 529237                     /lib/x86_64-linux-gnu/libpcre.so.3.13.2
7ff62a483000-7ff62a484000 rw-p 0006f000 08:14 529237                     /lib/x86_64-linux-gnu/libpcre.so.3.13.2
7ff62a484000-7ff62a486000 r-xp 00000000 08:14 3548161                    /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7ff62a486000-7ff62a686000 ---p 00002000 08:14 3548161                    /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7ff62a686000-7ff62a687000 r--p 00002000 08:14 3548161                    /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7ff62a687000-7ff62a688000 rw-p 00003000 08:14 3548161                    /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7ff62a688000-7ff62a6ea000 r-xp 00000000 08:14 3548200                    /usr/lib/x86_64-linux-gnu/libXt.so.6.0.0
7ff62a6ea000-7ff62a8ea000 ---p 00062000 08:14 3548200                    /usr/lib/x86_64-linux-gnu/libXt.so.6.0.0
7ff62a8ea000-7ff62a8eb000 r--p 00062000 08:14 3548200                    /usr/lib/x86_64-linux-gnu/libXt.so.6.0.0
7ff62a8eb000-7ff62a8f0000 rw-p 00063000 08:14 3548200                    /usr/lib/x86_64-linux-gnu/libXt.so.6.0.0
[... 127 lines cut ...]
7ff62f6dd000-7ff62f6de000 r--p 00017000 08:14 530571                     /lib/x86_64-linux-gnu/libpthread-2.23.so
7ff62f6de000-7ff62f6df000 rw-p 00018000 08:14 530571                     /lib/x86_64-linux-gnu/libpthread-2.23.so
7ff62f6df000-7ff62f6e3000 rw-p 00000000 00:00 0 
7ff62f6e3000-7ff62f720000 r-xp 00000000 08:14 3548565                    /usr/lib/x86_64-linux-gnu/libfontconfig.so.1.9.0
7ff62f720000-7ff62f91f000 ---p 0003d000 08:14 3548565                    /usr/lib/x86_64-linux-gnu/libfontconfig.so.1.9.0
7ff62f91f000-7ff62f921000 r--p 0003c000 08:14 3548565                    /usr/lib/x86_64-linux-gnu/libfontconfig.so.1.9.0
7ff62f921000-7ff62f926000 rw-p 0003e000 08:14 3548565                    /usr/lib/x86_64-linux-gnu/libfontconfig.so.1.9.0
7ff62f926000-7ff62f92d000 r-xp 00000000 08:14 530590                     /lib/x86_64-linux-gnu/librt-2.23.so
7ff62f92d000-7ff62fb2c000 ---p 00007000 08:14 530590                     /lib/x86_64-linux-gnu/librt-2.23.so
7ff62fb2c000-7ff62fb2d000 r--p 00006000 08:14 530590                     /lib/x86_64-linux-gnu/librt-2.23.so
7ff62fb2d000-7ff62fb2e000 rw-p 00007000 08:14 530590                     /lib/x86_64-linux-gnu/librt-2.23.so
7ff62fb2e000-7ff62fc36000 r-xp 00000000 08:14 530568                     /lib/x86_64-linux-gnu/libm-2.23.so
[... cut ...]
7ff6300ab000-7ff6302aa000 ---p 00070000 08:14 3801714                    /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1.2.0
7ff6302aa000-7ff6302ad000 r--p 0006f000 08:14 3801714                    /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1.2.0
7ff6302ad000-7ff6302ae000 rw-p 00072000 08:14 3801714                    /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1.2.0
7ff6302ae000-7ff6302af000 rw-p 00000000 00:00 0 
7ff6302af000-7ff630596000 r-xp 00000000 08:14 3548121                    /usr/lib/x86_64-linux-gnu/libQtCore.so.4.8.7Aborted (core dumped)

sunhaibo1, 2020-02-24:
How to deal with it

sunhaibo1, 2020-02-24:
> how can i fix that ? @RainerKuemmerle

How to deal with it

closed 2025-02-06 by github-actions[bot]