RainerKuemmerle/g2o #309: Memory Corruption and Segfault on basic example

Rookfighter, 2018-10-04:
Hi everybody,

first of all thanks for this great piece of software! I have used g2o before, but I get segfault and memory corruption errors with a fairly new version of g2o. Note that compiling works like a charm, no errors there. I have also used g2o before with an older version (latest tag).
Here are my details:

OS: Ubuntu 16.04
G2O Revision:  653c9eedc51f73394451cd8f55463c962b4092ab
GCC: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
CMake: cmake version 3.5.1

I have created a very simple example which crashes in my setup all the time. 

```cpp
#include <g2o/core/block_solver.h>
#include <g2o/core/optimization_algorithm_levenberg.h>
#include <g2o/solvers/csparse/linear_solver_csparse.h>
#include <g2o/types/slam3d/edge_se3.h>
#include <g2o/types/slam3d/vertex_se3.h>

typedef g2o::LinearSolverCSparse<g2o::BlockSolverX::PoseMatrixType> SlamLinearSolver;

int main(int argc, char **argv)
{
    g2o::SparseOptimizer optimizer;

    auto linearSolver = g2o::make_unique<SlamLinearSolver>();
    linearSolver->setBlockOrdering(false);

    auto blockSolver =
        g2o::make_unique<g2o::BlockSolverX>(std::move(linearSolver));

    g2o::OptimizationAlgorithm *algorithm =
        new g2o::OptimizationAlgorithmLevenberg(std::move(blockSolver));

    optimizer.setVerbose(true);
    optimizer.setAlgorithm(algorithm);

    g2o::VertexSE3 *v = new g2o::VertexSE3();
    v->setId(0);
    v->setEstimate(g2o::Isometry3::Identity());
    v->setFixed(false);
    optimizer.addVertex(v);

    v = new g2o::VertexSE3();
    v->setId(1);
    v->setEstimate(g2o::Isometry3::Identity());
    v->setFixed(false);
    optimizer.addVertex(v);

    g2o::EdgeSE3 *e = new g2o::EdgeSE3();
    e->setInformation(g2o::EdgeSE3::InformationType::Identity());
    e->setMeasurement(g2o::Isometry3::Identity());
    e->vertices()[0] = optimizer.vertex(0);
    e->vertices()[1] = optimizer.vertex(1);
    optimizer.addEdge(e);

    optimizer.initializeOptimization();
    optimizer.optimize(100);

    return 0;
}
```

The crash creates direct command line output stating a memory corruption as follows:

```
*** Error in `./devel/lib/sensor-calibration/calibrate_sensors': free(): invalid next size (fast): 0x0000000000b64770 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f4f082b77e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7f4f082c037a]
[... 149 lines cut ...]
7f4f09412000-7f4f0947f000 r-xp 00000000 00:33 1069120                    /home/fabian/develop/calib_ws/src/sensor_calibration/external/install/lib/libg2o_types_slam3d.so
7f4f0947f000-7f4f0967f000 ---p 0006d000 00:33 1069120                    /home/fabian/develop/calib_ws/src/sensor_calibration/external/install/lib/libg2o_types_slam3d.so
7f4f0967f000-7f4f09682000 r--p 0006d000 00:33 1069120                    /home/fabian/develop/calib_ws/src/sensor_calibration/external/install/lib/libg2o_types_slam3d.so
7f4f09682000-7f4f09683000 rw-p 00070000 00:33 1069120                    /home/fabian/develop/calib_ws/src/sensor_calibration/external/install/lib/libg2o_types_slam3d.so
7f4f09683000-7f4f096e1000 r-xp 00000000 00:33 1066238                    /home/fabian/develop/calib_ws/src/sensor_calibration/external/install/lib/libg2o_core.so
7f4f096e1000-7f4f098e0000 ---p 0005e000 00:33 1066238                    /home/fabian/develop/calib_ws/src/sensor_calibration/external/install/lib/libg2o_core.so
7f4f098e0000-7f4f098e2000 r--p 0005d000 00:33 1066238                    /home/fabian/develop/calib_ws/src/sensor_calibration/external/install/lib/libg2o_core.so
7f4f098e2000-7f4f098e3000 rw-p 0005f000 00:33 1066238                    /home/fabian/develop/calib_ws/src/sensor_calibration/external/install/lib/libg2o_core.so
7f4f098e3000-7f4f09909000 r-xp 00000000 08:03 3150652                    /lib/x86_64-linux-gnu/ld-2.23.so
7f4f09aae000-7f4f09abf000 rw-p 00000000 00:00 0 
7f4f09b05000-7f4f09b08000 rw-p 00000000 00:00 0 
7f4f09b08000-7f4f09b09000 r--p 00025000 08:03 3150652                    /lib/x86_64-linux-gnu/ld-2.23.so
7f4f09b09000-7f4f09b0a000 rw-p 00026000 08:03 3150652                    /lib/x86_64-linux-gnu/ld-2.23.so
7f4f09b0a000-7f4f09b0b000 rw-p 00000000 00:00 0 
7ffcd4e5f000-7ffcd4e81000 rw-p 00000000 00:00 0                          [stack]Aborted (core dumped)
```

```gdb``` returns the following backtrace:

```
(gdb) bt
#0  0x00007f81aef43425 in g2o::EdgeSE3::setMeasurement(Eigen::Transform<double, 3, 1, 0> const&) () from /home/fabian/develop/calib_ws/src/sensor_calibration/external/install/lib/libg2o_types_slam3d.so
#1  0x00000000004d7d29 in main (argc=7, argv=0x7fffcf22ae38) at /home/fabian/develop/calib_ws/src/sensor_calibration/src/calibrate_sensors.cpp:40

```

RainerKuemmerle, 2018-10-04:
I pushed 0d4e98b691988c400450c79d29cb3dae4f6bc8a5 which transforms your example into a unit test.
Does work fine in my local setup:
OS: Ubuntu 18.04
G2O Revision: 0d4e98b69
GCC: gcc (Ubuntu 7.3.0-16ubuntu3) 7.3.0
CMake: cmake version 3.10.2

Also seems to run on Travis without any problems
https://travis-ci.org/RainerKuemmerle/g2o/jobs/437266089

Rookfighter, 2018-10-05:
Tested it just now and the test works fine, no crash. Maybe something is off with my build system?

I created a very simple project which reproduces the issue.

You can find it here:

https://github.com/Rookfighter/g2o_test

I installed g2o locally in the ```external``` sub directory of the repo.

The cmake file looks like this:

```cmake
cmake_minimum_required(VERSION 2.8.3)

project(g2o_test)

set(CMAKE_CXX_STANDARD 11)
set(CMAKE_CXX_STANDARD_REQUIRED ON)

list(APPEND CMAKE_MODULE_PATH "${CMAKE_CURRENT_LIST_DIR}/cmake")

# find g2o
set(G2O_ROOT "${CMAKE_CURRENT_LIST_DIR}/external")
find_package(G2O REQUIRED)
# find required libraries for g2o
find_package(CSparse REQUIRED)
# find eigen 3
find_package(Eigen3 REQUIRED)

set(G2O_LIBRARIES
    ${G2O_SOLVER_CSPARSE}
    ${G2O_SOLVER_CSPARSE_EXTENSION}
    ${CSPARSE_LIBRARY}
    ${G2O_TYPES_SLAM3D}
    ${G2O_CORE_LIBRARY}
    ${G2O_STUFF_LIBRARY}
)

include_directories(
    ${G2O_INCLUDE_DIR}
    ${CSPARSE_INCLUDE_DIR}
    ${EIGEN3_INCLUDE_DIR}
)

add_executable(g2o_test "main.cpp")
target_link_libraries(g2o_test ${G2O_LIBRARIES})

```

RainerKuemmerle, 2018-10-05:
I gave it a try on my machine.
Seems to run smoothly.

```
.:[ goki@birdy:~/workspace/tmp/g2o_test/build ]:. (git)-[master] 
$ export LD_LIBRARY_PATH=$(pwd)/../external/lib
.:[ goki@birdy:~/workspace/tmp/g2o_test/build ]:. (git)-[master] 
$ ldd g2o_test
	linux-vdso.so.1 (0x00007fffdaf2e000)
	libg2o_csparse_extension.so => /home/goki/workspace/tmp/g2o_test/build/../external/lib/libg2o_csparse_extension.so (0x0000149d69c9c000)
	libcxsparse.so.3 => /usr/lib/x86_64-linux-gnu/libcxsparse.so.3 (0x0000149d69a71000)
	libg2o_types_slam3d.so => /home/goki/workspace/tmp/g2o_test/build/../external/lib/libg2o_types_slam3d.so (0x0000149d6980e000)
	libg2o_core.so => /home/goki/workspace/tmp/g2o_test/build/../external/lib/libg2o_core.so (0x0000149d695b5000)
	libg2o_stuff.so => /home/goki/workspace/tmp/g2o_test/build/../external/lib/libg2o_stuff.so (0x0000149d69399000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x0000149d6900b000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x0000149d68df3000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x0000149d68a02000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x0000149d68664000)
	libg2o_opengl_helper.so => /home/goki/workspace/tmp/g2o_test/build/../external/lib/libg2o_opengl_helper.so (0x0000149d6845f000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/libGL.so.1 (0x0000149d681d3000)
	/lib64/ld-linux-x86-64.so.2 (0x0000149d6a1e7000)
	libGLU.so.1 => /usr/lib/x86_64-linux-gnu/libGLU.so.1 (0x0000149d67f64000)
	libGLX.so.0 => /usr/lib/x86_64-linux-gnu/libGLX.so.0 (0x0000149d67d33000)
	libGLdispatch.so.0 => /usr/lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x0000149d67a7d000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x0000149d6785e000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x0000149d67526000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x0000149d67322000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x0000149d670fa000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x0000149d66ef6000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x0000149d66cf0000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x0000149d66adb000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x0000149d668d3000)
.:[ goki@birdy:~/workspace/tmp/g2o_test/build ]:. (git)-[master] 
$ ./g2o_test
iteration= 0	 chi2= 0.000000	 time= 8.6569e-05	 cumTime= 8.6569e-05	 edges= 1	 schur= 0	 lambda= 0.000020	 levenbergIter= 1
```
I assume I would need to test with an virtual machine running Ubuntu 16.04.

Rookfighter, 2018-10-06:
It's weird that the unit tests work fine, but my project does not! I'll check on this, maybe there is a problem in finding the libraries at runtime.

sjulier, 2018-10-06:
Hmm... Does valgrind provide any useful insights? I find it helps a lot with things like uninitialised variables and accidentally freeing memory (which sometimes continues to work for quite a while..)

Rookfighter, 2018-10-06:
Here's the output of valgrind. Note that g2o is compiled without specifying CMAKE_BUILD_TYPE (default value).

```
fabian@fabian-W110ER ~/develop/g2o_test/build (master=) $ valgrind ./g2o_test
==5428== Memcheck, a memory error detector
==5428== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
==5428== Using Valgrind-3.11.0 and LibVEX; rerun with -h for copyright info
==5428== Command: ./g2o_test
==5428== 
==5428== Invalid write of size 8
==5428==    at 0x52A0823: g2o::EdgeSE3::setMeasurement(Eigen::Transform<double, 3, 1, 0> const&) (in /home/fabian/develop/g2o_test/external/lib/libg2o_types_slam3d.so)
==5428==    by 0x4D6F38: main (main.cpp:40)
==5428==  Address 0x99271c0 is 0 bytes after a block of size 848 alloc'd
==5428==    at 0x4C2DB8F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==5428==    by 0x4DA652: Eigen::internal::aligned_malloc(unsigned long) (Memory.h:224)
==5428==    by 0x4DC452: void* Eigen::internal::conditional_aligned_malloc<true>(unsigned long) (Memory.h:305)
==5428==    by 0x4DBCE8: g2o::EdgeSE3::operator new(unsigned long) (edge_se3.h:46)
==5428==    by 0x4D6EAD: main (main.cpp:38)
==5428== 
==5428== Invalid read of size 8
==5428==    at 0x529FC3B: Eigen::internal::transform_transform_product_impl<Eigen::Transform<double, 3, 1, 0>, Eigen::Transform<double, 3, 1, 0>, false>::run(Eigen::Transform<double, 3, 1, 0> const&, Eigen::Transform<double, 3, 1, 0> const&) (in /home/fabian/develop/g2o_test/external/lib/libg2o_types_slam3d.so)
==5428==    by 0x529E1EE: g2o::EdgeSE3::computeError() (in /home/fabian/develop/g2o_test/external/lib/libg2o_types_slam3d.so)
==5428==    by 0x551BF40: g2o::SparseOptimizer::computeActiveErrors() (in /home/fabian/develop/g2o_test/external/lib/libg2o_core.so)
==5428==    by 0x552549F: g2o::OptimizationAlgorithmLevenberg::solve(int, bool) (in /home/fabian/develop/g2o_test/external/lib/libg2o_core.so)
==5428==    by 0x551D136: g2o::SparseOptimizer::optimize(int, bool) (in /home/fabian/develop/g2o_test/external/lib/libg2o_core.so)
==5428==    by 0x4D6FEA: main (main.cpp:46)
==5428==  Address 0x99271c0 is 0 bytes after a block of size 848 alloc'd
==5428==    at 0x4C2DB8F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==5428==    by 0x4DA652: Eigen::internal::aligned_malloc(unsigned long) (Memory.h:224)
==5428==    by 0x4DC452: void* Eigen::internal::conditional_aligned_malloc<true>(unsigned long) (Memory.h:305)
==5428==    by 0x4DBCE8: g2o::EdgeSE3::operator new(unsigned long) (edge_se3.h:46)
==5428==    by 0x4D6EAD: main (main.cpp:38)
==5428== 
==5428== 
==5428== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==5428==  General Protection Fault
==5428==    at 0x52977A9: void std::vector<Eigen::Transform<double, 3, 1, 0>, Eigen::aligned_allocator<Eigen::Transform<double, 3, 1, 0> > >::_M_emplace_back_aux<Eigen::Transform<double, 3, 1, 0> const&>(Eigen::Transform<double, 3, 1, 0> const&) (in /home/fabian/develop/g2o_test/external/lib/libg2o_types_slam3d.so)
==5428==    by 0x551C3D7: g2o::SparseOptimizer::push(std::vector<g2o::OptimizableGraph::Vertex*, std::allocator<g2o::OptimizableGraph::Vertex*> >&) (in /home/fabian/develop/g2o_test/external/lib/libg2o_core.so)
==5428==    by 0x552550F: g2o::OptimizationAlgorithmLevenberg::solve(int, bool) (in /home/fabian/develop/g2o_test/external/lib/libg2o_core.so)
==5428==    by 0x551D136: g2o::SparseOptimizer::optimize(int, bool) (in /home/fabian/develop/g2o_test/external/lib/libg2o_core.so)
==5428==    by 0x4D6FEA: main (main.cpp:46)
==5428== 
==5428== HEAP SUMMARY:
==5428==     in use at exit: 88,509 bytes in 230 blocks
==5428==   total heap usage: 301 allocs, 71 frees, 98,081 bytes allocated
==5428== 
==5428== LEAK SUMMARY:
==5428==    definitely lost: 0 bytes in 0 blocks
==5428==    indirectly lost: 0 bytes in 0 blocks
==5428==      possibly lost: 0 bytes in 0 blocks
==5428==    still reachable: 88,509 bytes in 230 blocks
==5428==         suppressed: 0 bytes in 0 blocks
==5428== Rerun with --leak-check=full to see details of leaked memory
==5428== 
==5428== For counts of detected and suppressed errors, rerun with: -v
==5428== ERROR SUMMARY: 3 errors from 2 contexts (suppressed: 0 from 0)
Segmentation fault
```

I tried to compile g2o with CMAKE_BUILD_TYPE=Debug and link the test application against it. In this case the error and crash is gone, the application works as expected.

sjulier, 2018-10-06:
Thanks. Strange. Could you try creating a temporary? e.g.,

g2o::Isometry3 I = g2o::Isometry3::Identity();
e->setMeasurement(I);

Rookfighter, 2018-10-06:
Tried it with a temp variable, bit the application still crashes.

I also tried to remove ```EIGEN_MAKE_ALIGNED_OPERATOR_NEW``` from EdgeSE3. But this also crashed. However the valgrind output changed:

```
fabian@fabian-W110ER ~/develop/g2o_test/build (master *=) $ valgrind ./g2o_test 
==14595== Memcheck, a memory error detector
==14595== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
==14595== Using Valgrind-3.11.0 and LibVEX; rerun with -h for copyright info
==14595== Command: ./g2o_test
==14595== 
==14595== Invalid write of size 8
==14595==    at 0x52A0823: g2o::EdgeSE3::setMeasurement(Eigen::Transform<double, 3, 1, 0> const&) (in /home/fabian/develop/g2o_test/external/lib/libg2o_types_slam3d.so)
==14595==    by 0x4D6FB8: main (main.cpp:42)
==14595==  Address 0x99271c0 is 0 bytes after a block of size 848 alloc'd
==14595==    at 0x4C2DB8F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==14595==    by 0x4DA6D2: Eigen::internal::aligned_malloc(unsigned long) (Memory.h:224)
==14595==    by 0x4DC49E: void* Eigen::internal::conditional_aligned_malloc<true>(unsigned long) (Memory.h:305)
==14595==    by 0x4DCC03: g2o::BaseBinaryEdge<6, Eigen::Transform<double, 3, 1, 0>, g2o::VertexSE3, g2o::VertexSE3>::operator new(unsigned long) (base_binary_edge.h:114)
==14595==    by 0x4D6F3C: main (main.cpp:40)
==14595== 
==14595== Invalid read of size 8
==14595==    at 0x529FC3B: Eigen::internal::transform_transform_product_impl<Eigen::Transform<double, 3, 1, 0>, Eigen::Transform<double, 3, 1, 0>, false>::run(Eigen::Transform<double, 3, 1, 0> const&, Eigen::Transform<double, 3, 1, 0> const&) (in /home/fabian/develop/g2o_test/external/lib/libg2o_types_slam3d.so)
==14595==    by 0x529E1EE: g2o::EdgeSE3::computeError() (in /home/fabian/develop/g2o_test/external/lib/libg2o_types_slam3d.so)
==14595==    by 0x551BF40: g2o::SparseOptimizer::computeActiveErrors() (in /home/fabian/develop/g2o_test/external/lib/libg2o_core.so)
==14595==    by 0x552549F: g2o::OptimizationAlgorithmLevenberg::solve(int, bool) (in /home/fabian/develop/g2o_test/external/lib/libg2o_core.so)
==14595==    by 0x551D136: g2o::SparseOptimizer::optimize(int, bool) (in /home/fabian/develop/g2o_test/external/lib/libg2o_core.so)
==14595==    by 0x4D706A: main (main.cpp:48)
==14595==  Address 0x99271c0 is 0 bytes after a block of size 848 alloc'd
==14595==    at 0x4C2DB8F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==14595==    by 0x4DA6D2: Eigen::internal::aligned_malloc(unsigned long) (Memory.h:224)
==14595==    by 0x4DC49E: void* Eigen::internal::conditional_aligned_malloc<true>(unsigned long) (Memory.h:305)
==14595==    by 0x4DCC03: g2o::BaseBinaryEdge<6, Eigen::Transform<double, 3, 1, 0>, g2o::VertexSE3, g2o::VertexSE3>::operator new(unsigned long) (base_binary_edge.h:114)
==14595==    by 0x4D6F3C: main (main.cpp:40)
==14595== 
==14595== 
==14595== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==14595==  General Protection Fault
==14595==    at 0x52977A9: void std::vector<Eigen::Transform<double, 3, 1, 0>, Eigen::aligned_allocator<Eigen::Transform<double, 3, 1, 0> > >::_M_emplace_back_aux<Eigen::Transform<double, 3, 1, 0> const&>(Eigen::Transform<double, 3, 1, 0> const&) (in /home/fabian/develop/g2o_test/external/lib/libg2o_types_slam3d.so)
==14595==    by 0x551C3D7: g2o::SparseOptimizer::push(std::vector<g2o::OptimizableGraph::Vertex*, std::allocator<g2o::OptimizableGraph::Vertex*> >&) (in /home/fabian/develop/g2o_test/external/lib/libg2o_core.so)
==14595==    by 0x552550F: g2o::OptimizationAlgorithmLevenberg::solve(int, bool) (in /home/fabian/develop/g2o_test/external/lib/libg2o_core.so)
==14595==    by 0x551D136: g2o::SparseOptimizer::optimize(int, bool) (in /home/fabian/develop/g2o_test/external/lib/libg2o_core.so)
==14595==    by 0x4D706A: main (main.cpp:48)
==14595== 
==14595== HEAP SUMMARY:
==14595==     in use at exit: 88,509 bytes in 230 blocks
==14595==   total heap usage: 301 allocs, 71 frees, 98,081 bytes allocated
==14595== 
==14595== LEAK SUMMARY:
==14595==    definitely lost: 0 bytes in 0 blocks
==14595==    indirectly lost: 0 bytes in 0 blocks
==14595==      possibly lost: 0 bytes in 0 blocks
==14595==    still reachable: 88,509 bytes in 230 blocks
==14595==         suppressed: 0 bytes in 0 blocks
==14595== Rerun with --leak-check=full to see details of leaked memory
==14595== 
==14595== For counts of detected and suppressed errors, rerun with: -v
==14595== ERROR SUMMARY: 3 errors from 2 contexts (suppressed: 0 from 0)
Segmentation fault
```

jlblancoc, 2018-10-06:
Just in case: are you in a 32bit system? Mem alignment and Eigen gets
tricky then and errors like this one typically arises if code was tested in
64bit only...

Cheers

Rookfighter, 2018-10-06:
Haha, good point! I'm not sure if anyone uses 32bit Ubuntu these days, anyway I use a 64bit system.

```uname``` returns the following:

```
Linux fabian-W110ER 4.15.0-36-generic #39~16.04.1-Ubuntu SMP Tue Sep 25 08:59:23 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

Rookfighter, 2018-10-06:
Just as a note I have Eigen 3.2 installed, from the standard Ubuntu repositories.

```
libeigen3-dev:
  Installed: 3.3~beta1-2
  Candidate: 3.3~beta1-2
  Version table:
 *** 3.3~beta1-2 500
        500 http://de.archive.ubuntu.com/ubuntu xenial/universe amd64 Packages
        500 http://de.archive.ubuntu.com/ubuntu xenial/universe i386 Packages
        100 /var/lib/dpkg/status
```
CMake output:

```
-- Checking for module 'eigen3'
--   Found eigen3, version 3.2.92
-- Found eigen: /usr/include/eigen3
```

Rookfighter, 2018-10-06:
Next try:

I removed the lines setting the default ```CMAKE_BUILD_TYPE ``` to ```Release``` in the build system of g2o . ```CMakeLists.txt``` l.14 - l.19:

``` cmake
# default built type
#if(NOT CMAKE_BUILD_TYPE)
#  set(CMAKE_BUILD_TYPE Release CACHE STRING
#      "Choose the type of build, options are: None Debug Release RelWithDebInfo MinSizeRel."
#      FORCE)
#endif(NOT CMAKE_BUILD_TYPE)
```

 So g2o is now built with the empty ```CMAKE_BUILD_TYPE ```. This resolved the SEGFAULT and the test application works as expected, no crash.

Seems like the optimization of the compiler breaks something inside g2o. I guess the next step would be to debug the resulting assembly when optimization is enabled. However, I'm afraid I currently don't have the time to do that. This workaround is good enough for me now. Hopefully the loss of speed is not too serious.

If I can be of any further help resolving this issue, let me know.

sjulier, 2018-10-06:
Thanks. This is very strange. It does look a bit like a compiler bug of some kind.

Not sure if it can be applied, but there are various tricks to selectively disable optimisation. e.g.,

https://stackoverflow.com/questions/2219829/how-to-prevent-gcc-optimizing-some-statements-in-c

closed 2018-11-18 by RainerKuemmerle

YangSiri, 2019-11-09:
Hi, @Rookfighter  
I may have run into similar problem. In the end, you changed the CMAKE_BUILD_TYPE and it worked? 
I'm using the latest version of g2o. Do you have any new solutions ?

Looking forward to your generous help.THANK

Rookfighter, 2019-11-10:
Hi @YangSiri,
so basically using ```-DCMAKE_BUILD_TYPE=Debug``` made the segfault disappear. However, ```Eigen3``` and ```g2o``` get super slow if all compiler optimizations are disabled. So it was not usable for me and I decided to take another route and use a different library. Anyway, this issue was closed over a year ago. If you run into similar problems with head of master, then I guess you should open a new issue.

YangSiri, 2019-11-10:
Thanks, @Rookfighter 
May you share the lib you are using instead?

PR #314 (2018-11-18): Fix for memory corruption and core dumps
This fixed a similar problem I ran into which is reported in #166. Might also fix #309.

Disable `-march native` because this will enable -mavx or -mavx2 on supported systems which override their corresponding -msse1/2/3/4 versions and lead to crashes whenever an external binary (which links g2o) is compiled with e.g. -msse4.2 and not -mavx/-mavx2

Eigen code compiled with -msse4.2 will be 16 bit aligned and Eigen code compiled with -avx will be 32bit aligned. If you have a binary compiled with -msee4.2 and create an Eigen::Isometry3d it will be 16bit m [... cut]