RainerKuemmerle/g2o #315: malloc():memory corruption(fast):0x0000000001055340 ***

Junkai-Liu, 2018-10-20:
*** Error in `./build/pose_graph_g2o_SE3': malloc(): memory corruption (fast): 0x0000000001055340 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7fca3f2977e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x82651)[0x7fca3f2a2651]
/lib/x86_64-linux-gnu/libc.so.6(__libc_malloc+0x54)[0x7fca3f2a4184]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(_Znwm+0x18)[0x7fca3f895e78]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZNSs4_Rep9_S_createEmmRKSaIcE+0x59)[0x7fca3f8d6e39]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZNSs4_Rep8_M_cloneERKSaIcEm+0x1b)[0x7fca3f8d7c6b]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZNSs7reserveEm+0x44)[0x7fca3f8d7d24]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZNKSt7num_getIcSt19istreambuf_iteratorIcSt11char_traitsIcEEE6do_getES3_S3_RSt8ios_baseRSt12_Ios_IostateRd+0x5e)[0x7fca3f91590e]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZNSi10_M_extractIdEERSiRT_+0x6f)[0x7fca3f90105f]
/usr/local/lib/libg2o_types_slam3d.so(_ZN3g2o7EdgeSE34readERSi+0x305)[0x7fca3fe9db05]
./build/pose_graph_g2o_SE3(main+0x5f8)[0x407208]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7fca3f240830]
./build/pose_graph_g2o_SE3(_start+0x29)[0x407399]
======= Memory map: ========
00400000-0041b000 r-xp 00000000 08:07 2489324                            /home/rush/projects/ch11/build/pose_graph_g2o_SE3
0061a000-0061b000 r--p 0001a000 08:07 2489324                            /home/rush/projects/ch11/build/pose_graph_g2o_SE3
0061b000-0061c000 rw-p 0001b000 08:07 2489324                            /home/rush/projects/ch11/build/pose_graph_g2o_SE3
00f2c000-01076000 rw-p 00000000 00:00 0                                  [heap]
7fca34000000-7fca34021000 rw-p 00000000 00:00 0 
7fca34021000-7fca38000000 ---p 00000000 00:00 0 
7fca3a148000-7fca3a186000 r-xp 00000000 08:06 666430                     /usr/lib/x86_64-linux-gnu/libquadmath.so.0.0.0
7fca3a186000-7fca3a385000 ---p 0003e000 08:06 666430                     /usr/lib/x86_64-linux-gnu/libquadmath.so.0.0.0
7fca3a385000-7fca3a386000 r--p 0003d000 08:06 666430                     /usr/lib/x86_64-linux-gnu/libquadmath.so.0.0.0
7fca3a386000-7fca3a387000 rw-p 0003e000 08:06 666430                     /usr/lib/x86_64-linux-gnu/libquadmath.so.0.0.0
7fca3a388000-7fca3a38d000 r-xp 00000000 08:06 666186                     /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7fca3a38d000-7fca3a58c000 ---p 00005000 08:06 666186                     /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7fca3a58c000-7fca3a58d000 r--p 00004000 08:06 666186                     /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7fca3a58d000-7fca3a58e000 rw-p 00005000 08:06 666186                     /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7fca3a590000-7fca3a592000 r-xp 00000000 08:06 666175                     /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7fca3a592000-7fca3a792000 ---p 00002000 08:06 666175                     /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7fca3a792000-7fca3a793000 r--p 00002000 08:06 666175                     /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7fca3a793000-7fca3a794000 rw-p 00003000 08:06 666175                     /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7fca3a798000-7fca3a8c1000 r-xp 00000000 08:06 677186                     /usr/lib/x86_64-linux-gnu/libgfortran.so.3.0.0
7fca3a8c1000-7fca3aac0000 ---p 00129000 08:06 677186                     /usr/lib/x86_64-linux-gnu/libgfortran.so.3.0.0
7fca3aac0000-7fca3aac1000 r--p 00128000 08:06 677186                     /usr/lib/x86_64-linux-gnu/libgfortran.so.3.0.0
7fca3aac1000-7fca3aac3000 rw-p 00129000 08:06 677186                     /usr/lib/x86_64-linux-gnu/libgfortran.so.3.0.0
7fca3aac8000-7fca3aacb000 r-xp 00000000 08:06 136318                     /lib/x86_64-linux-gnu/libdl-2.23.so
7fca3aacb000-7fca3acca000 ---p 00003000 08:06 136318                     /lib/x86_64-linux-gnu/libdl-2.23.so
7fca3acca000-7fca3accb000 r--p 00002000 08:06 136318                     /lib/x86_64-linux-gnu/libdl-2.23.so
7fca3accb000-7fca3accc000 rw-p 00003000 08:06 136318                     /lib/x86_64-linux-gnu/libdl-2.23.so
7fca3acd0000-7fca3ace8000 r-xp 00000000 08:06 136296                     /lib/x86_64-linux-gnu/libpthread-2.23.so
7fca3ace8000-7fca3aee7000 ---p 00018000 08:06 136296                     /lib/x86_64-linux-gnu/libpthread-2.23.so
7fca3aee7000-7fca3aee8000 r--p 00017000 08:06 136296                     /lib/x86_64-linux-gnu/libpthread-2.23.so
7fca3aee8000-7fca3aee9000 rw-p 00018000 08:06 136296                     /lib/x86_64-linux-gnu/libpthread-2.23.so
7fca3aee9000-7fca3aeed000 rw-p 00000000 00:00 0 
7fca3aef0000-7fca3af00000 r-xp 00000000 08:06 657667                     /usr/lib/x86_64-linux-gnu/libdrm.so.2.4.0
7fca3af00000-7fca3b100000 ---p 00010000 08:06 657667                     /usr/lib/x86_64-linux-gnu/libdrm.so.2.4.0
7fca3b100000-7fca3b101000 r--p 00010000 08:06 657667                     /usr/lib/x86_64-linux-gnu/libdrm.so.2.4.0
7fca3b101000-7fca3b102000 rw-p 00011000 08:06 657667                     /usr/lib/x86_64-linux-gnu/libdrm.so.2.4.0
7fca3b108000-7fca3b10d000 r-xp 00000000 08:06 666226                     /usr/lib/x86_64-linux-gnu/libXxf86vm.so.1.0.0
7fca3b10d000-7fca3b30c000 ---p 00005000 08:06 666226                     /usr/lib/x86_64-linux-gnu/libXxf86vm.so.1.0.0
7fca3b30c000-7fca3b30d000 r--p 00004000 08:06 666226                     /usr/lib/x86_64-linux-gnu/libXxf86vm.so.1.0.0
7fca3b30d000-7fca3b30e000 rw-p 00005000 08:06 666226                     /usr/lib/x86_64-linux-gnu/libXxf86vm.so.1.0.0
7fca3b310000-7fca3b331000 r-xp 00000000 08:06 667593                     /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0
7fca3b331000-7fca3b530000 ---p 00021000 08:06 667593                     /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0
7fca3b530000-7fca3b531000 r--p 00020000 08:06 667593                     /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0
7fca3b531000-7fca3b532000 rw-p 00021000 08:06 667593                     /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0
7fca3b538000-7fca3b53c000 r-xp 00000000 08:06 667561                     /usr/lib/x86_64-linux-gnu/libxcb-dri2.so.0.0.0
7fca3b53c000-7fca3b73b000 ---p 00004000 08:06 667561                     /usr/lib/x86_64-linux-gnu/libxcb-dri2.so.0.0.0
7fca3b73b000-7fca3b73c000 r--p 00003000 08:06 667561                     /usr/lib/x86_64-linux-gnu/libxcb-dri2.so.0.0.0
7fca3b73c000-7fca3b73d000 rw-p 00004000 08:06 667561                     /usr/lib/x86_64-linux-gnu/libxcb-dri2.so.0.0.0
7fca3b740000-7fca3b757000 r-xp 00000000 08:06 667565                     /usr/lib/x86_64-linux-gnu/libxcb-glx.so.0.0.0
7fca3b757000-7fca3b956000 ---p 00017000 08:06 667565                     /usr/lib/x86_64-linux-gnu/libxcb-glx.so.0.0.0
7fca3b956000-7fca3b958000 r--p 00016000 08:06 667565                     /usr/lib/x86_64-linux-gnu/libxcb-glx.so.0.0.0
7fca3b958000-7fca3b959000 rw-p 00018000 08:06 667565                     /usr/lib/x86_64-linux-gnu/libxcb-glx.so.0.0.0
7fca3b960000-7fca3ba95000 r-xp 00000000 08:06 657844                     /usr/lib/x86_64-linux-gnu/libX11.so.6.3.0
7fca3ba95000-7fca3bc95000 ---p 00135000 08:06 657844                     /usr/lib/x86_64-linux-gnu/libX11.so.6.3.0
7fca3bc95000-7fca3bc96000 r--p 00135000 08:06 657844                     /usr/lib/x86_64-linux-gnu/libX11.so.6.3.0
[... 77 lines cut ...]
7fca3ea8e000-7fca3ea8f000 rw-p 0006e000 08:06 676687                     /usr/lib/libblas/libblas.so.3.6.0
7fca3ea90000-7fca3eb00000 r-xp 00000000 08:06 924504                     /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1.2.0
7fca3eb00000-7fca3ecff000 ---p 00070000 08:06 924504                     /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1.2.0
7fca3ecff000-7fca3ed02000 r--p 0006f000 08:06 924504                     /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1.2.0
7fca3ed02000-7fca3ed03000 rw-p 00072000 08:06 924504                     /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1.2.0
7fca3ed03000-7fca3ed04000 rw-p 00000000 00:00 0 
7fca3ed08000-7fca3ed0b000 r-xp 00000000 08:06 657062                     /usr/local/lib/libg2o_opengl_helper.so
7fca3ed0b000-7fca3ef0b000 ---p 00003000 08:06 657062                     /usr/local/lib/libg2o_opengl_helper.so
7fca3ef0b000-7fca3ef0c000 r--p 00003000 08:06 657062                     /usr/local/lib/libg2o_opengl_helper.so
7fca3ef0c000-7fca3ef0d000 rw-p 00004000 08:06 657062                     /usr/local/lib/libg2o_opengl_helper.so
7fca3ef10000-7fca3f018000 r-xp 00000000 08:06 135967                     /lib/x86_64-linux-gnu/libm-2.23.so
[... cut ...]
7fca3f606000-7fca3f805000 ---p 00016000 08:06 136507                     /lib/x86_64-linux-gnu/libgcc_s.so.1
7fca3f805000-7fca3f806000 rw-p 00015000 08:06 136507                     /lib/x86_64-linux-gnu/libgcc_s.so.1
7fca3f808000-7fca3f97a000 r-xp 00000000 08:06 657632                     /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7fca3f97a000-7fca3fb7a000 ---p 00172000 08:06 657632                     /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7fca3fb7a000-7fca3fb84000 r--p 00172000 08:06 657632                     /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21Aborted (core dumped)

jinfagang, 2019-04-04:
Same issue here with a simple solver

RainerKuemmerle, 2019-04-07:
Can you please verify that you are using the same compiler flags for the g2o library and your binary.
It might be that the flags of the library are interfering here.

jinfagang, 2019-04-07:
@RainerKuemmerle  thanks for reply, for me, I just built g2o and installed into system path. When building myself tiny program, simply using CMake with c++11 support, nothing more flags, then link g2o  lib

closed 2025-02-06 by github-actions[bot]