RainerKuemmerle/g2o #322: Segmentation fault while iterating

pnakis, 2018-11-08:
Greetings everyone. I am in the process of building a SLAM algorithm and i am using g2o to perform all of my optimizations. The issue i have is that sometimes i get an unexpected Segmentation Fault while the optimizer is iterating. 

Platform: Ubuntu 16.04 LTS
Eigen v 3.2.92
` 
    
    g2o::SparseOptimizer optimizer;
    std::unique_ptr<g2o::BlockSolver_6_3::LinearSolverType> linearSolver;

    linearSolver = g2o::make_unique<g2o::LinearSolverEigen<g2o::BlockSolver_6_3::PoseMatrixType>>();

    g2o::OptimizationAlgorithmLevenberg* solver = new g2o::OptimizationAlgorithmLevenberg(
            g2o::make_unique<g2o::BlockSolver_6_3>(std::move(linearSolver)));

    optimizer.setAlgorithm( solver );
    optimizer.setVerbose( true );

    for (int keyframe = std::max (0, keyFrameNo-BAWindow) ; keyframe < keyFrameNo ; keyframe++){

        // Add Keyframes to be optimized
        g2o::VertexSE3Expmap * KF = new g2o::VertexSE3Expmap();

        int baseID = 1000000;
        KF -> setId( baseID + keyframe );
        // Set the first Keyframe of the optimization to FIXED
        if ( keyframe == std::max (0, keyFrameNo-BAWindow) ) KF -> setFixed( true );

        cv::Mat rodRot;
        cv::Rodrigues(R[keyframe], rodRot);

        Eigen::Matrix < double,3,3 > rot;

        rot <<  rodRot.at< double >(0,0), rodRot.at< double >(0,1), rodRot.at< double >(0,2),
                rodRot.at< double >(1,0), rodRot.at< double >(1,1), rodRot.at< double >(1,2),
                rodRot.at< double >(2,0), rodRot.at< double >(2,1), rodRot.at< double >(2,2);

        Eigen::Matrix < double,3,1 > trans(t[keyframe].at< double >(0,0), t[keyframe].at< double >(0,1), t[keyframe].at< double >(0,2));

        KF -> setEstimate( g2o::SE3Quat(rot, trans) );

        optimizer.addVertex( KF );
        // Add Keyframes - end

        // Add 3D Map Points
        for (int MP = 0 ; MP < keyframe3DPointIndices[keyframe].size() ; MP++){

            int ID = keyframe3DPointIndices[keyframe][MP];
            g2o::OptimizableGraph::Vertex* isInserted = optimizer.vertex( ID );

            //  if (points3DApp[ ID ] < 3) continue;

            // Add 3D Map Point if it's not inserted in the graph
            if (!isInserted){

                g2o::VertexSBAPointXYZ * mapPoint = new g2o::VertexSBAPointXYZ();

                mapPoint -> setId( ID );
                mapPoint -> setMarginalized( true );
                mapPoint -> setEstimate( Eigen::Vector3d( points3D[ID].x, points3D[ID].y, points3D[ID].z ) );

                if (points3DApp[ ID ] > 4) mapPoint -> setFixed( true );

                optimizer.addVertex( mapPoint );
            }

            // Add Edge between Keyframe Vertex and Map Point Vertex
            g2o::EdgeSE3ProjectXYZ * EDG = new g2o::EdgeSE3ProjectXYZ();

            EDG -> setVertex( 0, optimizer.vertex( ID ) ); // 3D Map Point
            EDG -> setVertex( 1, optimizer.vertex( baseID + keyframe ) ); // Keyframe

            cv::Point2d obs = cv::Point2d(matchedKeypoints[keyframe][MP].pt);
            EDG -> setMeasurement( Eigen::Vector2d( obs.x, obs.y) );

            EDG -> information() = Eigen::Matrix2d::Identity();

            EDG -> fx = cameraMatrix[0].at< double >(0,0);
            EDG -> fy = cameraMatrix[0].at< double >(1,1);
            EDG -> cx = cameraMatrix[0].at< double >(0,2);
            EDG -> cy = cameraMatrix[0].at< double >(1,2);

            optimizer.addEdge( EDG );
        }
    }

    optimizer.initializeOptimization();
    optimizer.optimize(10);`

In the code above, i am optimizing the last m Keyframes (BAWindow), R and t matrices hold the rotation in a Rodrigues format and the translation of the poses, keyframe3DPointIndices is a vector of vectors that holds the 3D Point Indices of every Keyframe, points3D is a vector of cv::Point3d that holds the 3D Map Points and matchedKeypoints is a vector of vectors cv::KeyPoint that hold the observations of every Keyframe. 

For example, two poses and 208 landmarks get optimized correctly with 10 iterations, when i change the number of iterations to 20, i get a Segmentation Fault (Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)) after the 14nth iteration. If i leave the iteration number at 10 and have a BAWindow of 8 (optimize the last 8 Keyframes), everything works as expected but i get a segmentation fault at the 31st Keyframe (optimize keyframes 24 - 31) at the 4th iteration. If i change the BAWindow parameter to a smaller or larger number, the segmentation fault appears somewhere else at a different iteration. I have tried this with LinearSolverCSparse and LinearSolverCholmod solvers, change the edge type to EdgeProjectXYZ2UV but that doesn't seem to fix the issue.

I tried debugging the problem but for the life of me, i couldn't find anything that explains this phenomenon.

closed 2018-11-24 by pnakis

EyeeHKao, 2019-06-22:
666

EyeeHKao, 2019-07-04:
I rebuild in the debug mode, but I found it runs very solwly(abuout 10 times), but i can not run my code in release mode successly(it will occur the question you mentioned) ,can you help me?

jonesd6, 2020-04-22:
Did you ever find a solution?