RainerKuemmerle/g2o #325: Correctly using EdgeSE3Expmap class

chait-desai, 2018-11-21:
Unless I am missing something, the EdgeSE3Expmap class implementation is not working the way it is supposed to.  
I am setting up a simple optimization problem with just 2  vertices representing poses in SE3: the pose of one of the vertices is fixed and the other is free to change. The idea is to get the free vertex to converge to the pose of the fixed vertex. The "odometry measurement" between the 2 poses is the identity isometry (no rotation and no translation), so that the error associated with the edge in only a function of the difference between the 2 poses.  I am able to get the optimization to behave correctly when the 2 vertices are represented using the VertexSE3 class and the edge is represented using the EdgeSE3 class. However, when I switch to VertexSE3Expmap and EdgeSE3Expmap, I get nonsensical results. Attached is the code (which should be added to the examples directory if you want to run the code)
[SE3test.tar.gz](https://github.com/RainerKuemmerle/g2o/files/2605597/SE3test.tar.gz)
 and the console output of running the code
[image]

steplee, 2018-12-05:
I am experiencing the same thing. If you have resolved this or anyone can help, please help!

steplee, 2018-12-05:
I think I fixed it, I can't run your unit test or afford the time right now to confirm.

In `g2o/types/sba/types_six_dof_expmap.h`, I made these changes to `EdgeSE3Expmap::computeError()`:
```       
         //SE3Quat error_= v2->estimate().inverse()*C*v1->estimate();                                                                                                                                                                                                              
         SE3Quat error_= C.inverse() * v1->estimate().inverse() * v2->estimate();
```
Which is the order of operations from `EdgeSE3::computeError()`

winf, 2019-01-16:
@chait-desai ,do not forget to set the information of the edge,
 `Eigen::MatrixXd Info = Eigen::MatrixXd::Identity(6,6);  constraint->setInformation(Info);`
after add these code, your SE3test should get the right result. Tested on the current master branch.

LongruiDong, 2020-12-15:
> Unless I am missing something, the EdgeSE3Expmap class implementation is not working the way it is supposed to.
> I am setting up a simple optimization problem with just 2 vertices representing poses [...]

Have you figured it out?

closed 2025-02-05 by github-actions[bot]