RainerKuemmerle/g2o #326: how to optimize on a part of a graph

capriciousZihao, 2018-11-27:
Does it possible to optimize only on a part of a graph?
for example, the graph have vertex1, vertex2, vertex 3, vertex4, edge12, edge 23, edge34. Can i only optimize on a subgraph composed of vertex3, edge34, and vertex4?
Thanks

sjulier, 2018-11-28:
One option is to call setFixed(true) on all the vertices you don't want to optimise.

capriciousZihao, 2018-11-28:
do you know is there any method or function that can return the vertexes that are fixed

sjulier, 2018-11-28:
Not directly. You could call getVertices() on the graph and then iterate over each vertex to see if it's fixed or not using the fixed() method.

capriciousZihao, 2018-11-28:
thanks a lot lot lot :)

RainerKuemmerle, 2018-11-28:
You should look at the different initializeOptimization() methods of the sparse optimizer in case the other vertices / edges should be completely ignored.
g2o/g2o/core/sparse_optimizer.h
```
/**
     * Initializes the structures for optimizing a portion of the graph specified by a subset of edges.
     * Before calling it be sure to invoke marginalized() and fixed() to the vertices you want to include in the 
     * schur complement or to set as fixed during the optimization.
     * @param eset: the subgraph to be optimized.
     * @returns false if somethings goes wrong
     */
    virtual bool initializeOptimization(HyperGraph::EdgeSet& eset);
```

capriciousZihao, 2018-11-28:
Thanks! I will look at the optimization initialization methods.

closed 2018-12-09 by RainerKuemmerle