RainerKuemmerle/g2o #334: SE(2) poses with XYZ landmarks?

means-to-meaning, 2018-12-29:
Can anyone comment on what is the best way to create a graph for a wheeled robot moving on a flat plane (robot pose only has 3 degrees of freedom and known sensor offsets) equipped with RGB-D (doing visual tracking and measurement of XYZ points) and wheel odometry?

Essentially, I'm wondering whether the right thing to do is to define a new constraint type edge_se2_pointXYZ, or maybe to incorporate the known pose constraints somehow into the se3 pose and do 3D SLAM?

Any pointers to the required maths, or code would be most welcome. Apologies if I missed some existing code - this is a very standard setup for any hobbyist building a home robot and the functionality might already be all there. Thanks!

means-to-meaning, 2018-12-31:
@TSC21 Thanks for your suggestion! Yes, casting both the poses and the landmarks into 2D and then doing 2D SLAM is something I thought about. Just wasn't sure if that's the best approach.

But if doing 2D SLAM is the answer for planar motion, I wonder how we can handle other types of sensors which are not so easily amenable to being converted to 2D, such as a stereo, or a mono camera (as covered in the BA and SBA examples).

I looked at the different g2o examples and read the original g2o paper, but it is not immediately clear to me which of the existing types can be applied to some new physical robot configuration. I hope this stuff starts clicking at some point. Thanks for your comment in the mean time!

sume-cn, 2019-08-12:
I've read a paper recently, it provided another way to deal with similar robot setups:
Odometry-Vision-Based Ground Vehicle Motion Estimation With SE(2)-Constrained SE(3) Poses

means-to-meaning, 2019-08-12:
That's a good pointer. I'll check it out. Thanks!

closed 2025-02-05 by github-actions[bot]