RainerKuemmerle/g2o #336: Levenberg Algorithm for EdgeSE3ProjectXYZOnlyPose class working poorly

Roysorus, 2019-01-02:
Hello,

I have a 3D points cloud that I'm tracking thoughout a video. I'm using g2o for Pose Opitmization  as in [https://github.com/Phylliida/orbslam-windows/blob/master/src/Optimizer.cc](url)
to optimize the pose by minimizing the reprojection error of the points. 
The problem is that, for the same video, g2o seems to work really bad for very big values of the 3D coordinates of the points(big values of the SE3 pose to optimize). However, when I rescaled the 3D coordinates of the points to be really smaller(small values of the SE3 pose to optimize), g2o seems to be working just fine.
I was wondering if there were another way to make Pose Optimization with g2o's implementation of Levenberg algorithm work on big scales?

ferreram, 2019-01-30:
Hi there,
If you have many 3D points which can be considered as being at infinity optimization you can try using an inverse depth parametrization (you can have a look at the example ba_anchored_inverse_depth).

MarceloContreras, 2024-09-11:
I found another solution that may help you. I was having the same problem as soon as I was inserting points w.r.t world position that were triangulated in forward camera poses. Try to set as the pose estimate the pose from world frame w.r.t camera frame (Tcw) just like in ORB-SLAM 2 PoseOptimization (src/Optimizer.cc) method

closed 2025-02-05 by github-actions[bot]