RainerKuemmerle/g2o #337: Image fusion backend optimization problem

woshidaye, 2019-01-22:
I have multiple images, and I can get the transformation matrix (odometer constraint) and closed loop constraint between the two images. I opened this g2o file with g2o_viewer and optimized it. The node information obtained through optimization cannot fusion many pictures very well. Why?

RainerKuemmerle, 2019-02-06:
You would need to provide the graph here to have a look into the problem.
Maybe you run into local minima, maybe some measurements are wrongly set, ...

woshidaye, 2019-02-11:
VERTEX_SE2 0 0 0 0
VERTEX_SE2 1 -1.20803 -0.930108 0.00349066
VERTEX_SE2 2 -0.472017 0.201772 0.00349066
VERTEX_SE2 3 -1.44849 -0.133693 -0.00698132
VERTEX_SE2 4 -0.569271 0.474725 -0.00349066
VERTEX_SE2 5 -1.29152 -0.473624 -0.00349066
VERTEX_SE2 6 -1.42955 0.0844847 -0.00698132
VERTEX_SE2 7 -1.28582 -0.361636 -0.00698132
VERTEX_SE2 8 -0.577007 -0.0931204 -0.00349066
VERTEX_SE2 9 -1.39489 -1.12234 -0.00349066
VERTEX_SE2 10 -0.661996 -1.06246 -0.00349066
VERTEX_SE2 11 -0.632563 -1.61829 -0.00349066
VERTEX_SE2 12 0.143054 -1.23199 -0.00698132
VERTEX_SE2 13 -0.551833 -2.30843 -0.010472
VERTEX_SE2 14 -0.848906 -2.55026 -0.00349066
VERTEX_SE2 15 -0.468046 -1.66366 -0.00698132
VERTEX_SE2 16 -0.407775 -1.91697 0.00349066
VERTEX_SE2 17 -1.25037 -2.73455 0.00349066
VERTEX_SE2 18 -0.464237 -2.31887 0.00349066
VERTEX_SE2 19 -0.372069 -2.38932 -8.5259e-17
VERTEX_SE2 20 -0.3513 -2.25066 -0.00349066
VERTEX_SE2 21 -0.430359 -2.48428 0.00349066
VERTEX_SE2 22 -0.448344 -2.50231 0.00698132
VERTEX_SE2 23 -0.256943 -2.69068 0.00698132
VERTEX_SE2 24 -1.08311 -3.34149 0.00698132
VERTEX_SE2 25 -0.384305 -2.03545 1.41452e-16
VERTEX_SE2 26 -0.244396 -1.86471 0.00349066
VERTEX_SE2 27 -0.195817 -1.7011 -0.0139626
EDGE_SE2 0 1 -1.20803 -0.930108 0.00349066 172.14 0 0 172.14 0 0.101321
EDGE_SE2 0 6 -1.44413 0.137386 -0.0349066 172.14 0 0 172.14 0 0.101321
EDGE_SE2 0 7 -1.32149 -0.311414 -0.0383972 172.14 0 0 172.14 0 0.101321
EDGE_SE2 0 8 -0.597115 0.0200464 -0.101229 172.14 0 0 172.14 0 0.101321
EDGE_SE2 1 2 0.739958 1.1293 3.77852e-17 172.14 0 0 172.14 0 0.101321
EDGE_SE2 1 4 0.640647 1.40942 -0.0139626 172.14 0 0 172.14 0 0.101321
EDGE_SE2 1 5 -0.0842753 0.461218 -0.010472 172.14 0 0 172.14 0 0.101321
EDGE_SE2 1 6 -0.226305 1.03419 -0.020944 172.14 0 0 172.14 0 0.101321
EDGE_SE2 1 7 -0.0921807 0.591825 -0.0244346 172.14 0 0 172.14 0 0.101321
EDGE_SE2 2 3 -0.977636 -0.332054 -0.010472 172.14 0 0 172.14 0 0.101321
EDGE_SE2 2 4 -0.0968234 0.276137 -0.010472 172.14 0 0 172.14 0 0.101321
EDGE_SE2 2 5 -0.817532 -0.67409 -0.00698132 172.14 0 0 172.14 0 0.101321
EDGE_SE2 2 6 -0.952469 -0.115797 -0.010472 172.14 0 0 172.14 0 0.101321
EDGE_SE2 3 4 0.874949 0.614541 0.00349066 172.14 0 0 172.14 0 0.101321
EDGE_SE2 4 5 -0.718933 -0.950864 3.77852e-17 172.14 0 0 172.14 0 0.101321
EDGE_SE2 4 6 -0.853375 -0.40122 3.77852e-17 172.14 0 0 172.14 0 0.101321
EDGE_SE2 5 6 -0.139977 0.557623 -0.00349066 172.14 0 0 172.14 0 0.101321
EDGE_SE2 6 7 0.146835 -0.445107 3.77852e-17 172.14 0 0 172.14 0 0.101321
EDGE_SE2 6 8 0.853449 -0.169927 0.00349066 172.14 0 0 172.14 0 0.101321
EDGE_SE2 6 9 0.0465836 -1.20592 0.00349066 172.14 0 0 172.14 0 0.101321
EDGE_SE2 7 8 0.706925 0.273458 0.00349066 172.14 0 0 172.14 0 0.101321
EDGE_SE2 7 9 -0.10806 -0.755525 3.77852e-17 172.14 0 0 172.14 0 0.101321
EDGE_SE2 8 9 -0.814285 -1.03207 3.77852e-17 172.14 0 0 172.14 0 0.101321
EDGE_SE2 9 10 0.73268 0.0624395 3.77852e-17 172.14 0 0 172.14 0 0.101321
EDGE_SE2 10 11 0.0313722 -0.555723 3.77852e-17 172.14 0 0 172.14 0 0.101321
EDGE_SE2 10 12 0.799632 -0.158513 -0.010472 172.14 0 0 172.14 0 0.101321
EDGE_SE2 11 12 0.774265 0.388998 -0.00349066 172.14 0 0 172.14 0 0.101321
EDGE_SE2 12 13 -0.687356 -1.08126 -0.00349066 172.14 0 0 172.14 0 0.101321
EDGE_SE2 12 27 -0.388515 -0.378137 -0.115192 172.14 0 0 172.14 0 0.101321
EDGE_SE2 13 14 -0.294524 -0.24492 0.00698132 172.14 0 0 172.14 0 0.101321
EDGE_SE2 13 25 0.240073 0.368139 -0.0802851 172.14 0 0 172.14 0 0.101321
EDGE_SE2 13 26 0.286865 0.474786 -0.0349066 172.14 0 0 172.14 0 0.101321
EDGE_SE2 13 27 0.331346 0.653111 -0.0698132 172.14 0 0 172.14 0 0.101321
EDGE_SE2 14 15 0.377762 0.887918 -0.00349066 172.14 0 0 172.14 0 0.101321
EDGE_SE2 14 25 0.429515 0.564946 -0.0523599 172.14 0 0 172.14 0 0.101321
EDGE_SE2 14 26 0.587374 0.716188 -0.0418879 172.14 0 0 172.14 0 0.101321
EDGE_SE2 14 27 0.630624 0.896614 -0.0802851 172.14 0 0 172.14 0 0.101321
EDGE_SE2 15 16 0.0620384 -0.25288 0.010472 172.14 0 0 172.14 0 0.101321
EDGE_SE2 15 17 -0.771495 -1.06956 0.010472 172.14 0 0 172.14 0 0.101321
EDGE_SE2 15 25 0.0602279 -0.327364 -0.0418879 172.14 0 0 172.14 0 0.101321
EDGE_SE2 16 17 -0.845447 -0.814636 3.77852e-17 172.14 0 0 172.14 0 0.101321
EDGE_SE2 16 25 0.0870714 -0.162726 0.0558505 172.14 0 0 172.14 0 0.101321
EDGE_SE2 17 18 0.787582 0.412934 3.77852e-17 172.14 0 0 172.14 0 0.101321
EDGE_SE2 17 19 0.87881 0.345521 -0.00698132 172.14 0 0 172.14 0 0.101321
EDGE_SE2 17 20 0.908702 0.475733 3.77852e-17 172.14 0 0 172.14 0 0.101321
EDGE_SE2 17 23 0.97949 0.0565196 -0.0139626 172.14 0 0 172.14 0 0.101321
EDGE_SE2 17 24 0.141154 -0.577676 -0.0139626 172.14 0 0 172.14 0 0.101321
EDGE_SE2 18 19 0.091921 -0.0707681 -0.00349066 172.14 0 0 172.14 0 0.101321
EDGE_SE2 18 20 0.115387 0.0641114 -0.00349066 172.14 0 0 172.14 0 0.101321
EDGE_SE2 19 20 0.0207692 0.138661 -0.00349066 172.14 0 0 172.14 0 0.101321
EDGE_SE2 19 23 0.0929344 -0.281291 -0.0174533 172.14 0 0 172.14 0 0.101321
EDGE_SE2 19 24 -0.69597 -0.961586 0.0139626 172.14 0 0 172.14 0 0.101321
EDGE_SE2 20 21 -0.0782432 -0.233895 0.00698132 172.14 0 0 172.14 0 0.101321
EDGE_SE2 20 23 0.076803 -0.422174 -0.010472 172.14 0 0 172.14 0 0.101321
EDGE_SE2 20 24 -0.76067 -1.0579 -0.010472 172.14 0 0 172.14 0 0.101321
EDGE_SE2 21 22 -0.0180472 -0.0179664 0.00349066 172.14 0 0 172.14 0 0.101321
EDGE_SE2 21 23 0.160268 -0.195444 -0.010472 172.14 0 0 172.14 0 0.101321
EDGE_SE2 22 23 0.190081 -0.189701 3.77852e-17 172.14 0 0 172.14 0 0.101321
[... cut ...] 0 0 172.14 0 0.101321
EDGE_SE2 23 24 -0.830694 -0.645033 3.77852e-17 172.14 0 0 172.14 0 0.101321
EDGE_SE2 24 25 0.707909 1.30113 -0.00698132 172.14 0 0 172.14 0 0.101321
EDGE_SE2 25 26 0.139909 0.170735 0.00349066 172.14 0 0 172.14 0 0.101321
EDGE_SE2 26 27 0.0491501 0.16344 -0.0174533 172.14 0 0 172.14 0 0.101321

The above is my pose-graph information, where the constraint between the edges is the transformation matrix between the two images, and the node information is the pose of the image.

closed 2025-02-05 by github-actions[bot]