RainerKuemmerle/g2o #338: Edge PoseOnly vs fixed nodes

phildue, 2019-02-04:
Hi, 

I would like to perform pose graph optimization. Hence, all 3D points should be fixed and only R,t should be optimized. I was wondering if that is possibly by simply setting all 3D point nodes as fixed, whereby still using the ProjectXYZ2UV edge?

Thanks!

RainerKuemmerle, 2019-02-06:
In general something along the lines you drafted should be possible.

pegjul, 2019-02-11:
Did you manage to solve this problem since I'm currently working on a similar one?

sjulier, 2019-04-15:
Would the Optimizer::PoseOptimization method of ORBSLAM2 be of help? See:

https://github.com/raulmur/ORB_SLAM2

This uses a somewhat older version of g2o, but fixes all the points in the map and just optimises for the camera pose.

closed 2025-02-05 by github-actions[bot]