RainerKuemmerle/g2o #340: Extracting Jacobian Matrix

B0Bftl, 2019-02-11:
Hello,

I am implementing a Conjugate Gradient linear solver with a special preconditioning. To use it, I need the Jacobian of the error vector, instead of the Hessian, which g2o passes to the solver.
Right now I'm trying to work my way through g2o internals, to understand how to extract the jacobian.

My current idea is to adjust `constructQuadraticForm()` and write the current Jacobian block into the jacobian of the complete problem. Is is advisable, or is there an easier way to accomplish getting the jacobian to a linear solver? Has anyone done this before, or knows what adjustments are needed? 

Thanks for your help

billamiable, 2021-01-27:
Hi, I am also interested in this question. Have you ever succeeded in extracting the jacobian matrix from g2o? Thanks! Appreciate your help!

sjulier, 2021-01-27:
I haven't  done it myself. However, I'd hazard a guess you could build it up by iterating over all the edges, extracting all the Jacobians and using them to populate a sparse block matrix (or another sparse matrix type of your own choosing) whose number of rows and columns is that of the overall system. You'd need to use the Hessian indices from the vertices to specify the block coordinates in the sparse matrix. I think the `buildSystem` method does a similar kind of thing to construct both the Hessian and the Jacobian times the system information matrix.

B0Bftl, 2021-01-28:
Hello,
I made an implementation a couple of years ago. My Implementation of extraction of the jacobian is NOT efficient, but it worked good enough for me. My focus was on the linear solver, so the jacobian was just a requirement i needed.
You can take a look at my fork https://github.com/B0Bftl/g2o
Maybe it's useful to you :)

closed 2025-02-04 by github-actions[bot]