RainerKuemmerle/g2o #345: Different result with g2o_viewer

ceccocats, 2019-03-11:
Im struggling to achieve the same result with g2o_viewer, g2o and c++ code.
with g2o_viewer I can optimize my graph with poses and landmarks like in the picture below
[image]
with g2o executable i tried a lot of configuration but the result is always something like this:
```
$ g2o graph.g2o -solver gn_var_cholmod -guess 
# Using CSparse poseDim -1 landMarkDim -1 blockordering 0
Read input from graph.g2o
Loaded 141 vertices
Loaded 272 edges
# graph is fixed by priors or already fixed vertex
Initial chi2 = 0.001323
int g2o::csparse_extension::cs_cholsolsymb(const cs*, number_t*, const css*, number_t*, int*): cholesky failed!
Cholesky failure, writing debug.txt (Hessian loadable by Octave)`
```
same problem when i optimize from code, I tried reading the g2o_viewer code but still can't tell the difference.
This is how i optimize by code:
```c++
  graph = new g2o::SparseOptimizer();
  std::string g2o_solver_name = "gn_var_cholmod";
  std::cout << "construct solver... " << std::endl;
  g2o::OptimizationAlgorithmFactory* solver_factory = g2o::OptimizationAlgorithmFactory::instance();
  g2o::OptimizationAlgorithmProperty solver_property;
  g2o::OptimizationAlgorithm* solver = solver_factory->construct(g2o_solver_name, solver_property);
  graph->setAlgorithm(solver);
[...]
  graph->initializeOptimization();
  graph->computeInitialGuess();
  graph->computeActiveErrors();
  graph->setVerbose(true);
  int iterations = graph->optimize(10);
```
and the result
```
construct solver... 
# Using CHOLMOD poseDim -1 landMarkDim -1 blockordering 1
done
iteration= 0     chi2= 0.001279  time= 0.00174919        cumTime= 0.00174919     edges= 272      schur= 0
iteration= 1     chi2= 0.452561  time= 0.00113027        cumTime= 0.00287946     edges= 272      schur= 0
iteration= 2     chi2= 2489.505225       time= 0.0010361         cumTime= 0.00391557     edges= 272      schur= 0
iteration= 3     chi2= 32051.708081      time= 0.00088962        cumTime= 0.00480519     edges= 272      schur= 0
iteration= 4     chi2= 1800908828.515487         time= 0.000856505       cumTime= 0.00566169     edges= 272      schur= 0
iteration= 5     chi2= 1551661108644.207275      time= 0.000841392       cumTime= 0.00650308     edges= 272      schur= 0
iteration= 6     chi2= 504233743410214.875000    time= 0.00085931        cumTime= 0.00736239     edges= 272      schur= 0
iteration= 7     chi2= 557693785548149056.000000         time= 0.000843036       cumTime= 0.00820543     edges= 272      schur= 0
iteration= 8     chi2= 210290525349022138368.000000      time= 0.000840777       cumTime= 0.00904621     edges= 272      schur= 0
iteration= 9     chi2= 264824788336802848047104.000000   time= 0.000839956       cumTime= 0.00988616     edges= 272      schur= 0
```

It must be something about the _solverProperties_ but Im really stuck, does anyone can tell me what's wrong?

Thanks,
Francesco

<details> 
 <summary>here is my not optimized graph</summary>

```
PARAMS_SE3OFFSET 0 0 0 0 0 0 0 1 
VERTEX_SE3:QUAT 0 0 0 0 0 0 0 1 
FIX 0
VERTEX_SE3:QUAT 1 1.00911 0.0468775 0 0 0 0.0230096 0.999735 
VERTEX_TRACKXYZ 2 2.55092 -0.759609 0.0001 
VERTEX_TRACKXYZ 3 6.79318 -0.166535 0.0001 
VERTEX_TRACKXYZ 4 6.78116 0.744037 0.0001 
VERTEX_SE3:QUAT 5 1.62431 0.0692605 0 0 0 0.0157094 0.999877 
VERTEX_SE3:QUAT 6 1.81794 0.0740222 0 0 0 0.0156703 0.999877 
VERTEX_SE3:QUAT 7 2.75567 0.07309 0 0 0 0.013333 0.999911 
VERTEX_SE3:QUAT 8 2.88904 0.076758 0 0 0 0.0113753 0.999935 
VERTEX_SE3:QUAT 9 3.81439 0.0877005 0 0 0 0.0125067 0.999922 
[... 379 lines cut ...]
EDGE_SE3_TRACKXYZ 129 123 0 1.15885 -2.51374 0.0001 0.0399982 0 0 0 0.0399982 0 
EDGE_SE3:QUAT 129 128 -0.0911856 0.020231 0 0 0 0.0307703 0.999526 4.75604 0 0 0 0 0 4.75604 0 0 0 0 4.75604 0 0 0 357.078 0 0 357.078 0 357.078 
EDGE_SE3_TRACKXYZ 130 123 0 1.18734 -2.46161 0.0001 0.0399982 0 0 0 0.0399982 0 
EDGE_SE3:QUAT 130 129 -0.0463195 0.0110986 0 0 0 0.0173078 0.99985 7.01798 0 0 0 0 0 7.01798 0 0 0 0 7.01798 0 0 0 370.533 0 0 370.533 0 370.533 
EDGE_SE3_TRACKXYZ 131 123 0 1.24738 -2.37088 0.0001 0.0399982 0 0 0 0.0399982 0 
EDGE_SE3:QUAT 131 130 -0.0855987 0.0221529 0 0 0 0.0324354 0.999474 3.59651 0 0 0 0 0 3.59651 0 0 0 0 3.59651 0 0 0 344.562 0 0 344.562 0 344.562 
EDGE_SE3_TRACKXYZ 132 123 0 1.29754 -2.29339 0.0001 0.0399982 0 0 0 0.0399982 0 
EDGE_SE3:QUAT 132 131 -0.0607367 0.0193992 0 0 0 0.0225349 0.999746 6.61532 0 0 0 0 0 6.61532 0 0 0 0 6.61532 0 0 0 368.755 0 0 368.755 0 368.755 
EDGE_SE3_TRACKXYZ 133 123 0 1.35871 -2.1744 0.0001 0.0399982 0 0 0 0.0399982 0 
EDGE_SE3:QUAT 133 132 -0.0942783 0.0221438 0 0 0 0.0348439 0.999393 4.37831 0 0 0 0 0 4.37831 0 0 0 0 4.37831 0 0 0 353.641 0 0 353.641 0 353.641 
EDGE_SE3_TRACKXYZ 134 123 0 1.39948 -2.09447 0.0001 0.0399982 0 0 0 0.0399982 0 
EDGE_SE3:QUAT 134 133 -0.0646808 0.0151958 0 0 0 0.0250603 0.999686 4.29168 0 0 0 0 0 4.29168 0 0 0 0 4.29168 0 0 0 352.778 0 0 352.778 0 352.778 
EDGE_SE3_TRACKXYZ 135 123 0 1.49359 -1.91171 0.0001 0.0399982 0 0 0 0.0399982 0 
EDGE_SE3:QUAT 135 134 -0.106597 0.0261106 0 0 0 0.0482502 0.998835 1.92396 0 0 0 0 0 1.92396 0 0 0 0 1.92396 0 0 0 306.284 0 0 306.284 0 306.284 
EDGE_SE3_TRACKXYZ 136 123 0 1.52341 -1.84804 0.0001 0.0399982 0 0 0 0.0399982 0 
EDGE_SE3:QUAT 136 135 -0.0465021 0.0131168 0 0 0 0.0197106 0.999806 6.05743 0 0 0 0 0 6.05743 0 0 0 0 6.05743 0 0 0 365.935 0 0 365.935 0 365.935 
EDGE_SE3_TRACKXYZ 137 123 0 1.54353 -1.76006 0.0001 0.0399982 0 0 0 0.0399982 0 
EDGE_SE3:QUAT 137 136 -0.0444441 0.0157475 0 0 0 0.0196228 0.999807 5.98031 0 0 0 0 0 5.98031 0 0 0 0 5.98031 0 0 0 365.508 0 0 365.508 0 365.508 
EDGE_SE3_TRACKXYZ 138 123 0 1.59389 -1.60786 0.0001 0.0399982 0 0 0 0.0399982 0 
EDGE_SE3:QUAT 138 137 -0.091064 0.0237393 0 0 0 0.0393431 0.999226 1.6382 0 0 0 0 0 1.6382 0 0 0 0 1.6382 0 0 0 294.031 0 0 294.031 0 294.031 
EDGE_SE3_TRACKXYZ 139 123 0 1.61261 -1.517 0.0001 0.0399982 0 0 0 0.0399982 0 
EDGE_SE3:QUAT 139 138 -0.0428972 0.0153551 0 0 0 0.0202514 0.999795 1.7047 0 0 0 0 0 1.7047 0 0 0 0 1.7047 0 0 0 297.153 0 0 297.153 0 297.153 
EDGE_SE3_TRACKXYZ 140 123 0 1.64433 -1.35548 0.0001 0.0399982 0 0 0 0.0399982 0 
EDGE_SE3:QUAT 140 139 -0.090579 0.0219483 0 0 0 0.0415071 0.999138 0.292493 0 0 0 0 0 0.292493 0 0 0 0 0.292493 0 0 0 131.313 0 0 131.313 0 131.313 
```

feroze, 2019-03-11:
Your g2o executable is using the wrong solver - you have given the options after the input file and not before. Usage: g2o [options] [graph-input]

`└> λ g2o -solver gn_var_cholmod -v graph.g2o
# Used Compiler: GNU /usr/bin/c++
# Using CHOLMOD poseDim -1 landMarkDim -1 blockordering 1
Read input from graph.g2o
Loaded 141 vertices
Loaded 272 edges
# graph is fixed by priors or already fixed vertex
Initial chi2 = 0.001323
iteration= 0	 chi2= 0.000288	 time= 0.00126068	 cumTime= 0.00126068	 edges= 272	 schur= 0
iteration= 1	 chi2= 0.000288	 time= 0.000674143	 cumTime= 0.00193483	 edges= 272	 schur= 0
iteration= 2	 chi2= 0.000288	 time= 0.000736182	 cumTime= 0.00267101	 edges= 272	 schur= 0
iteration= 3	 chi2= 0.000288	 time= 0.000633675	 cumTime= 0.00330468	 edges= 272	 schur= 0
iteration= 4	 chi2= 0.000288	 time= 0.000630234	 cumTime= 0.00393492	 edges= 272	 schur= 0
`

ceccocats, 2019-03-12:
thank you,
I still have problem with the optimizer in the c++ code.
I noticed that if I save the graph and reload it, like this:
```
for(int i=0; i<2; i++) {
  save("graph.g2o");
  graph->initializeOptimization();
  graph->computeActiveErrors();
  graph->computeInitialGuess();
  int iterations = graph->optimize(5);
  load("graph.g2o");
}
```
the result is correct once i reloaded the saved graph:
```
# graph is fixed by priors or already fixed vertex
iteration= -1    chi2= 0.00132569        time= 0.0       cumTime= 0.0    (using initial guess from spanning tree)
iteration= 0     chi2= 0.001289  time= 0.00184881        cumTime= 0.00184881     edges= 272      schur= 0
iteration= 1     chi2= 0.477088  time= 0.00101728        cumTime= 0.00286609     edges= 272      schur= 0
iteration= 2     chi2= 3360.335133       time= 0.000905111       cumTime= 0.0037712      edges= 272      schur= 0
iteration= 3     chi2= 37564.002301      time= 0.000873246       cumTime= 0.00464445     edges= 272      schur= 0
iteration= 4     chi2= 1045285.693482    time= 0.00146753        cumTime= 0.00611198     edges= 272      schur= 0
virtual bool g2o::OptimizableGraph::load(std::istream&, bool): Parameter of type:PARAMS_SE3OFFSET id:0 already defined
iteration= -1    chi2= 0.00132337        time= 0.0       cumTime= 0.0    (using initial guess from spanning tree)
iteration= 0     chi2= 0.000288  time= 0.00107288        cumTime= 0.00107288     edges= 272      schur= 0
iteration= 1     chi2= 0.000288  time= 0.00112479        cumTime= 0.00219767     edges= 272      schur= 0
iteration= 2     chi2= 0.000288  time= 0.00103262        cumTime= 0.0032303      edges= 272      schur= 0
iteration= 3     chi2= 0.000288  time= 0.00108759        cumTime= 0.00431788     edges= 272      schur= 0
iteration= 4     chi2= 0.000288  time= 0.000894814       cumTime= 0.0052127      edges= 272      schur= 0
```
could be a problem in the graph construction?

closed 2022-02-03 by ceccocats