RainerKuemmerle/g2o #346: optimize fail

sylin211, 2019-03-13:
Hi! I am using optimize() function to optimize a graph in which there is only one pose vertex and the edges are re-projection errors, but I find that it only optimize once even though I have set iteration times for 50, like, optimizer.optimize(50). I set verbose true and it just output one-line info:
iteration= 0 chi2= 5674.536900 time= 0.000129743 cumTime= 0.000129743 edges= 427 schur= 0 lambda= 1880.815151 levenbergIter= 1
I have check that initializeOptimization() work normally. Why would this happen?

sylin211, 2019-03-13:
I've solved it. It turns out that there is something wrong in the computeError() function re-defined by myself. After I correct it, the procedure work normally now.

closed 2019-03-13 by sylin211

kaixinbear, 2020-03-05:
Same issue as yours.Could you plz tell me what did you re-defined in computeError()?