RainerKuemmerle/g2o #347: How to deal with '[Setjac] infinite jac' Issue

kylxyz, 2019-03-19:
Hi, Rainer

I met a problem when I was using **[rtabmap](https://github.com/introlab/rtabmap)**  ,
with some  image, g2o will abort the rtabmap processing and throw a message:

 [[Setjac] infinite jac](https://github.com/introlab/rtabmap/issues/360#issuecomment-468103937)

What I expected is to keep the rtabmap processing not to abort,
is there any suggestion to overcome this? Thank you.

Cheers

AnselmC, 2020-08-11:
Hi, I'm also encountering the  `infinite jac` issue. Did you ever figure out what caused this/how to solve it?
Thanks!

jinshengye-git, 2020-08-11:
Hi, I did not figure it out. I just modify the   `g2o/g2o/types/sba/types_sba.cpp ` to walk around the abort(); 

```
// g2o - General Graph Optimization
// Copyright (C) 2011 Kurt Konolige
// All rights reserved.
//
// Redistribution and use in source and binary forms, with or without
// modification, are permitted provided that the following conditions are
// met:
//
// * Redistributions of source code must retain the above copyright notice,
//   this list of conditions and the following disclaimer.
// * Redistributions in binary form must reproduce the above copyright
//   notice, this list of conditions and the following disclaimer in the
//   documentation and/or other materials provided with the distribution.
//
// THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS
// IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED
// TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A
// PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT
// HOLDER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL,
// SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED
// TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR
// PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF
// LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING
// NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS
// SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.

#include "types_sba.h"
#include <iostream>

#include "g2o/core/factory.h"
#include "g2o/stuff/macros.h"

namespace g2o {

  using namespace std;

  G2O_REGISTER_TYPE_GROUP(sba);

  G2O_REGISTER_TYPE(VERTEX_CAM, VertexCam);
  G2O_REGISTER_TYPE(VERTEX_XYZ, VertexSBAPointXYZ);
  G2O_REGISTER_TYPE(VERTEX_INTRINSICS, VertexIntrinsics);

  G2O_REGISTER_TYPE(EDGE_PROJECT_P2MC, EdgeProjectP2MC);
  G2O_REGISTER_TYPE(EDGE_PROJECT_P2MC_INTRINSICS, EdgeProjectP2MC_Intrinsics);
  G2O_REGISTER_TYPE(EDGE_PROJECT_P2SC, EdgeProjectP2SC);
  G2O_REGISTER_TYPE(EDGE_CAM, EdgeSBACam);
  G2O_REGISTER_TYPE(EDGE_SCALE, EdgeSBAScale);

  // constructor
  VertexIntrinsics::VertexIntrinsics() 
  {
    _estimate << cst(1.), cst(1.), cst(.5), cst(.5), cst(.1);
  }

  bool VertexIntrinsics::read(std::istream& is)
  {
    for (int i=0; i<5; i++)
      is >> _estimate[i];
    return true;
  }

  bool VertexIntrinsics::write(std::ostream& os) const
  {
    for (int i=0; i<5; i++)
      os << _estimate[i] << " ";
    return os.good();
  }
[... 490 lines cut ...]
      SE3Quat delta = (v1->estimate().inverse()*v2->estimate());
      number_t norm =  delta.translation().norm();
      number_t alpha = _measurement/norm;
      delta.setTranslation(delta.translation()*alpha);
      v2->setEstimate(v1->estimate()*delta);
    } else {
      SE3Quat delta = (v2->estimate().inverse()*v1->estimate());
      number_t norm =  delta.translation().norm();
      number_t alpha = _measurement/norm;
      delta.setTranslation(delta.translation()*alpha);
      v1->setEstimate(v2->estimate()*delta);
    }
  }

  bool EdgeSBACam::setMeasurementFromState()
  {
    const VertexCam* v1 = dynamic_cast<const VertexCam*>(_vertices[0]);
    const VertexCam* v2 = dynamic_cast<const VertexCam*>(_vertices[1]);
    _measurement = (v1->estimate().inverse()*v2->estimate());
    _inverseMeasurement = _measurement.inverse();
    return true;
  }

} // end namespace
```

matlabbe, 2020-09-15:
We made a new commit to make sure valid 3D points are passed to g2o optimization: https://github.com/introlab/rtabmap/commit/4e6e40495150ea6024dc5b557359523c9cd63a62, this may fix the problem.

jinshengye-git, 2020-09-16:
Thank you matlabbe

closed 2020-09-16 by jinshengye-git

matlabbe, 2023-10-06:
Just find out that before https://github.com/RainerKuemmerle/g2o/commit/4145b3ead71c2f54f191980e65a277552c09e29a, if cholesky failed for some reason, it will crash with `[SetJac] infinite jac` error:
```
int g2o::csparse_extension::cs_cholsolsymb(const cs_di*, number_t*, const cs_dis*, number_t*, int*): cholesky failed!
Cholesky failure, writing debug.txt (Hessian loadable by Octave)
[SetJac] infinite jac
```