RainerKuemmerle/g2o #348: simple snippets got core dumped

jinfagang, 2019-04-04:
here is a simple program:

```
#include <iostream>
// for opencv
#include <opencv2/core/core.hpp>
#include <opencv2/highgui/highgui.hpp>
#include <opencv2/features2d/features2d.hpp>
#include <boost/concept_check.hpp>
// for g2o
#include <g2o/core/sparse_optimizer.h>
#include <g2o/core/block_solver.h>
#include <g2o/core/robust_kernel.h>
#include <g2o/core/robust_kernel_impl.h>
#include <g2o/core/optimization_algorithm_levenberg.h>
#include <g2o/solvers/cholmod/linear_solver_cholmod.h>
#include <g2o/types/slam3d/se3quat.h>
#include <g2o/types/sba/types_six_dof_expmap.h>

using namespace std;
using namespace cv;

double cx = 325.5;
double cy = 253.5;
double fx = 518.0;
double fy = 519.0;

int findCorrespondingPoints(const cv::Mat &img1,
                            const cv::Mat &img2,
                            vector<cv::Point2f> &points1,
                            vector<cv::Point2f> &points2) {
  Ptr<ORB> orb = ORB::create(4000, 1.2f, 8, 31, 0, 2, ORB::HARRIS_SCORE, 31, 20);
  vector<cv::KeyPoint> kp1, kp2;
  cv::Mat desp1, desp2;
  orb->detectAndCompute(img1, Mat(), kp1, desp1);
  orb->detectAndCompute(img2, Mat(), kp2, desp2);

  cout << "分别找到了" << kp1.size() << "和" << kp2.size() << "个特征点" << endl;
  cv::Ptr<cv::DescriptorMatcher> matcher = cv::DescriptorMatcher::create("BruteForce-Hamming");

  double knn_match_ratio = 0.8;
  vector<vector<cv::DMatch> > matches_knn;
  matcher->knnMatch(desp1, desp2, matches_knn, 2);
  vector<cv::DMatch> matches;
  for (size_t i = 0; i < matches_knn.size(); i++) {
    if (matches_knn[i][0].distance < knn_match_ratio * matches_knn[i][1].distance)
      matches.push_back(matches_knn[i][0]);
  }

  if (matches.size() <= 20) //匹配点太少
    return false;

  for (auto m:matches) {
    points1.push_back(kp1[m.queryIdx].pt);
    points2.push_back(kp2[m.trainIdx].pt);
  }

  return true;
}

int main(int argc, char **argv) {

  if (argc != 3) {
    cout << "Usage: a img1 img2\n";
  }

  cv::Mat img1 = cv::imread(argv[1]);
  cv::Mat img2 = cv::imread(argv[1]);

  vector<cv::Point2f> pts1, pts2;
[... 297 lines cut ...]
7f3fa88c4000-7f3fa88df000 r-xp 00000000 103:08 918717                    /usr/local/lib/libg2o_stuff.so
7f3fa88df000-7f3fa8adf000 ---p 0001b000 103:08 918717                    /usr/local/lib/libg2o_stuff.so
7f3fa8adf000-7f3fa8ae0000 r--p 0001b000 103:08 918717                    /usr/local/lib/libg2o_stuff.so
7f3fa8ae0000-7f3fa8ae1000 rw-p 0001c000 103:08 918717                    /usr/local/lib/libg2o_stuff.so
7f3fa8ae1000-7f3fa8ae2000 rw-p 00000000 00:00 0 
7f3fa8ae2000-7f3fa8b3f000 r-xp 00000000 103:08 918719                    /usr/local/lib/libg2o_core.so
7f3fa8b3f000-7f3fa8d3e000 ---p 0005d000 103:08 918719                    /usr/local/lib/libg2o_core.so
7f3fa8d3e000-7f3fa8d40000 r--p 0005c000 103:08 918719                    /usr/local/lib/libg2o_core.so
7f3fa8d40000-7f3fa8d41000 rw-p 0005e000 103:08 918719                    /usr/local/lib/libg2o_core.so
7f3fa8d41000-7f3fa8d67000 r-xp 00000000 103:08 1316495                   /lib/x86_64-linux-gnu/ld-2.23.so
7f3fa8f06000-7f3fa8f18000 rw-p 00000000 00:00 0 
7f3fa8f64000-7f3fa8f66000 rw-p 00000000 00:00 0 
7f3fa8f66000-7f3fa8f67000 r--p 00025000 103:08 1316495                   /lib/x86_64-linux-gnu/ld-2.23.so
7f3fa8f67000-7f3fa8f68000 rw-p 00026000 103:08 1316495                   /lib/x86_64-linux-gnu/ld-2.23.so
7f3fa8f68000-7f3fa8f69000 rw-p 00000000 00:00 0 
7fff11f01000-7fff11f21000 rwxp 00000000 00:00 0                          [stack]
7fff11f21000-7fff11f22000 rw-p 00000000 00:00 0 
7fff11f62000-7fff11f65000 r--p 00000000 00:00 0                          [vvar]
7fff11f65000-7fff11f67000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
[1]    11714 abort (core dumped)  ./ba_example

```

Anybody know why?

sjulier, 2019-04-13:
Can you build in debug and try valgrind? That's typically pretty good at spotting things like memory corruption issues and initialized variables.

jinfagang, 2019-04-15:
@sjulier  I haven't managed built which this repo contains commit g2o version, but with newest g2o clone from master branch, it's easy to reveal this bug. And currently still have no idea what's going on..

sjulier, 2019-04-15:
I don't have a machine in front of me to compile and test your code. valgrind will give a deep dive of what the code is doing.

YangSiri, 2019-11-09:
Hi, @jinfagang , I'm facing the same problem. Did you solve it? 

The code you mentioned here is like a 'textbook'. My problem comes from when i new a vertex or a edge  like _[g2o::VertexSE3Expmap *v = new g2o::VertexSE3Expmap();]_ which is defined in g2o, the whole project collapsed as SIGART no matter what is in your cpp. However, using custom vertex or edge is okay. Do you have any comments on this? 

Looking forward to generous help.

closed 2025-02-04 by github-actions[bot]