RainerKuemmerle/g2o #350: Homebrew formula uses old archive from 2017

jackxujh, 2019-04-05:
First of all, Homebrew formula was migrated. To help with the situation, I made a pull request to update the proper command for installing *g2o* on Mac: #349.

Then when I install *g2o* with the migrated formula, it seems that the formula has not been updated, since it was archived when the Homebrew team made the migration. The current formula that Homebrew uses is at https://github.com/brewsci/homebrew-science/blob/master/Formula/g2o.rb, which uses a commit in Summer 2017. 

It would be great if the formula is maintained so that all new features and enhancements are available for Mac users; and at the same time potential issues can be eliminated.

jackxujh, 2019-04-05:
This is the installation process at this moment.

    $ brew tap brewsci/science
    
    ...
   
    $ brew install g2o
    Updating Homebrew...
    ==> Installing g2o from brewsci/science
    ==> Downloading https://homebrew.bintray.com/bottles-science/g2o-2017-07-30.sierra.bottle.tar.gz
    curl: (22) The requested URL returned error: 404 Not Found
    Error: Failed to download resource "g2o"
    Download failed: https://homebrew.bintray.com/bottles-science/g2o-2017-07-30.sierra.bottle.tar.gz
    Warning: Bottle installation failed: building from source.
    ==> Downloading https://github.com/RainerKuemmerle/g2o/archive/20170730_git.tar.gz
    ==> Downloading from https://codeload.github.com/RainerKuemmerle/g2o/tar.gz/20170730_git
    ######################################################################## 100.0%
    ==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFI
    ==> make install
    🍺  /usr/local/Cellar/g2o/2017-07-30: 387 files, 6.3MB, built in 6 minutes 41 seconds

RainerKuemmerle, 2020-10-24:
Not a Mac OS user. Contributions are always welcome.

closed 2020-10-24 by RainerKuemmerle

iamkrs9, 2021-05-17:
> This is the installation process at this moment.
>
> ```
> $ brew tap brewsci/science
>
> ...
>
> $ brew install g2o
> Updating Homebrew...
> ==> Installing g2o from brewsci/science
> ==> Downloading [...]

homebrew doesn't complete this by itself, the CMake and make install commands are to be typed by the user