RainerKuemmerle/g2o #351: Eigen assertion issue

wang9964, 2019-04-11:
When I use "g2o::VertexSim3Expmap *VSim3 = new g2o::VertexSim3Expmap();"  with debug. I get the fail:
Assertion failed: (internal::UIntPtr(array) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****", file <private...>\openmvg\src\third_party\eigen\eigen\src/Core/DenseStorage.h, line 128
here my setup:
Windows 10
Visual Studio 2015 (Community Edition)
How can I solve it? Thanks everyone.

sjulier, 2019-04-13:
What's the full context of the code?

wang9964, 2019-04-15:
It's my code:
for (size_t i = 0, iend = vKFs.size(); i < iend; i++)
    {
        std::shared_ptr<Frame> pKF = vKFs[i];

        g2o::VertexSim3Expmap *VSim3 = new g2o::VertexSim3Expmap();

        const int nIDi = pKF->mViewid;

        FrameAndPose::const_iterator it = CorrectedSim3.find(pKF->mViewid);

        // pose with sim3
        if (it != CorrectedSim3.end())
        {
            vScw[nIDi] = it->second;
            VSim3->setEstimate(it->second);
        }
        else // pose without sim3 (pose itself)
        {
            Eigen::Matrix<double, 3, 3> Rcw;
            pKF->GetRotation(Rcw);
            Eigen::Matrix<double, 3, 1> tcw;
            pKF->GetTranslation(tcw);
            g2o::Sim3 Siw(Rcw, tcw, 1.0);
            vScw[nIDi] = Siw;
            VSim3->setEstimate(Siw);
        }

        if (pKF->mViewid == pLoopKF->mViewid)
            VSim3->setFixed(true);

        VSim3->setId(nIDi);
        VSim3->setMarginalized(false);
        VSim3->_fix_scale = bFixScale;

        optimizer.addVertex(VSim3);
    }

wang9964, 2019-04-15:
It`s my code,@sjulier :
`for (size_t i = 0, iend = vKFs.size(); i < iend; i++)
    {
        std::shared_ptr<Frame> pKF = vKFs[i];

        g2o::VertexSim3Expmap *VSim3 = new g2o::VertexSim3Expmap();

        const int nIDi = pKF->mViewid;

        FrameAndPose::const_iterator it = CorrectedSim3.find(pKF->mViewid);

        // pose with sim3
        if (it != CorrectedSim3.end())
        {
            vScw[nIDi] = it->second;
            VSim3->setEstimate(it->second);
        }
        else // pose without sim3 (pose itself)
        {
            Eigen::Matrix<double, 3, 3> Rcw;
            pKF->GetRotation(Rcw);
            Eigen::Matrix<double, 3, 1> tcw;
            pKF->GetTranslation(tcw);
            g2o::Sim3 Siw(Rcw, tcw, 1.0);
            vScw[nIDi] = Siw;
            VSim3->setEstimate(Siw);
        }

        if (pKF->mViewid == pLoopKF->mViewid)
            VSim3->setFixed(true);

        VSim3->setId(nIDi);
        VSim3->setMarginalized(false);
        VSim3->_fix_scale = bFixScale;

        optimizer.addVertex(VSim3);
    }`

sjulier, 2019-04-17:
Hmm... I had a quick look and the only thing I can see is this:

https://github.com/mp3guy/ElasticFusion/issues/93

Can you confirm if you are compiling 32-bit or 64-bits?

wang9964, 2019-04-17:
64-bits. The bug appears on windosw, but it`s fine on linux. Maybe it's a compiler problem.

sjulier, 2019-04-17:
Hmm... We've had problems with the MS compiler. For example, we couldn't get it to compile GTSAM (we use it for teaching because of the Matlab bindings) because it seemed to mess up eigen's templates and the eigen assert failed (as far as I can tell, it thought 1 > 3).

Is it possible to try 2017 to see if that fixes the problem?

jlblancoc, 2019-04-18:
On revisiting this, I finally found a solution: simply defining  `EIGEN_NO_STATIC_ASSERT` when using MSVC **and** building DLLs.

wang9964, 2019-04-19:
@jlblancoc  Thank you so much. I will try it.

closed 2019-08-08 by wang9964