RainerKuemmerle/g2o #352: Mac Mojave compile error

QinZiwen, 2019-04-13:
On my Mac Mojave 10.14.4, complie errors like:
```
-- Compiling on OSX
-- Found CHOLMOD and its dependencies
-- Compiling with OpenGL support
CMake Error at /usr/local/lib/cmake/Qt5Core/Qt5CoreConfig.cmake:15 (message):
  The imported target "Qt5::Core" references the file

     "/usr/local/.//mkspecs/macx-clang"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/local/lib/cmake/Qt5Core/Qt5CoreConfigExtras.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /usr/local/lib/cmake/Qt5Core/Qt5CoreConfigExtras.cmake:50 (_qt5_Core_check_file_exists)
  /usr/local/lib/cmake/Qt5Core/Qt5CoreConfig.cmake:166 (include)
  /usr/local/lib/cmake/Qt5/Qt5Config.cmake:28 (find_package)
  cmake_modules/FindQGLViewer.cmake:1 (find_package)
  CMakeLists.txt:174 (find_package)
```

but Mac dose not support OpenGL.

sjulier, 2019-04-13:
How is Qt5 installed? It works for me.

QinZiwen, 2019-04-15:
Yes, Thank you!

sjulier, 2019-04-15:
Please can you clarify?

QinZiwen, 2019-04-16:
After installing Qt5, the program will work normally.

closed 2019-04-20 by RainerKuemmerle

michaelgruner, 2020-02-17:
I had Qt properly installed via Homebrew and still got this. For anyone still battling with the problem I had to manually link these files:

```
sudo ln -s /usr/local/Cellar/qt/5.13.2/mkspecs/ /usr/local/mkspecs
sudo ln -s /usr/local/Cellar/qt5/5.13.2/plugins /usr/local/plugins
```

Make sure you replace your correct Qt version.