RainerKuemmerle/g2o #358: g2o_viewer cannot load .g2o file success

ZJU-kingdong, 2019-05-16:
I just run ./tutorial_slam2d and it output two files: "tutorial_before.g2o" and "tutorial_after.g2o"

when i want to see my result by g2o_viewer,however,it seems g2o_viewer cannot load the file success.

the output like this:
[image]

ZJU-kingdong, 2019-05-16:
After i delete all “TUTORIAL_”,g2o_viewer works good,but i still don't know why g2o_viewer unknow these types

closed 2019-05-16 by ZJU-kingdong

senselessDev, 2022-01-09:
Just for reference, the same questions were answered in the following issues:
https://github.com/RainerKuemmerle/g2o/issues/52#issuecomment-58566495
https://github.com/RainerKuemmerle/g2o/issues/169#issuecomment-293024642

wsp666, 2022-01-30:
> but why I encounter the problem 'unknown type: EDGE_SE3:QUAT'.
``` 
[31mvirtual bool g2o::OptimizableGraph::load(std::istream&) unknown type: VERTEX_SE3:QUAT[0m
[31mvirtual bool g2o::OptimizableGraph::load(std::istream&) unknown type: EDGE_SE3:QUAT[0m
loaded sphere.g2o with 0 vertices and 0 measurements
``` 
my g2o file is:
``` 
VERTEX_SE3 : QUAT 0 -0.125664 -1.53894e-17 99.9999 0.706662 4.32706e-17 0.707551 -4.3325e-17 
FIX 0
VERTEX_SE3 : QUAT 1 -0.2447 -0.0287003 99.9832 0.704617 0.0231823 0.70586 -0.0688426 
VERTEX_SE3 : QUAT 2 -0.370162 -0.0884744 99.9776 0.697957 0.0591073 0.703232 -0.121767 
VERTEX_SE3 : QUAT 3 -0.478226 -0.175864 99.9885 0.68874 0.100025 0.698127 -0.16808 
VERTEX_SE3 : QUAT 4 -0.56488 -0.291548 99.975 0.676789 0.139138 0.689709 -0.216564 
VERTEX_SE3 : QUAT 5 -0.624303 -0.425521 99.9516 0.661416 0.182572 0.679895 -0.258725 
......
EDGE_SE3 : QUAT 2350 2399 -0.221166 -1.57428 -6.25248 0.0722386 0.0252205 -0.00972285 0.997021 10000 0 0 0 0 0 10000 0 0 0 0 10000 0 0 0 40000 0 0 40000 0 40000 
EDGE_SE3 : QUAT 2350 2400 -0.215575 -0.0051028 -6.29581 -0.00802616 0.0373789 0.00768615 0.999239 10000 0 0 0 0 0 10000 0 0 0 0 10000 0 0 0 40000 0 0 40000 0 40000 
EDGE_SE3 : QUAT 2350 2401 -0.242948 1.5426 -6.49575 -0.0606068 0.0339903 0.0106538 0.997526 10000 0 0 0 0 0 10000 0 0 0 0 10000 0 0 0 40000 0 0 40000 0 40000 
EDGE_SE3 : QUAT 2351 2400 -0.199069 -1.55932 -6.25306 0.0534525 0.0233561 -0.00671425 0.998275 10000 0 0 0 0 0 10000 0 0 0 0 10000 0 0 0 40000 0 0 40000 0 40000 
EDGE_SE3 : QUAT 2351 2401 -0.202152 -0.00510787 -6.26985 -0.00533501 0.0418398 0.004802 0.999099 10000 0 0 0 0 0 10000 0 0 0 0 10000 0 0 0 40000 0 0 40000 0 40000 
EDGE_SE3 : QUAT 2351 2402 -0.206978 1.53837 -6.48552 -0.0736597 0.0285196 0.0150613 0.996762 10000 0 0 0 0 0 10000 0 0 0 0 10000 0 0 0 40000 0 0 40000 0 40000 
EDGE_SE3 : QUAT 2352 2401 -0.212408 -1.53492 -6.24696 0.0591696 0.0380121 -0.00935306 0.99748 10000 0 0 0 0 0 10000 0 0 0 0 10000 0 0 0 40000 0 0 40000 0 40000 
```

RainerKuemmerle, 2022-01-30:
see #565