RainerKuemmerle/g2o #359: /usr/bin/ld: 找不到 -lcore /usr/bin/ld: 找不到 -lsolver_csparse

YL13609244502, 2019-05-29:
Scanning dependencies of target tutorial_slam2d_library
[ 10%] Building CXX object CMakeFiles/tutorial_slam2d_library.dir/edge_se2_pointxy.cpp.o
[ 20%] Building CXX object CMakeFiles/tutorial_slam2d_library.dir/edge_se2.cpp.o
[ 30%] Building CXX object CMakeFiles/tutorial_slam2d_library.dir/vertex_se2.cpp.o
[ 40%] Building CXX object CMakeFiles/tutorial_slam2d_library.dir/vertex_point_xy.cpp.o
[ 50%] Building CXX object CMakeFiles/tutorial_slam2d_library.dir/parameter_se2_offset.cpp.o
[ 60%] Building CXX object CMakeFiles/tutorial_slam2d_library.dir/types_tutorial_slam2d.cpp.o
[ 70%] Building CXX object CMakeFiles/tutorial_slam2d_library.dir/simulator.cpp.o
[ 80%] Linking CXX static library libtutorial_slam2d.a
[ 80%] Built target tutorial_slam2d_library
Scanning dependencies of target tutorial_slam2d
[ 90%] Building CXX object CMakeFiles/tutorial_slam2d.dir/tutorial_slam2d.cpp.o
[100%] Linking CXX executable tutorial_slam2d
/usr/bin/ld: 找不到 -lcore
/usr/bin/ld: 找不到 -lsolver_csparse
collect2: error: ld returned 1 exit status
CMakeFiles/tutorial_slam2d.dir/build.make:95: recipe for target 'tutorial_slam2d' failed
make[2]: *** [tutorial_slam2d] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/tutorial_slam2d.dir/all' failed
make[1]: *** [CMakeFiles/tutorial_slam2d.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

WLwind, 2019-09-25:
沙发！😂

RainFromCN, 2021-03-06:
解决了吗，我也遇到这个问题了

closed 2025-02-04 by github-actions[bot]