RainerKuemmerle/g2o #362: /usr/bin/ld: cannot find  -lopengl_helper

foolishair, 2019-06-14:
[ 41%] Built target stuff
[ 41%] Built target core
[ 41%] Built target g2o_cli_library
[ 41%] Built target g2o_cli_application
[ 41%] Built target g2o_hierarchical_library
[ 41%] Built target types_slam3d
[ 41%] Built target g2o_hierarchical_application
[ 41%] Linking CXX shared library ../../../../lib/libg2o_viewer.so
/usr/bin/ld: 找不到 -lopengl_helper
collect2: error: ld returned 1 exit status
g2o/apps/g2o_viewer/CMakeFiles/viewer_library.dir/build.make:230: recipe for target '../lib/libg2o_viewer.so' failed
make[2]: *** [../lib/libg2o_viewer.so] Error 1
CMakeFiles/Makefile2:475: recipe for target 'g2o/apps/g2o_viewer/CMakeFiles/viewer_library.dir/all' failed
make[1]: *** [g2o/apps/g2o_viewer/CMakeFiles/viewer_library.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

feroze, 2019-06-24:
You're missing the opengl library.

ArashJavan, 2019-12-04:
@feroze it is not right.
@foolishair opengl_helper is part of the  [g2o](https://github.com/RainerKuemmerle/g2o/blob/master/g2o/stuff/CMakeLists.txt#L29) .
IMHO [this part ](https://github.com/RainerKuemmerle/g2o/blob/master/g2o/apps/CMakeLists.txt#L8) causes the problem, since it is configurable, even tough you set opengl to OFF, as long as you have installed opengl and qt on your system g2o_viewer will be build but not opengl_helper, as a consequence the build fails. 

One solution would be, at least in my case, to add following to your cmake:
-DG2O_HAVE_OPENGL=ON

liaozihang, 2020-11-16:
In my case, add -DG2O_USE_OPENGL=ON to your cmake;

Airplane5, 2021-03-05:
> @feroze it is not right.
> @foolishair opengl_helper is part of the [g2o](https://github.com/RainerKuemmerle/g2o/blob/master/g2o/stuff/CMakeLists.txt#L29) .
> IMHO [this part [...]

I add -DG2O_HAVE_OPENGL=ON, but when make there is a new eroor:
`
[ 63%] Linking CXX executable ../../../../bin/simple_optimize
../../../../lib/libg2o_types_slam3d.so：对‘glMultMatrixd’未定义的引用    undefined reference
../../../../lib/libg2o_types_slam2d.so：对‘glPopAttrib’未定义的引用
../../../../lib/libg2o_types_slam2d.so：对‘glPointSize’未定义的引用
../../../../lib/libg2o_opengl_helper.so：对‘gluCylinder’未定义的引用
../../../../lib/libg2o_types_slam2d.so：对‘glBegin’未定义的引用
../../../../lib/libg2o_types_slam2d.so：对‘glDisable’未定义的引用
../../../../lib/libg2o_opengl_helper.so：对‘gluQuadricNormals’未定义的引用
../../../../lib/libg2o_opengl_helper.so：对‘gluSphere’未定义的引用
../../../../lib/libg2o_types_slam3d.so：对‘glNormal3f’未定义的引用
../../../../lib/libg2o_opengl_helper.so：对‘gluDeleteQuadric’未定义的引用
../../../../lib/libg2o_opengl_helper.so：对‘glEnable’未定义的引用
../../../../lib/libg2o_opengl_helper.so：对‘gluDisk’未定义的引用
../../../../lib/libg2o_opengl_helper.so：对‘gluNewQuadric’未定义的引用
../../../../lib/libg2o_opengl_helper.so：对‘glVertex2f’未定义的引用
../../../../lib/libg2o_types_slam2d.so：对‘glEnd’未定义的引用
../../../../lib/libg2o_opengl_helper.so：对‘glIsEnabled’未定义的引用
../../../../lib/libg2o_types_slam2d.so：对‘glPushAttrib’未定义的引用
../../../../lib/libg2o_opengl_helper.so：对‘gluPartialDisk’未定义的引用
../../../../lib/libg2o_opengl_helper.so：对‘glScalef’未定义的引用
../../../../lib/libg2o_types_slam2d.so：对‘glTranslatef’未定义的引用
../../../../lib/libg2o_types_slam2d.so：对‘glPushMatrix’未定义的引用
../../../../lib/libg2o_types_slam2d.so：对‘glRotatef’未定义的引用
../../../../lib/libg2o_types_slam2d.so：对‘glPopMatrix’未定义的引用
../../../../lib/libg2o_types_slam2d.so：对‘glColor3f’未定义的引用
../../../../lib/libg2o_types_slam2d.so：对‘glVertex3f’未定义的引用
collect2: error: ld returned 1 exit status
g2o/examples/simple_optimize/CMakeFiles/simple_optimize.dir/build.make:89: recipe for target '../bin/simple_optimize' failed
make[2]: *** [../bin/simple_optimize] Error 1
CMakeFiles/Makefile2:2867: recipe for target 'g2o/examples/simple_optimize/CMakeFiles/simple_optimize.dir/all' failed
make[1]: *** [g2o/examples/simple_optimize/CMakeFiles/simple_optimize.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....

`

liaozihang, 2021-03-05:
> I add -DG2O_HAVE_OPENGL=ON, but when make there is a new eroor:
> `
> [ 63%] Linking CXX executable ../../../../bin/simple_optimize
> ../../../../lib/libg2o_types_slam3d.so：对‘glMultMatrixd’未定义的引用 [...]

It seems that OpenGL is not linked,

Airplane5, 2021-03-05:
@liaozihang 
thank u for your reply.
I realize that I set G2O_USE_OPENGL=OFF.

closed 2025-02-03 by github-actions[bot]