RainerKuemmerle/g2o #363: undefined reference to `cholmod_start'

vyi, 2019-06-18:
I'm trying to build the lsd_slam_noros package and I followed [issue 206](https://github.com/RainerKuemmerle/g2o/issues/206) to fix the compile error.. However now I am getting the link error 

`[ 15%] Building CXX object lsd_slam/CMakeFiles/lsd_slam.dir/global_mapping/key_frame_graph.cc.o
[ 18%] Linking CXX static library ../../lib/liblsd_slam.a
[ 84%] Built target lsd_slam
[ 87%] Linking CXX executable ../../../bin/main_on_images
../../../lib/liblsd_slam.a(key_frame_graph.cc.o): In function `g2o::LinearSolverCholmod<Eigen::Matrix<double, 7, 7, 0, 7, 7> >::LinearSolverCholmod()':
key_frame_graph.cc:(.text._ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEEC2Ev[_ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEEC5Ev]+0xac): undefined reference to `cholmod_start'
../../../lib/liblsd_slam.a(key_frame_graph.cc.o): In function `g2o::LinearSolverCholmod<Eigen::Matrix<double, 7, 7, 0, 7, 7> >::~LinearSolverCholmod()':
key_frame_graph.cc:(.text._ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEED2Ev[_ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEED5Ev]+0x62): undefined reference to `cholmod_free_factor'
key_frame_graph.cc:(.text._ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEED2Ev[_ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEED5Ev]+0x81): undefined reference to `cholmod_finish'
../../../lib/liblsd_slam.a(key_frame_graph.cc.o): In function `g2o::LinearSolverCholmod<Eigen::Matrix<double, 7, 7, 0, 7, 7> >::init()':
key_frame_graph.cc:(.text._ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE4initEv[_ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE4initEv]+0x35): undefined reference to `cholmod_free_factor'
../../../lib/liblsd_slam.a(key_frame_graph.cc.o): In function `g2o::LinearSolverCholmod<Eigen::Matrix<double, 7, 7, 0, 7, 7> >::solve(g2o::SparseBlockMatrix<Eigen::Matrix<double, 7, 7, 0, 7, 7> > const&, double*, double*)':
key_frame_graph.cc:(.text._ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_[_ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_]+0x13c): undefined reference to `cholmod_factorize'
key_frame_graph.cc:(.text._ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_[_ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_]+0x21e): undefined reference to `cholmod_solve'
key_frame_graph.cc:(.text._ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_[_ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_]+0x26b): undefined reference to `cholmod_free_dense'
../../../lib/liblsd_slam.a(key_frame_graph.cc.o): In function `g2o::LinearSolverCholmod<Eigen::Matrix<double, 7, 7, 0, 7, 7> >::solveBlocks(double**&, g2o::SparseBlockMatrix<Eigen::Matrix<double, 7, 7, 0, 7, 7> > const&)':
key_frame_graph.cc:(.text._ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE11solveBlocksERPPdRKNS_17SparseBlockMatrixIS3_EE[_ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE11solveBlocksERPPdRKNS_17SparseBlockMatrixIS3_EE]+0x209): undefined reference to `cholmod_factorize'
key_frame_graph.cc:(.text._ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE11solveBlocksERPPdRKNS_17SparseBlockMatrixIS3_EE[_ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE11solveBlocksERPPdRKNS_17SparseBlockMatrixIS3_EE]+0x265): undefined reference to `cholmod_change_factor'
../../../lib/liblsd_slam.a(key_frame_graph.cc.o): In function `g2o::LinearSolverCholmod<Eigen::Matrix<double, 7, 7, 0, 7, 7> >::solvePattern(g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> >&, std::vector<std::pair<int, int>, std::allocator<std::pair<int, int> > > const&, g2o::SparseBlockMatrix<Eigen::Matrix<double, 7, 7, 0, 7, 7> > const&)':
key_frame_graph.cc:(.text._ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE12solvePatternERNS_17SparseBlockMatrixINS2_IdLin1ELin1ELi0ELin1ELin1EEEEERKSt6vectorISt4pairIiiESaISB_EERKNS5_IS3_EE[_ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE12solvePatternERNS_17SparseBlockMatrixINS2_IdLin1ELin1ELi0ELin1ELin1EEEEERKSt6vectorISt4pairIiiESaISB_EERKNS5_IS3_EE]+0xed): undefined reference to `cholmod_factorize'
key_frame_graph.cc:(.text._ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE12solvePatternERNS_17SparseBlockMatrixINS2_IdLin1ELin1ELi0ELin1ELin1EEEEERKSt6vectorISt4pairIiiESaISB_EERKNS5_IS3_EE[_ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE12solvePatternERNS_17SparseBlockMatrixINS2_IdLin1ELin1ELi0ELin1ELin1EEEEERKSt6vectorISt4pairIiiESaISB_EERKNS5_IS3_EE]+0x149): undefined reference to `cholmod_change_factor'
../../../lib/liblsd_slam.a(key_frame_graph.cc.o): In function `g2o::LinearSolverCholmod<Eigen::Matrix<double, 7, 7, 0, 7, 7> >::computeSymbolicDecomposition(g2o::SparseBlockMatrix<Eigen::Matrix<double, 7, 7, 0, 7, 7> > const&)':
key_frame_graph.cc:(.text._ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE28computeSymbolicDecompositionERKNS_17SparseBlockMatrixIS3_EE[_ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE28computeSymbolicDecompositionERKNS_17SparseBlockMatrixIS3_EE]+0x91): undefined reference to `cholmod_analyze'
key_frame_graph.cc:(.text._ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE28computeSymbolicDecompositionERKNS_17SparseBlockMatrixIS3_EE[_ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE28computeSymbolicDecompositionERKNS_17SparseBlockMatrixIS3_EE]+0x244): undefined reference to `cholmod_amd'
key_frame_graph.cc:(.text._ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE28computeSymbolicDecompositionERKNS_17SparseBlockMatrixIS3_EE[_ZN3g2o19LinearSolverCholmodIN5Eigen6MatrixIdLi7ELi7ELi0ELi7ELi7EEEE28computeSymbolicDecompositionERKNS_17SparseBlockMatrixIS3_EE]+0x4a0): undefined reference to `cholmod_analyze_p'
collect2: error: ld returned 1 exit status
apps/slam/CMakeFiles/main_on_images.dir/build.make:142: recipe for target '../bin/main_on_images' failed
make[2]: *** [../bin/main_on_images] Error 1
CMakeFiles/Makefile2:266: recipe for target 'apps/slam/CMakeFiles/main_on_images.dir/all' failed
make[1]: *** [apps/slam/CMakeFiles/main_on_images.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
`

mohamed1915, 2020-04-01:
Did you find any solutions to this ?

samsdolphin, 2020-04-04:
I am not sure if we encountered the same problem, but you could try to add `cholmod` to your target_link_libraries(), e.g., 

> target_link_libraries(your_executable ${someother_libs} cholmod)

closed 2025-02-03 by github-actions[bot]