RainerKuemmerle/g2o #364: What edge to use for a distance constraint?

antithing, 2019-06-19:
I have a data set which is N noisy Poses (XYZ), and known (less noisy) distances from every Pose to every other Pose.

I have it structured like:

```
struct structDistance
{
	int id_from;
	int id_to;
	double distance;
};

struct structPose
{
	int id;
	Eigen::Vector3d position;
};

```
```
std::vector<structPose> poses;
std::vector<structDistance> distances;
```

I set up the Poses as Vertices in my graph, fixing the first one:

```
for (size_t i = 0; i < poses.size(); ++i)
	{		
		Eigen::Vector3d xyz = poses[i].position;
		g2o::VertexPointXYZ* vertex(new g2o::VertexPointXYZ());
		vertex->setId(poses[i].id);
		vertex->setEstimate(xyz);
		if (i == 0)
		{
			vertex->setFixed(true);
		}
		optimizer.addVertex(vertex);		
	}
```

As edges, I know the distance from each Pose to every other Pose, from range measurements. I want to now add these distances as constraints.

I have this:

```
	//add edges. These are the distances between the poses.
	for (size_t i = 0; i < distances.size(); ++i)
	{		
			g2o::EdgePointXYZ* edge = new g2o::EdgePointXYZ();
			edge->setMeasurement(distances[i].distance); //THIS NEEDS VECTOR3D			
			edge->vertices()[0] = optimizer.vertices()[distances[i].id_from];
			edge->vertices()[1] = optimizer.vertices()[distances[i].id_to];
			optimizer.addEdge(edge);		
	}
```

But the `edge->setMeasurement` expects a `Vector3d`, and I have a double value as a distance reading.

What edge should I be using for this kind of constraint? r do i need to get a `Vector3d` from my distance measurement somehow?

Additionally, what does this line do? Do I need it?

`edge->setInformation(EdgePointXYZ::InformationType::Identity()); `

Thank you very much.

sjulier, 2019-06-22:
The constraint you have is that the distance between two vertices, x_i and x_j obeys the constraint:
d(x_i,x_j)=z
where z is your measurement. This is a 1D constraint, not a 3D one and so it's completely different from the one represented by `EdgePointXYZ`. 

I think you'll have to create your own custom edge (I don't see one in the standard types to do this). You'd need to create an edge which:

1. Inherits from g2o::base_binary_edge (because it connects two vertices). You'd specify the two vertices as VertexPointXYZ, the observation dimension as 1 and the observation type as double. (See `base_edge.h` to see how the template parameters work.)

2. You'd need to write the error function to specify the distance between the two vertices. e.g., something like:

    VertexPointXY * xy1 = static_cast<VertexPointXY *> (_vertices[0]);
    VertexPointXY * xy2 = static_cast<VertexPointXY *> (_vertices[1]);

_error[0] = _measurement - (xy1->estimate()-xy2->estimate()).norm();

You need to set the information. This is the inverse of the covariance and specifies how well you know the measurement. The more accurately you know the measurement, the higher this value. e.g., if you knew the distance with mm accuracy, the error standard deviation is 1e-3m, so the covariance is 1e-6m^2. The information is the inverse of this, so the information in this case would be 1e6. Conversely, if you knew the distance with a standard deviation of 1km, the covariance is 1e6m^2 and the information is 1e-6.

antithing, 2019-06-24:
Thank you @sjulier . Very much appreciated. I have written a new edge type, but am a little stuck on the information matrix.

When i run with an Identity matrix, i get:

```
g2o::csparse_extension::cs_cholsolsymb: cholesky failed!
Cholesky failure, writing debug.txt (Hessian loadable by Octave)
g2o::csparse_extension::cs_cholsolsymb: cholesky failed!
Cholesky failure, writing debug.txt (Hessian loadable by Octave)
g2o::csparse_extension::cs_cholsolsymb: cholesky failed!
Cholesky failure, writing debug.txt (Hessian loadable by Octave)
g2o::csparse_extension::cs_cholsolsymb: cholesky failed!
Cholesky failure, writing debug.txt (Hessian loadable by Octave)
g2o::csparse_extension::cs_cholsolsymb: cholesky failed!
Cholesky failure, writing debug.txt (Hessian loadable by Octave)
g2o::csparse_extension::cs_cholsolsymb: cholesky failed!
Cholesky failure, writing debug.txt (Hessian loadable by Octave)
g2o::csparse_extension::cs_cholsolsymb: cholesky failed!
Cholesky failure, writing debug.txt (Hessian loadable by Octave)
g2o::csparse_extension::cs_cholsolsymb: cholesky failed!
Cholesky failure, writing debug.txt (Hessian loadable by Octave)
g2o::csparse_extension::cs_cholsolsymb: cholesky failed!
Cholesky failure, writing debug.txt (Hessian loadable by Octave)
g2o::csparse_extension::cs_cholsolsymb: cholesky failed!
Cholesky failure, writing debug.txt (Hessian loadable by Octave)
iteration= 0     chi2= 0.000000  time= 0.0923078         cumTime= 0.0923078      edges= 72       schur= 0        lambda= 0.000000        levenbergIter= 10
```

I have attached my edge class, if you have a second, would you be able to take a quick look and tell me where i am wrong? Thank you very much.

[src.zip](https://github.com/RainerKuemmerle/g2o/files/3320093/src.zip)

sjulier, 2019-07-02:
Sorry - only got a chance to look at this now. What does an example dataset look like?

antithing, 2019-07-02:
Hi again, thank you, but i have this working now. I was able to adjust an edge type from here:

`https://github.com/wang-chen/localization/blob/master/src/types/types_edge_se3range.cpp`

for my use. 

Additionally, the crash that I was seeing vanished when I switched from `BlockSolverX `to:

```
std::unique_ptr<g2o::BlockSolver_6_3::LinearSolverType> linearSolver	
linearSolver = g2o::make_unique<g2o::LinearSolverDense<g2o::BlockSolver_6_3::PoseMatrixType>>();
```

Thank you again for your time.

sjulier, 2019-07-02:
Excellent.

closed 2019-07-02 by sjulier

YangSiri, 2019-11-11:
Hi, @antithing 
I run into this problem these days. But i still wonder your solution. You said your pose is 3 dimension (XYZ). How come BlockSolver63 help you? Isn't that for 6-d pose and 3-d landmark ?

Looking forward to your reply.