RainerKuemmerle/g2o #371: Custom unary edge with vse2 to reproject 3d point throws memory error

SupunPerera1994, 2019-08-23:


SupunPerera1994, 2019-08-23:
[image]

SupunPerera1994, 2019-08-23:
[image]

SupunPerera1994, 2019-08-23:
The only change was in the SE2.h class where the map function was changed to accommodate the re-projection calculation

[image]

sjulier, 2019-08-23:
Could you compile with debug enabled to get the full stack trace?

SupunPerera1994, 2019-08-26:
> Could you compile with debug enabled to get the full stack trace?

Thank you for the suggestion, i will try it that way. For now I have only tried it as a Thirdparty package within orbslam2 and have no errors at compile time.

The error i have shown above only occurs at runtime.

closed 2025-02-03 by github-actions[bot]