RainerKuemmerle/g2o #378: Difference pose prior and pose set fixed

MinaHenein, 2019-10-11:
Hi there,
I'm new to g2o and I'd like to understand the difference between setting a first camera pose to fixed vs setting a prior edge on this camera pose?

Is it that the 'setFixed' is really locking the first camera pose, and the edge prior is trying to put a prior on a pose however it is still allowed to vary based on the covariance/information associated with this edge? Is this correct?

Please clarify.

RainerKuemmerle, 2020-06-28:
setFixed is a hard constraint. That vertex will stay at its initial values.
A Prior is a "normal" error term to be considered in the optimization.
Your summary is correct.

closed 2020-06-28 by RainerKuemmerle

sjulier, 2020-06-28:
I have wondered if it might be better to describe a vertex as conditioned on rather than fixed.

The reason is that (in Kalman filter land) there is something called a Schmidt Kalman Filter. In the SKF, the numerical value of some states can be fixed. This means that the state values do not change during a predict and update step. However, the covariance on these states is non-zero and unchanging. As a result, correlations can exist between both your fixed and updated states. (The SKF was used a lot in an inertial navigation system where you can have more than a hundred calibration states which you don't want to update in real time, but you'd like to know how they affect the accuracy of the position solution.)

In g2o, when you set a vertex as fixed, it's not included in the optimization step and implicitly is assigned a covariance of zero. Setting a covariance to zero implies that we are conditioning on the values of the states.

Anyway, minor thought...