RainerKuemmerle/g2o #379: edge setParameterId

MinaHenein, 2019-10-11:
Hi there,
I'm new to g2o and I can't really understand the use of edge set parameter id, and I'm failing to find enough examples to understand this.

Basically for every landmark observation and the camera prior, I had to add a line:
edge->setParameterId(0, 0); otherwise it doesn't run, but I'm not sure what it actually means.

Thanks a lot

YangSiri, 2019-10-26:
Hi, MinaHenein, do you have any thoughts about that ?

MinaHenein, 2019-10-28:
Hi YangSiri, I think that for every landmark observation, setParameterId(cam_id) makes the system know that this observation happens from the camera with id = cam_id and I guess it is related to: 

g2o::ParameterSE3Offset* cameraOffset = new g2o::ParameterSE3Offset;
cameraOffset->setId(0);
optimizer.addParameter(cameraOffset); 

however these are all my thoughts/understandings, I can't confirm anything yet.

YangSiri, 2019-10-28:
yes, i thought about that too, its all about the camera paras

nickponline, 2020-03-13:
@MinaHenein @YangSiri  Did either of you get an answer? I'm have the same question .. :)

MinaHenein, 2020-03-13:
I didn't get any answer, but I'm fairly confident my hypothesis is valid

RainerKuemmerle, 2020-06-13:
Yes, it relates to the given ParameterSE3Offset for which you have to provide the ID.

closed 2020-06-13 by RainerKuemmerle