RainerKuemmerle/g2o #380: when i use EdgeSE3Prior together with EdgeSE3XYZPrior, the optmizer cannot converge the right place

rjasd1128hf, 2019-10-11:
g2o::SparseOptimizer optimizer;
std::unique_ptr<g2o::BlockSolverX::LinearSolverType> linearSolver(new g2o::LinearSolverCholmod<g2o::BlockSolverX::PoseMatrixType>());
std::unique_ptr<g2o::BlockSolverX> blockSolver(new g2o::BlockSolverX(std::move(linearSolver)));     
g2o::OptimizationAlgorithmLevenberg* optimizationAlgorithm = new g2o::OptimizationAlgorithmLevenberg(std::move(blockSolver));

optimizer.setAlgorithm(optimizationAlgorithm);
optimizer.setVerbose(true);
g2o::ParameterSE3Offset* cameraOffset = new g2o::ParameterSE3Offset;
cameraOffset->setId(0);
optimizer.addParameter(cameraOffset);

Eigen::Matrix<double, 6, 6> inf = Eigen::Matrix<double, 6, 6>::Zero();

inf(0, 0) = 1.0;	inf(0, 1) = 0.0;	inf(0, 2) = 0.0;
inf(1, 0) = 0.0;	inf(1, 1) = 1.0;	inf(1, 2) = 0.0;
inf(2, 0) = 0.0;	inf(2, 1) = 0.0;	inf(2, 2) = 1.0;
inf(3, 3) = 1.0;	inf(3, 4) = 0.0;	inf(3, 5) = 0.0;
inf(4, 3) = 0.0;	inf(4, 4) = 1.0;	inf(4, 5) = 0.0;
inf(5, 3) = 0.0;	inf(5, 4) = 0.0;	inf(5, 5) = 1.0;

Eigen::Isometry3d t;
Eigen::Matrix3d rotrpy = g2o::internal::fromEuler(Eigen::Vector3d(-58.0118418734 * deg2rad, -4.2868610819 * deg2rad, 27.3365702914 * deg2rad));//which deg2rad = PI/180.0;
t = rotrpy;
t.translation() = Eigen::Vector3d(-2848321.07828, 4660858.32744, 3282028.80025);
g2o::VertexSE3* v = new g2o::VertexSE3;
v->setId(0);
v->setEstimate(t);
optimizer.addVertex(v);

size_t m_unaryEdgeNum = 0;
g2o::EdgeSE3Prior* e = new g2o::EdgeSE3Prior;
e->setVertex(0, v);
e->setId(m_unaryEdgeNum++);
e->setParameterId(0, 0);
e->setMeasurement(t);
e->setInformation(inf);
optimizer.addEdge(e);

Eigen::Vector3d pos_new_z(t.translation()(0) + 10.0, t.translation()(1) + 10.0, t.translation()(2) - 5.0);
Eigen::Matrix3d pos_inf_new_z = Eigen::Matrix3d::Zero();
pos_inf_new_z(0, 0) = 1E8;
pos_inf_new_z(1, 1) = 1E8;
pos_inf_new_z(2, 2) = 1E8;

g2o::EdgeSE3XYZPrior* e_new_z = new g2o::EdgeSE3XYZPrior;
e_new_z->setVertex(0, v);
e_new_z->setId(m_unaryEdgeNum++);
e_new_z->setParameterId(0, 0);
e_new_z->setMeasurement(pos_new_z);
e_new_z->setInformation(pos_inf_new_z);
optimizer.addEdge(e_new_z);

bool psdFlag = optimizer.verifyInformationMatrices(true);
if (!psdFlag)
{
	std::cout <<"no psd.\n";
	return;
}
	optimizer.initializeOptimization();
	int iteration_count = optimizer.optimize(2000);

the result converge

rjasd1128hf, 2019-10-11:
below is the iteration information, the chi2 decrease very slowly ,and lambda increase
iteration= 0     chi2= 4623606996.417544         time= 8126.51   cumTime= 8126.51        edges= 2        schur= 0       lambda= 1333.333333      levenbergIter= 1
iteration= 1     chi2= 4278804538.230045         time= 94.9293   cumTime= 8221.44        edges= 2        schur= 0       lambda= 29127111.111111  levenbergIter= 6
iteration= 2     chi2= 4278804538.015624         time= 6.41647   cumTime= 8227.86        edges= 2        schur= 0       lambda= 1334399889591257600.000000       levenbergIter= 9
iteration= 3     chi2= 4278804537.795496         time= 13.569    cumTime= 8241.43        edges= 2        schur= 0       lambda= 889599926394171648.000000        levenbergIter= 1
iteration= 4     chi2= 4278804537.579648         time= 2.52369   cumTime= 8243.95        edges= 2        schur= 0       lambda= 1186133235192228864.000000       levenbergIter= 2
iteration= 5     chi2= 4278804537.570018         time= 1.35656   cumTime= 8245.31        edges= 2        schur= 0       lambda= 1581510980256305152.000000       levenbergIter= 2
iteration= 6     chi2= 4278804537.352566         time= 0.626045  cumTime= 8245.93        edges= 2        schur= 0       lambda= 1054340653504203392.000000       levenbergIter= 1
iteration= 7     chi2= 4278804537.341733         time= 1.3018    cumTime= 8247.23        edges= 2        schur= 0       lambda= 1405787538005604352.000000       levenbergIter= 2
iteration= 8     chi2= 4278804537.122474         time= 0.833714  cumTime= 8248.07        edges= 2        schur= 0       lambda= 937191692003736192.000000        levenbergIter= 1
iteration= 9     chi2= 4278804536.907278         time= 3.53564   cumTime= 8251.6         edges= 2        schur= 0       lambda= 1249588922671648256.000000       levenbergIter= 2
iteration= 10    chi2= 4278804536.898137         time= 4.69e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 1666118563562197504.000000       levenbergIter= 2
iteration= 11    chi2= 4278804536.681418         time= 2.76e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 1110745709041464960.000000       levenbergIter= 1
iteration= 12    chi2= 4278804536.671135         time= 3.33e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 1480994278721953280.000000       levenbergIter= 2
iteration= 13    chi2= 4278804536.452702         time= 2.72e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 987329519147968768.000000        levenbergIter= 1
iteration= 14    chi2= 4278804536.238124         time= 4.62e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 1316439358863958272.000000       levenbergIter= 2
iteration= 15    chi2= 4278804536.229448         time= 3.37e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 1755252478485277696.000000       levenbergIter= 2
iteration= 16    chi2= 4278804536.013425         time= 2.67e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 1170168318990184960.000000       levenbergIter= 1
iteration= 17    chi2= 4278804536.003664         time= 5.34e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 1560224425320246528.000000       levenbergIter= 2
iteration= 18    chi2= 4278804535.786014         time= 5.72e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 1040149616880164352.000000       levenbergIter= 1
iteration= 19    chi2= 4278804535.775033         time= 3.98e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 1386866155840219136.000000       levenbergIter= 2
iteration= 20    chi2= 4278804535.555553         time= 5e-05     cumTime= 8251.6         edges= 2        schur= 0       lambda= 924577437226812672.000000        levenbergIter= 1
iteration= 21    chi2= 4278804535.340191         time= 3.74e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 1232769916302416896.000000       levenbergIter= 2
iteration= 22    chi2= 4278804535.330925         time= 6e-05     cumTime= 8251.6         edges= 2        schur= 0       lambda= 1643693221736555776.000000       levenbergIter= 2
iteration= 23    chi2= 4278804535.114019         time= 4.93e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 1095795481157703808.000000       levenbergIter= 1
iteration= 24    chi2= 4278804535.103596         time= 6.98e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 1461060641543604992.000000       levenbergIter= 2
iteration= 25    chi2= 4278804534.884952         time= 6.28e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 974040427695736576.000000        levenbergIter= 1
iteration= 26    chi2= 4278804534.670217         time= 5.46e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 1298720570260982016.000000       levenbergIter= 2
iteration= 27    chi2= 4278804534.661422         time= 3.29e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 1731627427014642688.000000       levenbergIter= 2
iteration= 28    chi2= 4278804534.445221         time= 4.53e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 1154418284676428288.000000       levenbergIter= 1
iteration= 29    chi2= 4278804534.435328         time= 4.53e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 1539224379568570880.000000       levenbergIter= 2
[... cut ...]
iteration= 36    chi2= 4278804533.534058         time= 5.22e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 1441395303775232000.000000       levenbergIter= 2
iteration= 37    chi2= 4278804533.315201         time= 2.99e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 960930202516821248.000000        levenbergIter= 1
iteration= 38    chi2= 4278804533.100306         time= 3.91e-05  cumTime= 8251.6         edges= 2        schur= 0       lambda= 1281240270022428160.000000       levenbergIter= 2

RainerKuemmerle, 2020-06-28:
Provide the g2o file for analyzing or a ready to compile example project.

LongruiDong, 2020-07-28:
Hi  Have you known where the problem is?

closed 2025-02-02 by github-actions[bot]