RainerKuemmerle/g2o #400: Compilation error - unable to locate cs.h

krishnakumarranganathan, 2019-12-15:
I am compiling openvslam as per the installation instructions provided.

from openvslam/openvslam/src/openvslam/optimize/local_bundle_adjuster.cc:20:
/usr/local/include/g2o/solvers/csparse/csparse_extension.h:27:16: fatal error: cs.h: No such file or directory

I did a cmake with the following options
cmake
-DBUILD_WITH_MARCH_NATIVE=ON
-DUSE_PANGOLIN_VIEWER=ON
-DUSE_SOCKET_PUBLISHER=OFF
-DUSE_STACK_TRACE_LOGGER=OFF
-DBOW_FRAMEWORK=DBoW2
-DBUILD_TESTS=ON
..

Stack Trace logging was disabled as I was seeing glog cmake errors while running cmake (which I believe is ok).

I followed the comments at "#53" and also installed libsuitesparse-dev, but no luck. 
cs.h is installed at /usr/local//include/g2o/EXTERNAL/csparse/. g2o compiles file and am able to execute few binaries from the g2o/bin directory.

Any help would be appreciated.

Thanks
KK

taronegeorage, 2019-12-20:
You should add CSparse to your cmake project in order to find the file.
`find_package(CSparse REQUIRED)`
`include_directories(${CSPARSE_INCLUDE_DIR})`

closed 2020-08-28 by RainerKuemmerle

mgrallos, 2023-02-23:
> `find_package(CSparse REQUIRED)`
> `include_directories(${CSPARSE_INCLUDE_DIR})`

hi I added this to my cmakelist.txt. And this occurs 
[image]

mgrallos, 2023-02-23:
> I am compiling openvslam as per the installation instructions provided.
>
> from openvslam/openvslam/src/openvslam/optimize/local_bundle_adjuster.cc:20: [...]

I was doing the same thing and encountered the same error. Please let me know if you have solved this error.